sle118/squeezelite-esp32 #121: TAS5711 Support?

danielehrhardt, 2021-11-07:
Hi 
one Question. 

Do you think the TAS5711 does also work with squeezelite-esp32?
I created a custom amplifier board with the TAS5711. 
But currently i am not able to get it to work. With TAS57XX and TAS5713 config. 

Best,
Daniel

sle118, 2021-11-17:
These amplifier chips are initialized at startup with some i2c instructions. If the chip isn't detected, you will get a warning message in the logs.

Could you share some details of your design and also share some logs please?

philippe44, 2021-11-17:
And you can definitively use dac_controlset to build your own initialization with the default I2S build. Once you are sure of it, I can include it in the well-known codecs

closed 2021-12-07 by philippe44