sle118/squeezelite-esp32 #123: Other settings for esp32 A1S Audio Kit v2.2 3378 series

superjasiek, 2021-11-16:
Hi philippe44,
I'm using your software for a year or so but finally i could test it on esp32 A1S Audio Kit v2.2 3378 series. Unfortunately the setting that you provided does not work for me (this version of kit use for ES8388 codec). Manufacturer changed pin assignment and for this model settings should be like this:

- dac_config: model=ES8388,bck=27,ws=25,do=26,sda=33,scl=32,i2c=16
(more about the problem you can find [here](https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/issues/26))

After successful initialization i cant find correct setting for jack and built-in amp - when i connect jack the music does not stop playing  from the speakers. (i was trying to use set_GPIO: 21=amp,22=green:0,39=jack:0) 
Any idea?
Thanks for your work and looking forward to read from you.

philippe44, 2021-11-17:
These boards seem to be a real mess. I have a version that works with the settings I've mentionned, but it seems they have done yet another version and changed again the pinout. I don't believe their schematics. Seing a button created for jack is normal.

superjasiek, 2021-11-17:
Thanks for advice. Unfortunately no luck - flashed new firmware I2S-4MFlash.16.751.master-cmake, mode: Squeezelite-ESP32. gpio:  tried for jack 
19
35
29
39
without any success.  The only change i can see is the lack of blinking (or constantly up during playing music) red led (D4) after changing firmware
Music still playes from speakers and headphones in the same time.

philippe44, 2021-11-17:
What do you have in the log when it boots now?

superjasiek, 2021-11-17:
Do not know what part is interesting for you so here it goes:
`I (61) boot: Chip Revision: 1
I (63) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.0.2-206-g5630b17e1 2nd stage bootloader
I (39) boot: compile time 03:19:55
I (39) boot: Enabling RNG early entropy source...
I (44) qio_mode: Enabling default flash chip QIO
I (50) boot: SPI Speed      : 80MHz
I (54) boot: SPI Mode       : QIO
I (58) boot: SPI Flash Size : 4MB
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 recovery         factory app      00 00 00010000 00140000
I (103) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (110) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (118) boot: End of partition table
I (122) boot_comm: chip revision: 1, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=0x00150020 vaddr=0x3f400020 size=0xdfa5c (916060) map
I (433) esp_image: segment 1: paddr=0x0022fa84 vaddr=0x3ffbdb60 size=0x0058c (  1420) load
I (434) esp_image: segment 2: paddr=0x00230018 vaddr=0x400d0018 size=0x17c284 (1557124) map
I (940) esp_image: segment 3: paddr=0x003ac2a4 vaddr=0x3ffbe0ec size=0x064a8 ( 25768) load
I (949) esp_image: segment 4: paddr=0x003b2754 vaddr=0x40080000 size=0x00400 (  1024) load
I (950) esp_image: segment 5: paddr=0x003b2b5c vaddr=0x40080400 size=0x1e44c (123980) load
I (1004) esp_image: segment 6: paddr=0x003d0fb0 vaddr=0x400c0000 size=0x00068 (   104) load
I (1023) boot: Loaded app from partition at offset 0x150000
I (1023) boot: Disabling RNG early entropy source...
I (1024) psram: This chip is ESP32-D0WD
I (1028) spiram: Found 64MBit SPI RAM device
I (1033) spiram: SPI RAM mode: flash 80m sram 80m
I (1038) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1046) cpu_start: Pro cpu up.
I (1050) cpu_start: Application information:
I (1055) cpu_start: Project name:     Squeezelite-ESP32
I (1060) cpu_start: App version:      I2S-4MFlash.16.751.master-cmake
I (1068) cpu_start: Compile time:     Nov 14 2021 03:21:27
I (1074) cpu_start: ELF file SHA256:  4c40a6e8dcfb8641...
I (1080) cpu_start: ESP-IDF:          v4.0.2-206-g5630b17e1
I (1086) cpu_start: Starting app cpu, entry point is 0x40081a3c
I (0) cpu_start: App cpu up.
I (1592) spiram: SPI SRAM memory test OK
I (1595) heap_init: Initializing. RAM available for dynamic allocation:
I (1596) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1599) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1605) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1612) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1618) heap_init: At 3FFC7650 len 000189B0 (98 KiB): DRAM
I (1624) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1631) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1637) heap_init: At 4009E84C len 000017B4 (5 KiB): IRAM
I (1643) cpu_start: Pro cpu start user code
I (1648) spiram: Adding pool of 4048K of external SPI memory to heap allocator
I (1669) spi_flash: detected chip: generic
I (1669) spi_flash: flash io: qio
I (1669) esp_core_dump_common: Init core dump to UART
E (1678) esp_core_dump_common: No core dump partition found!
I (1679) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1690) spiram: Reserving pool of 64K of internal memory for DMA/internal allocations
I (1700) esp_app_main: Reset reason is: 1
I (1700) esp_app_main: Starting app_main
I (1710) nvs_utilities: Initializing flash nvs
I (1740) nvs_utilities: Initializing nvs partition settings
I (1770) esp_app_main: Setting up telnet.
I (1770) telnet: Telnet support disabled
I (1770) esp_app_main: Setting up config subsystem.
I (1820) esp_app_main: Registering default values
I (1820) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[... 55 lines cut ...]
To automatically execute lines at startup:
        Set NVS variable autoexec (U8) = 1 to enable, 0 to disable automatic execution.
        Set NVS variable autoexec[1~9] (string)to a command that should be executed automatically

W (3270) console: Processing autoexec commands while wifi_manager active.  Wifi related commands will be ignored.
I (3270) console: autoexec is set to perform auto-process
I (3280) console: Running command autoexec1 = squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n squeezelite-49790
W (3290) config: Value not found for key autoexec2
I (3290) squeezelite_cmd: Calling squeezelite
I (3290) wifi:I (3300) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
[00:00:01.650307] stream_init:387 init stream
[00:00:01.651715] sb_controls_init:244 initializing audio (buttons/rotary/ir) controls (raw:0)
                                                                                                                                                                                                              I (3330) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[00:00:01.675766] sb_displayer_init:312 no display or led visualizer for LMS
[... cut ...]
[00:00:01.690789] output_init_common:434 supported rates: 192000 176400 96000 88200 48000 44100 32000 24000 22050 16000 12000 11025 8000
[00:00:01.703042] output_init_embedded:90 init I2S/SPDIF
I (3380) DAC core: DAC uses I2C port:0, sda:33, scl:32
I (3380) DAC external: DAC on I2C @16
I (3380) DAC external: Configuring MCLK on GPIO0
I (3400) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=12
I (3400) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
`

philippe44, 2021-11-17:
You don't have the latest build, you need to update

superjasiek, 2021-11-17:
Flashed the newest firmware, tried gpio:
19
35
29
39
with same result - playing both speakers and headphones
Only difference i see here:

I (2871) monitor: Adding jack (low) detection GPIO 29
I (2881) buttons: Creating button using GPIO 29, type 0, pull-up/down 0, long press 0 shifter -1
E (2891) gpio: gpio_set_direction(234): GPIO number error
I (2891) buttons: starting button tasks
E (2901) gpio: gpio_set_intr_type(101): GPIO number error
E (2901) gpio: gpio_isr_handler_add(383): GPIO number error
E (2911) gpio: gpio_intr_enable(129): GPIO number error
I (2921) monitor: Heap internal:230536 (min:230184) external:4114992 (min:4114992)

after pressing pause i get:
[00:20:53.347740] output_thread_i2s:449 Output state is 1
[00:20:53.350537] output_thread_i2s:512 Restarting I2S.
[00:20:53.355183] equalizer_open:71 equalizer initialized 0
[00:20:53.396326] mad_decode:247 setting track_start
[00:20:53.469772] _output_frames:65 start buffer frames: 2927
[00:20:53.481139] _output_frames:150 track start sample rate: 44100 replay_gain: 0
[00:20:53.483275] output_thread_i2s:449 Output state is 2
[00:22:31.921572] output_thread_i2s:449 Output state is 0
[00:23:02.525067] output_thread_i2s:449 Output state is -1
[00:23:02.526226] output_thread_i2s:453 switching off amp GPIO 21

philippe44, 2021-11-17:
It's strange - you should see, when you set 39, an entry log that mentions that 39 is a gpio with polling. What is the version number you see?

superjasiek, 2021-11-17:
(1055) cpu_start: Project name:     Squeezelite-ESP32
I (1061) cpu_start: App version:      I2S-4MFlash.16.759.master-cmake
I (1068) cpu_start: Compile time:     Nov 17 2021 18:37:26
I (1074) cpu_start: ELF file SHA256:  656f82f93f4556bb...
I (1080) cpu_start: ESP-IDF:          v4.0.2-206-g5630b17e1

philippe44, 2021-11-17:
oh, you probably have not set "jack_mutes_amp" to "y". The log for the jack insertion is silenced, unless jack_mutes_amp is enabled

superjasiek, 2021-11-18:
Flashed newest version - fw: I2S-4MFlash.16.761.master-cmake, set jack_mutes_amp" to "y" and set jack gpio to 39 but unfortunately is still the same. Also do no see any information about insertion. Same as yesteday:

I (2859) led: Configuring LEDs green:22 (active:0 0%), red:0 (active:0 0%)
I (2859) battery: No battery
I (2869) monitor: Initializing monitoring
I (2869) monitor: Adding jack (low) detection GPIO 39
I (2879) buttons: Creating button using GPIO 39, type 0, pull-up/down 0, long press 0 shifter -1
I (2889) buttons: starting button tasks
W (2889) buttons: creating polled gpio 39, level 1
I (2899) monitor: Heap internal:230488 (min:230184) external:4114992 (min:4114992)

philippe44, 2021-11-18:
The I really don't know because I checked on the same board and it worked

philippe44, 2021-11-18:
Of course you tried both jacks?

superjasiek, 2021-11-18:
I really appreciate your help. So for sure for version 3378 it is gpio39. Would you be so kind and tell me dip switches configuration? i will try to make fresh start. There are 2 jacks inputs - one for headphones other is line in - tried both.

philippe44, 2021-11-19:
DIP switch is on,on,on,off,off (reverse of yours)

philippe44, 2021-11-20:
I'm sorry but I really don't know what can be the issue with that board then

superjasiek, 2021-11-20:
No problem, anyway the software you make is grate :-) I made some measurements and first of all - the pad from jack for was not soldiered properly. 
[image]
For model 3378 GPIO for jack detection is 39. Voltage when jack is disconnected is about 3,12v and when connected is 0v.  Software does recognize the state of jack:
[00:01:10.821786] jack_handler:153 switching amplifier ON
[00:01:50.521777] jack_handler:153 switching amplifier OFF
[00:03:03.621878] jack_handler:153 switching amplifier ON
[00:03:27.721769] jack_handler:153 switching amplifier OFF
But unfortunately the amp is still active an i can hear both - speakers and headphones.
Some guy has mention about ISR detection mechanism [here ](https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/issues/26#issuecomment-974556076) but i do no have a clue what it means.

philippe44, 2021-11-20:
ah... well the AC101 had a speaker on/off method and I just saw that ES8388 does not

philippe44, 2021-11-20:
Should be fixed in build 762

superjasiek, 2021-11-20:
Yeah! Works grate. Thank you very much for your patience.

closed 2021-11-20 by superjasiek

bootrino, 2021-12-28:
This blog post might help:

https://www.pschatzmann.ch/home/2021/12/06/the-ai-thinker-audio-kit-experience-or-nothing-is-right/

mwolter805, 2022-06-18:
Here's another tutorial. This one's for the ESP32 Audio Kit V2.2 A149 OR 3478 (Chip ID 2974)
https://screenzone.eu/esp32-squeezelite-on-esp32-audio-kit/

```
dac_config: model=ES8388,bck=27,ws=25,do=26,sda=33,scl=32,i2c=16
set_GPIO: 21=amp,22=green:0,39=jack:0
```

NuclearEMP, 2025-11-16:
Took me a few days to get control of my A1S but finally have a working web radio with webUI and stereo output and button control with Vol+, Vol-, ch+,ch- and power on and off.