sle118/squeezelite-esp32 #126: Coredump with BT enabled

dkotte, 2021-12-25:
If i enable BT i get the following coredump
(the only way to recover is to reflash the device ... is there another way)

`I (2421) battery: No battery
I (2421) monitor: Initializing monitoring
I (2421) monitor: Heap internal:270660 (min:269828) external:4140956 (min:4140956)
I (2431) esp_app_main: Initializing display
I (2441) display: Trying to configure display with SPI,width=240,height=320,cs=22,reset=17,speed=8000000,driver=ILI9341
I (2451) ILI9341: ILI9341_Detect 391 Driver= driver=ILI9341   Depth=16
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400902d7  PS      : 0x00060d33  A0      : 0x8009258e  A1      : 0x3ffbc570
A2      : 0xffffffff  A3      : 0x00060120  A4      : 0x00000001  A5      : 0x0000cdcd
A6      : 0xb33fffff  A7      : 0x00000001  A8      : 0x0000abab  A9      : 0x3ffbc690
A10     : 0x0000cdcd  A11     : 0x3ffc28f0  A12     : 0x00000000  A13     : 0x00000000
A14     : 0xffffffff  A15     : 0x00000001  SAR     : 0x0000000f  EXCCAUSE: 0x0000001d
EXCVADDR: 0xffffffff  LBEG    : 0x4008b9ce  LEND    : 0x4008b9d9  LCOUNT  : 0x00000000

ELF file SHA256: a2dfa563d06f01b2

Backtrace: 0x400902d4:0x3ffbc570 0x4009258b:0x3ffbc5a0 0x400925c0:0x3ffbc5c0 0x40083c2a:0x3ffbc5f0 0x40083c4e:0x3ffbc620 0x4009295d:0x3ffbc640 0x400ef3e6:0x3ffbc660 0x400ee181:0x3ffbc690 0x400eefcc:0x3ffbc6d0 0x400edb77:0x3ffbc740 0x400d91ac:0x3ffbc7a0 0x400d45c4:0x3ffbc7c0

I (2639) esp_core_dump_uart: Press Enter to print core dump to UART...
I (2646) esp_core_dump_uart: Print core dump to uart...
I (2652) esp_core_dump_common: Found tasks: (7)!
================= CORE DUMP START =================
OBcAAAEAAAAHAAAAZAEAAA==
UMj7P7DE+z9IyPs/
sMX7P+DH+z+EAAAArCH8P6wh/D9QyPs/pCH8PxgAAABgSv0/YEr9P1DI+z8AAAAA
AQAAAEym+z9tYWluAJ9w4tBZX//6NlwAAAAAAEjI+z8AAAAAIAsGAAEAAAAAAAAA
AAAAAAAAAAAAAAAAtImAPxyKgD+EioA/AAAAAAAAAAABAAAAAAAAAAAAAADATP0/
GK8IQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
`

dkotte, 2021-12-28:
Today i cleared all config (Video/SPI/... everything with "clear" button)
I enabled BT and reboot. Now the Coredump looks other ...
Seems there is am "-R" parameter within squeezelite autostart ...
`
****************************
Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
****************************
W (3195) console: Processing autoexec commands while wifi_manager active.  Wifi related commands will be ignored.
I (3205) console: autoexec is set to perform auto-process
I (3205) console: Running command autoexec1 = squeezelite -b 500:2000 -d all=info -C 30 -W -n "squeezelite-ESP32" -o "BT" -R -Z 192000
W (3225) config: Value not found for key autoexec2
I (3225) squeezelite_cmd: Calling squeezelite
I (3235) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
squeezelite-esp32>
Option error: -R

Squeezelite v1.0-859-32, Copyright 2012-2015 Adrian Smith, 2015-2019 Ralph Irving. See -t for license terms
Usage: squeezelite [options]
  -s <server>[:<port>]  Connect to specified server, otherwise uses autodiscovery to find server
  -a <f>                Specify sample format (16|24|32) of output file when using -o - to output samples to stdout (interleaved little endian only)
  -b <stream>:<output>  Specify internal Stream and Output buffer sizes in Kbytes
  -c <codec1>,<codec2>  Restrict codecs to those specified, otherwise load all available codecs; known codecs: flac,pcm,mp3,ogg (mad,mpg for specific mp3 codec)
                        Codecs reported to LMS in order listed, allowing codec priority refinement.
  -C <timeout>          Close output device when idle after timeout seconds, default is to keep it open while player is 'on'
  -d <log>=<level>      Set logging level, logs: all|slimproto|stream|decode|output, level: info|debug|sdebug
  -e <codec1>,<codec2>  Explicitly exclude native support of one or more codecs; known codecs: flac,pcm,mp3,ogg (mad,mpg for specific mp3 codec)
  -f <logfile>          Write debug to logfile
  -m <mac addr>         Set mac address, format: ab:cd:ef:12:34:56
  -M <modelname>        Set the squeezelite player model name sent to the server (default: SqueezeESP32)
  -n <name>             Set the player name
  -N <filename>         Store player name in filename to allow server defined name changes to be shared between servers (not supported with -n)
  -W                    Read wave and aiff format from header, ignore server parameters
  -r <rates>[:<delay>]  Sample rates supported, allows output to be off when squeezelite is started; rates = <maxrate>|<minrate>-<maxrate>|<rate1>,<rate2>,<rate3>; delay = optional delay switching rates in ms
  -t                    License terms
  -?                    Display this help text

Build options: EMBEDDED LOOPBACK NO_FAAD LINKALL

abort() was called at PC 0x400d4b74 on core 1

ELF file SHA256: a57daf7349ccc36b

Backtrace: 0x40094e0f:0x3f801bf0 0x4009517d:0x3f801c10 0x400d4b74:0x3f801c30 0x400ec437:0x3f801c50 0x400da67c:0x3f801f70

I (2329) esp_core_dump_uart: Press Enter to print core dump to UART...
I (2336) esp_core_dump_uart: Print core dump to uart...
I (2342) esp_core_dump_common: Found tasks: (13)!
================= CORE DUMP START =================
QCQAAAEAAAALAAAAZAEAAA==
KNf7P6DV+z8g1/s/
oNX7P8DW+z/XA1CvwFH8P8TP+z8o1/s/uFH8PxkAAABg3z0Ipv72qijX+z8AAAAA
AAAAACTR+z9JRExFMQCClJdgt2MefIMAAQAAACDX+z8AAAAAIQAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAVPmAP7z5gD8k+oA/AAAAAAAAAAABAAAAAAAAAAAAAAAAAAAA
tBYJQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAADctmk=
VEEIQALkI0AwAQYA1G0NgGDW+z8AAAAAIwQGACAEBgCwgfw/AwAAACMABgCWaA2A
MNb7PwAAAADo7vs/6O77PwAAAAAgAAYAAAAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAAFRDCEAAAAAAQHAJQAxj7z8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABLiwmAgNb7P7RP/D8IAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAACg1vs/AAAAAEhR/D8BAAAAAAAAAAAAAAABAACA
AAAAAMDW+z8AAAAAAAAAAAEAAAABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAADM1vs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
vM/7PzDO+z+0z/s/
MM77P1DP+z+Y3kbFMNf7P8BR/D+8z/s/uFH8PxkAAAAsIoFdwEDt5LzP+z8AAAAA
[... 180 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAylPw=
VEEIQPgaCEAwDQYARHkJgFCd+z8BAAAA/////7BT/D/M6vs//gAAACADBgD4GgiA
MJ37P+AA8D8BAAAAKNz7PwEAAAD+AAAAHAD0PwAAAAAAAAAAVEMIQAAAAAAAAAAA
AAAAAFRDCEABAAAAQHAJQBwq7z8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAMSNCYBQnfs/zOr7PyAEBgDbOQiAcJ37P9h++z8AAAAA
QJ77PwEAAAD+AAAAHAD0PwAAAACwnfs/yE/8PwAAAAAAAAAAhwAAAP4AAAAgAwYA
/////wAAAADIT/w/AAAAACR/+z8AAAAA2H77PwAAAAAAAAAA0J37PwAAAAAAAAAA
AAAAADjCCEABAAAAAAAAAEwYCICAff4/KwAAACsAAAAAAAAAAAAAANyd+z8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
================= CORE DUMP END =================
E (3463) esp_core_dump_common: Skipped 2 tasks with bad TCB!
I (3469) esp_core_dump_uart: Core dump has been written to uart.
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
`

philippe44, 2022-01-07:
BT source has been updated recently - is this still happening?

dkotte, 2022-01-08:
no, won't help ... unfortunately.
Do you see/check the "-R" parameter comming with BT-Config and killing Squeezelite?
And do I have to reflash the system now or is there another way to switch to recovery mode?

17:33:58.814 -> [0;32mI (3745) console: autoexec is set to perform auto-process[0m
17:33:58.814 -> [0;32mI (3745) console: Running command autoexec1 = squeezelite -b 500:2000 -d all=info -C 30 -W -n "squeezelite-ESP32" -o "BT" -R -Z 192000[0m
17:33:58.814 -> [0;33mW (3765) config: Value not found for key autoexec2[0m
17:33:58.814 -> [0;32mI (3765) squeezelite_cmd: Calling squeezelite[0m
17:33:58.814 -> [0;32mI (3775) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
**17:33:58.814 -> squeezelite-esp32> 
17:33:58.814 -> Option error: -R**
17:33:58.814 -> 
17:33:58.814 -> Squeezelite v1.0-859-32, Copyright 2012-2015 Adrian Smith, 2015-2019 Ralph Irving. See -t for license terms
17:33:58.814 -> Usage: squeezelite [options]
17:33:58.814 ->   -s <server>[:<port>]	Connect to specified server, otherwise uses autodiscovery to find server
17:33:58.814 ->   -a <f>		Specify sample format (16|24|32) of output file when using -o - to output samples to stdout (interleaved little endian only)
17:33:58.814 ->   -b <stream>:<output>	Specify internal Stream and Output buffer sizes in Kbytes
17:33:58.814 ->   -c <codec1>,<codec2>	Restrict codecs to those specified, otherwise load all available codecs; known codecs: flac,pcm,mp3,ogg (mad,mpg for specific mp3 codec)
17:33:58.814 ->   			Codecs reported to LMS in order listed, allowing codec priority refinement.
17:33:58.814 ->   -C <timeout>		Close output device when idle after timeout seconds, default is to keep it open while player is 'on'
17:33:58.814 ->   -d <log>=<level>	Set logging level, logs: all|slimproto|stream|decode|output, level: info|debug|sdebug
17:33:58.814 ->   -e <codec1>,<codec2>	Explicitly exclude native support of one or more codecs; known codecs: flac,pcm,mp3,ogg (mad,mpg for specific mp3 codec)
17:33:58.814 ->   -f <logfile>		Write debug to logfile
17:33:58.814 ->   -m <mac addr>		Set mac address, format: ab:cd:ef:12:34:56
17:33:58.814 ->   -M <modelname>	Set the squeezelite player model name sent to the server (default: SqueezeESP32)
17:33:58.814 ->   -n <name>		Set the player name
17:33:58.814 ->   -N <filename>		Store player name in filename to allow server defined name changes to be shared between servers (not supported with -n)
17:33:58.814 ->   -W			Read wave and aiff format from header, ignore server parameters
17:33:58.814 ->   -r <rates>[:<delay>]	Sample rates supported, allows output to be off when squeezelite is started; rates = <maxrate>|<minrate>-<maxrate>|<rate1>,<rate2>,<rate3>; delay = optional delay switching rates in ms
17:33:58.814 ->   -t 			License terms
17:33:58.814 ->   -? 			Display this help text
17:33:58.814 -> 
17:33:58.814 -> Build options: EMBEDDED LOOPBACK NO_FAAD LINKALL
17:33:58.814 -> 
17:33:58.814 -> abort() was called at PC 0x400d4b74 on core 1
17:33:58.814 -> 
17:33:58.814 -> ELF file SHA256: a57daf7349ccc36b
17:33:58.814 -> 
17:33:58.814 -> Backtrace: 0x40094e0f:0x3f801bf0 0x4009517d:0x3f801c10 0x400d4b74:0x3f801c30 0x400ec437:0x3f801c50 0x400da67c:0x3f801f70
17:33:58.814 -> 
17:33:58.814 -> [0;32mI (2869) esp_core_dump_uart: Press Enter to print core dump to UART...[0m
17:33:58.814 -> [0;32mI (2876) esp_core_dump_uart: Print core dump to uart...[0m
17:33:58.814 -> [0;32mI (2882) esp_core_dump_common: Found tasks: (14)![0m
17:33:58.814 -> ================= CORE DUMP START =================
17:33:58.814 -> MCQAAAEAAAALAAAAZAEAAA==
17:33:58.814 -> KNf7P6DV+z8g1/s/
17:33:58.814 -> oNX7P8DW+z/XA1EtwFH8P8TP+z8o1/s/uFH8PxkAAABg/30Ip/7yqijX+z8AAAAA
17:33:58.814 -> AAAAACTR+z9JRExFMQCClJdgt+MefIIAAQAAACDX+z8AAAAAIQAGAAAAAAAAAAAA
17:33:58.814 -> AAAAAAAAAAAAAAAAVPmAP7z5gD8k+oA/AAAAAAAAAAABAAAAAAAAAAAAAAAAAAAA

closed 2022-01-08 by dkotte