sle118/squeezelite-esp32 #127: LMS Plugin is able to kill LMS

dkotte, 2021-12-31:
After adding the SqueezeESP32 plugin, I got a lot of new options for Squeezelite-esp32 devices (as expected).
After configure different Options within "Player / ESP32 settings" the LMS goes to 100%, GUI is reachable no more and LMS stops playing for other devices.
I have to remove the plugin from file system, restart LMS-service and Squeezelite-esp32 is working again.
After readding the plugin, the error occur immediately after connecting player. Seems the old settings are not deleted with plugin. I try to reset the "player settings" at the global page .. without success.
I think it is a bug...?
Additional question: where does the plugin store the settings made for a specific player? Possible i can find the specific problem-configuration ..
Can't find something within LMS-Server.log
Environment:
Debian bullseye / Logitech Media Server Version: 8.1.1

philippe44, 2022-01-26:
Go to /var/lib/squeezeboxserver/prefs/plugin/squeezeESP32.prefs

ponpon1013, 2022-01-28:
I have the same problem. Additionnaly I have this messsage in log window of LMS:

Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Web::HTTP::processHTTP: Undefined subroutine &Plugins::SqueezeESP32::FirmwareHelper::initFirmwareDownload called at /var/lib/squeezeboxserver/cache/InstalledPlugins/Plugins/SqueezeESP32/PlayerSettings.pm line 134.
; fh=Slim::Web::HTTP::ClientConn=GLOB(0x559609a78038)

dkotte, 2022-01-28:
After deleting the squeezeESP32.prefs file i am able to restart LMS.
Seems the image position / size triggers the problem.

hannemann, 2022-07-20:
I'm also affected by these errors and cannot access the settings page. Nginx says Gateway Timeout when i try to configure my Player.

I see errors like these in the logs:
```
Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Web::HTTP::processHTTP: Undefined subroutine &Plugins::SqueezeESP32::FirmwareHelper::initFirmwareDownload called at /var/lib/squeezeboxserver/cache/InstalledPlugins/Plugins/SqueezeESP32/PlayerSettings.pm line 134.
; fh=Slim::Web::HTTP::ClientConn=GLOB(0x55fd223a48e0)
```
```
Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Web::HTTP::processHTTP: Can't locate object method "send_equalizer" via package "Slim::Player::SqueezePlay" at /var/lib/squeezeboxserver/cache/InstalledPlugins/Plugins/SqueezeESP32/Plugin.pm line 34.
; fh=Slim::Web::HTTP::ClientConn=GLOB(0x556c4e9f1868)
```
I tried to delete the prefs file and commented out the calls to these subs so the error messages disappeared but i still cannot open the settings page.

Server Version:
```
Logitech Media Server Version: 8.2.0 - 1627922070 @ Tue Aug 3 11:37:35 CEST 2021
```
Would be great to be able to have a display. How can i help to solve the issue?

hannemann, 2022-07-23:
I was able to get the Plugin working by:
- Commenting in Lines 41,42 in Player.pm
  - enable the following to test the firmware downloading code without a SqueezeliteESP32 player
- Commenting out Lines 33 - 35
  - call to send_equalizer
- the settings page did load
- applied settings to my liking
- now i have a prefs file with much more settings than before
- reverted all changes and after restart of lms the plugin works
- deleted the new preferences file and restarted lms, plugin works...
- the plugins is able to find a cached firmwarefile
- deleted the firmwarefile, deleted preferences file and restarted... plugin works

Confusing... so i uninstalled the plugin, deleted the prefs and firmware files and installed it again... same game as before. Comment in/out lines mentioned above, restarted lms and player... Plugin and display work 

### Prefs file before commenting in/out
```
---
_client:c8:c9:a3:d4:0a:80:
  _ts_equalizer: 1658563784
  _version: 0
  equalizer: ~
_version: 0
```
### and after
```
---
_client:c8:c9:a3:d4:0a:80:
  _ts_artwork: 1658566355
  _ts_equalizer: 1658565238
  _ts_height: 1658566997
  _ts_small_VU: 1658565238
  _ts_spectrum: 1658566355
  _ts_width: 1658566997
  _version: 2
  artwork:
    enable: 1
    x: 1
    y: 40
  equalizer:
  - 0
  - 0
  - 0
  - 0
  - 0
  - 0
  - 0
  - 0
  - 0
  - 0
  height: 320
  small_VU: 15
  spectrum:
    full:
      band: 8
    scale: '25'
    small:
      band: 5.33
      size: 25
  width: 240
_version: 0
```

philippe44, 2022-07-23:
I'll fix that when I'm back

hannemann, 2022-07-24:
i think i should also see an equalizer on the settings page which is not there.

philippe44, 2022-07-24:
It means that the player model is simply not recognized at all. There is a different model for squeezeesp32 devices which is sent when connecting to server

hannemann, 2022-07-25:
I should have read the readme more carefully. I have a muse proto with the 32 bit firmware version (installed via raspiaudio.github.io) which has no equalizer support.
Player information:
```
Player-Modell: SqueezeESP32
Gerätetyp: squeezeesp32
Firmware: v1.0-764-32
IP-Adresse des Players: 192.168.179.106
MAC-Adresse des Players: c8:c9:a3:d4:0a:80
Stärke des drahtlosen Signals: 55%
Spannung: 0.00
```

sle118, 2022-08-31:
Closing this since it looks like the question was answered

closed 2022-08-31 by sle118

hannemann, 2022-08-31:
> Closing this since it looks like the question was answered

IMO the question is not answered. If i have a fresh install of the plugin i cannot reach the settings page. I got it working only by temporarily modifying the code of the plugin.
https://github.com/sle118/squeezelite-esp32/issues/127#issuecomment-1193100759