sle118/squeezelite-esp32 #131: The Clipping Distortion In Max Volume

Skylakc, 2022-01-14:
Hello, There!
Yesterday I compiled a firmware by ESP-IDF v4.0.3. But I found that there will occur "Clipping Distortion" when I turning volume max(100%) at LMS Web control panel. The specific performance is broken sound and a great rustling sound.
I checked the input signal with an oscilloscope, the results show that the output signal from DAC produce severe distortion. The problem can be solved by adjusting the balance volume for time being (I chosed mono channel background music system).

I think the essence of this problem is that the control panel is adjusted to 100% when the DAC output amplitude too large.
Could we add a new function in program to set the percentage of maximum amplitude from DAC output amplitude?

My Hardware：
MCU Module: ESP32-WROVER ESP-A1S
DAC: AC101 (For A1s) ,ES8388 (For Normal ESP32 WROVER & ESP-A1s)

philippe44, 2022-01-26:
For 1/ I'm not sure I'm following. Do you use AC101 or ES8388? Are they both showing distortion? And yes, the ES8388 also has a volume set that, depending on your dac_controlset, can add gain. I don't understand either the connection with balance. Can you please describe more precisely what you are doing.
For 2/, it is already there. After X seconds (-C X on the squeezelite command line), the gpio that commands the amp amplifier is turned off (polarity is a parameter)

Skylakc, 2022-01-26:
Thanks for your reply! I like this project so much!
For 2, I found the function pin to control my AMP stand by pin, and I will share my PCBDoc(Altium or Cadence) in my repository (If it could work).  I used TDA7498 to AMP.
For 1, Both, the Clipping Distortion will appear in these two DAC chips. My intention to change the balance volume is decrease the DAC output volume (the output amplitude). Maybe this can be solved by reducing the output gain. I will continue to report my testing in this issue.

Skylakc, 2022-02-01:
After adding the dac_controlset parameter, the clipping distortion problem disappeared, but I also found that there is an option for audio gain in the LMS WEB UI settings.
[image]
When u config the NEW ESP32 Squeezelite, Please make sure the option is close.
[image]

closed 2022-02-01 by Skylakc