sle118/squeezelite-esp32 #133: DAC GPIOs not be shown after update

networxnet, 2022-01-16:
DAC GPIOs show <n> after update I2S 16-736 and higher, but still working.

[image]

sle118, 2022-01-23:
I am currently revisiting the UI and fixing bugs.  Could you please post your nvs configuration so I have something to test with?

networxnet, 2022-01-23:
`{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "WC",
  "a2dp_sink_name": "WC BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "WC",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "WC",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n WC",
  "bat_config": "",
  "bt_name": "WC",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "dac_config": "model=I2S,bck=25,ws=26,do=33",
  "dac_controlset": "",
  "display_config": "",
  "enable_airplay": "N",
  "enable_bt_sink": "J",
  "Equalizer": "-12,20,20,12,5,0,3,6,12,16",
  "eth_config": "",
  "gpio_exp_config": "",
  "host_name": "WC",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "lms_ctrls_raw": "n",
  "metadata_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "n",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "J",
  "wifi_smode": "A"
}`

networxnet, 2022-01-23:
```
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "WC",
  "a2dp_sink_name": "WC BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "WC",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "WC",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n WC",
  "bat_config": "",
  "bt_name": "WC",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "dac_config": "model=I2S,bck=25,ws=26,do=33",
  "dac_controlset": "",
  "display_config": "",
  "enable_airplay": "N",
  "enable_bt_sink": "J",
  "Equalizer": "-12,20,20,12,5,0,3,6,12,16",
  "eth_config": "",
  "gpio_exp_config": "",
  "host_name": "WC",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "lms_ctrls_raw": "n",
  "metadata_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "n",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "J",
  "wifi_smode": "A"
}
```

sle118, 2022-07-23:
@networxnet could you try the 4.3 branch and see if the issue is still present?

sle118, 2022-07-28:
I'll go ahead and close the issue for now.

closed 2022-07-28 by sle118