sle118/squeezelite-esp32 #139: Addition to documentation: easy way to start

drogfild, 2022-02-01:
I found really hard to find information how to easily start with squeezelite-esp32.
Here's my take how documentation could be improved. Please correct/add/modify. It might not be complete but could help many who find this project. I hope you find it useful.

Requirements:
- Linux or Windows Subsystem for Linux (WSL)
- Python (sudo apt install python)
- Unzip (sudo apt install unzip)

1) Download latest release as a zip file. Recommended to get 16bit I2S version for compatibility.
https://github.com/sle118/squeezelite-esp32/releases
For example:
wget https://github.com/sle118/squeezelite-esp32/releases/download/I2S-4MFlash.16.1012.master-cmake/squeezelite-esp32-master-cmake-I2S-4MFlash-16-1.1012.zip

2) Install Esptool
pip install esptool

3) Unzip package and go to build folder
unzip squeezelite-esp32-master-cmake-I2S-4MFlash-16-1.1012.zip
cd build

4) Plug in your board and find correct serial port
esptool.py read_mac

5) Run esptool to flash. You need to fill out correct serial port
esptool.py --port /dev/ttyS9 --baud 921600 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0xD000 ota_data_initial.bin 0x10000 recovery.bin

6) Finish setup:
- Open the wifi AP and enter configuration UI
- Select your wifi and enter credentials

7) Go to updates, select same build version and flash it

ykutovoy, 2022-03-09:
Ok looks like i did it :) 
problem was in saved commandline 1 
I couldn't understand from where does it get it, but it was saved corrupted one 
Working command for my device

squeezelite -o "BT -n 'Tempo'" -b 500:2000 -Z 44100 -r "44100-44100"

balloob, 2022-04-22:
My suggestion is to add a GitHub Pages to this repository and add [ESP Web Tools](https://esphome.github.io/esp-web-tools/). That way anyone can install the firmware using the Chrome/Edge browsers and don't have to install any tooling.

pwcutajar, 2022-05-16:
> Ok looks like i did it :) problem was in saved commandline 1 I couldn't understand from where does it get it, but it was saved corrupted one Working command for my device
> 
> squeezelite -o "BT -n 'Tempo'" -b 500:2000 -Z 44100 -r "44100-44100"

I also have this issue but do not understand your fix, could you explain more?

balloob, 2022-05-16:
I've added a link to an ESP Web Tools based installer in #144. Try installing with that.

pwcutajar, 2022-05-16:
> I've added a link to an ESP Web Tools based installer in #144. Try installing with that.

Smashing worked first time........but

Changing the output to bluetooth kills it, dead in the water

drogfild, 2022-05-17:
Thanks @balloob ! I think ESP Web Tools is the way to go for an easy start. 

Maybe my method can be saved as an intermediate level guide. As it allows additional tinkering with parameters before uploading.

Additionally I would love to see Imrov implemented in the project.

ykutovoy, 2022-05-17:
> > I've added a link to an ESP Web Tools based installer in #144. Try installing with that.
> 
> Smashing worked first time........but
> 
> Changing the output to bluetooth kills it, dead in the water

Yes You are right almost works :(  (strongly distorted sound)
I didn't manage how to decode stream for device, 
So now I'm stop experiments with BT and waiting i2s encoder will arrive to proceed

ykutovoy, 2022-05-26:
Finally I've got DAC Module UDA1334A 
With I2S everything works perfectly

sle118, 2022-07-28:
I'm planning to explore the web installer route suggested by @balloob so for now I'll go ahead and close this issue as a web installer will dramatically cut the complexity associated with getting up to speed

closed 2022-07-28 by sle118