sle118/squeezelite-esp32 #140: [Feature request] Volume scaling

iMartyn, 2022-02-15:
Love the project, and it's locked me into LMS a little more lol

I am using some of the annoying Ai-Thinker boards, and one of the not-so-documented problems with them is that once you start driving them hard (I like my bathroom music loud so I can hear it in the shower), some component internally spikes enough power to upset the whole board.

It'd be nice if I could set a scale factor of about 0.7 to make 100% volume 70% so I could avoid going into the red zone and crashing the board.

FWIW it only seems to happen with the speaker outputs rather than the headphones so my gut instinct is that the amps are drawing too much for the power regulation to handle.  I checked ampage usage and it is drawing about .2a with peaks upto .4a, but have seen it crash at .2a, so I don't think it's the USB limits, rather more likely a poorly selected 5v->3.3v regulator.

philippe44, 2022-02-24:
You probably can add a DAC_controlset to set one of the registers of the codec of that board to set such back-off

sle118, 2022-07-23:
@iMartyn was your question answered?

iMartyn, 2022-07-26:
not really, the answer given is a bit vague for me to attempt to integrate into my devices and only a "probably".  The device is poorly documented so this is not entirely the project's fault, but the answer also wasn't a yes or no to the feature request either.

philippe44, 2022-07-26:
The ai-tinker's DAC has some register to control gain/max volume I think. You should look at it and then just add the register write in dac_controlset

philippe44, 2023-04-02:
I think we should close this one unless the OP chimes in

sle118, 2023-04-02:
I'll close. OP can reopen if appropriate.

closed 2023-04-02 by sle118