sle118/squeezelite-esp32 #144: Web installer for Squeezelite ESP32

balloob, 2022-04-30:
Using [ESP Web Tools](https://esphome.github.io/esp-web-tools/) it's possible to offer a web installer for Squeezelite ESP32. This will make starting with Squeezelite ESP32 very easy. The web installer requires using either Chrome or Edge.

I've created a demo here: https://paulusschoutsen.nl/squeezelite-esp32-install/. It will install v1.1023. The source for the installer can be found here: https://github.com/balloob/squeezelite-esp32-install.

I've tested it locally with my Muse box and was able to connect it to Wi-Fi and control it using Home Assistant.

GitHub can be used to host websites, and GitHub Actions can be used to automatically update files. That way you could offer a website that always offers the latest version. Tasmota does something like that: https://github.com/tasmota/install

CC @RASPIAUDIO @kheperV3

RASPIAUDIO, 2022-04-30:
Great work👍 And thanks as  is something that was on our task list for next week ☺️
I think this is will contribute a lot to make Squeezelite more accessible for everyone.

balloob, 2022-05-02:
Great. Hope you can adopt it and make it available for ESP Muse Box users. It will make installing Squeezelite on the ESP Muse Box very accessible.

One step that would make it _even more_ accessible is adopting [Improv Wifi](https://www.improv-wifi.com/). It's a free + open standard to configure Wi-Fi. It's implemented on the device as a very small serial protocol (5 msgs). ESP Web Tools can detect it and offer users to configure Wi-Fi while installing Squeezelite.

sle118, 2022-07-04:
I will definitely have a look into this at a later point, as it sure can simplify adoption. I severely lack bandwidth for this right now, though, so it will have to wait for later this year.

Thank you for the excellent suggestion

balloob, 2022-07-05:
@sle118 I created an example repo that you can copy 1:1: https://github.com/balloob/squeezelite-esp32-install

RASPIAUDIO, 2022-07-05:
We are also trying to work on this Improv subject on our side for Squeezelite, we will share as soon as we have something.

sle118, 2022-07-28:
@balloob I'll get to it later in the summer. I really appreciate the efforts you put into that, so thank you. 

I will likely include some steps in the make file and automated build to keep an alignment with our different targets and I think this will lower the complexity for anyone wanting to get started.

sle118, 2022-07-28:
@RASPIAUDIO if you intend to try the improv implementation, please do it in branch 4.3 as the network manager was fully rewritten to support Ethernet. Integration should be easier as well there, as the network manager is now built around a week structured state machine that clearly defines states and transitions.

sle118, 2022-07-29:
@balloob FYI I'm a huge fan of home assistant, which I use here at home. If there's any way to better integrate this project into home assistant ecosystem, let me know. I understand how improv and a web installer can lower the adoption complexity threshold and I wish more users would flock to the LMS platform.

By the way, I had to ditch the LMS add-on in home assistant because it was eating all of my LMS server TCP ports. :D

balloob, 2022-07-29:
Great to hear! We recently added the [`slimproto`](https://www.home-assistant.io/integrations/slimproto) integration to directly control Squeezelite-ESP32 devices. The community started [Music Assistant](https://github.com/music-assistant/hass-music-assistant) as an LMS-alternative powered by Home Assistant media players. So yeah, this project + the hardware from Raspiaudio fits in perfectly to build a great local music stack 🎉

sle118, 2022-07-30:
If there's anything outside of slimproto that could be implemented and that would help with the integration, aside of what is being discussed here, let me know.

sle118, 2022-09-09:
I've started working on this. Cracking open the actions workflow ended up exposing some tools I was using which are no longer maintained, so I'll take the opportunity to revamp the process a bit.

RASPIAUDIO, 2022-09-13:
Hi all!
We (Raspiaudio) are trying to implement the wifi credential setup from a Chrome browser using the Web Serial API to have an easier start for new Squeezelite's users.

We would like to do it using the existing serial command set of Squeezelite, rather than adding the Improv.

But for now it does not work, I have tried these leads:

- Initialize the credentials directly (via Web Serial ==> "join <ssid> <password>)
BUT WiFiManager does not take this into account, as it is not saved
 
- Disable WifiManager as expected (via Web Serial ==> "nvs_set bypass_wm str -v 1")
BUT the replacement sequence remains to be written! And we lack information

Any clue of why this is happening?

Thanks!
Louis

balloob, 2022-09-13:
@RASPIAUDIO can you please open a new issue for this. Note that not using Improv means it will not be compatible with ESP Web Tools.

sle118, 2022-10-07:
@RASPIAUDIO  Configuring the wifi through the serial interface has long been out of my priorities and I'm afraid that I might have broken it along the way. 

I'm currently focusing on implementing a full blown web installer that will be connected directly to our build system and which will automatically be updated on every new release (thanks to @balloob for the suggestion). You will see lots of repetitive commits in the upcoming days because running in the github actions environment is somewhat different from running locally.   However, I'm happy to report that the whole build chain is almost completed by now, with only one last issue to address... pushing the commit to the web installer branch after a release build.

`2022-10-07 02:24:41,358 Changes found. Preparing commit
2022-10-07 02:24:43,133 Opening repository from /__w/squeezelite-esp32/squeezelite-esp32
2022-10-07 02:24:43,133 Last commit for master-v4.3 is 2a22284 WIP workflow [skip actions] (Thu Oct 6 16:01:59 2022-0400) <Sebastien L>
2022-10-07 02:24:43,134 Pushing commit 0d46fd5 Web installer for None-None-None-NoneNone (Thu Oct 6 16:01:59 2022-0400) <Sebastien L> to url https://github.com/sle118/squeezelite-esp32-installer.git
Traceback (most recent call last):
  File "/usr/sbin/build_tools.py", line 801, in <module>
    main()
  File "/usr/sbin/build_tools.py", line 795, in main
    func(args)
  File "/usr/sbin/build_tools.py", line 762, in handle_pushinstaller
    push_if_change(repo,args.token,args.cwd)
  File "/usr/sbin/build_tools.py", line 729, in push_if_change
    remote.push([reference],callbacks= RemoteCallbacks(pygit2.UserPass(auth_method, token)))
  File "/opt/esp/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/pygit2/remote.py", line 257, in push
    payload.check_error(err)
  File "/opt/esp/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/pygit2/callbacks.py", line 93, in check_error
    check_error(error_code)
  File "/opt/esp/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/pygit2/errors.py", line 65, in check_error
    raise GitError(message)
_pygit2.GitError: unexpected http status code: 403
Error: Process completed with exit code 1.`

Once this is done' I'll focus on implementing Improv for network configuration, so I suggest you wait for that to happen before you spend too much time on a different solution.

sle118, 2022-10-07:
Didn't mean to clear this. Reopening it

sle118, 2022-10-07:
@balloob After a long detour where I came up with a new build system, I finally have a rough web installer up and running
https://sle118.github.io/squeezelite-esp32-installer/

RASPIAUDIO, 2022-10-08:
@sle118 Great work! I've tested the web installer with the Muse and it is working like a charm.

We share your idea and Balloob that it makes more sense to use Improv as a de facto standard for provisioning of wifi , rather than the route that we first took using the serial command interface of Squeezelite.
So we were also these last days humbly starting to work integrating Improv for Squeezelite. We don't have your expertise in Squeezelite, but we would be happy to offer our help if you wish.

sle118, 2022-10-08:
@RASPIAUDIO I have completely re-written the network manager using a simple state machine in version 4.3, as it would have been very difficult to integrate support for Ethernet while coexisting with the previous Wifi-Manager, for which the event management had grown hard to maintain. 

If your attempt was done on master-v4.3, then you could send a pull request for me to review and possibly start from it. Otherwise, I'll skip the offer and bite the bullet myself in the upcoming days.

thank you!

sle118, 2022-10-08:
Since a rough installer is now available, I'll go ahead and close this issue.

closed 2022-10-08 by sle118