sle118/squeezelite-esp32 #147: Build error

FightMan01, 2022-05-15:
```
[ 88%] Linking CXX static library libbt.a
[ 88%] Built target __idf_bt
make: *** [all] Error 2
make failed with exit code 2
```

esp-idf version: 4.0

FightMan01, 2022-05-15:
```
/Users/nemetharon/Documents/squeezelite-esp32/components/services/i2s.c: In function 'i2s_param_config':
/Users/nemetharon/Documents/squeezelite-esp32/components/services/i2s.c:883:9: error: implicit declaration of function 'adc_power_always_on'; did you mean 'adc_power_release'? [-Werror=implicit-function-declaration]
         adc_power_always_on();
         ^~~~~~~~~~~~~~~~~~~
         adc_power_release
```

sle118, 2022-05-16:
Please provide more details. The code will build on a very specific version of the ESP idf. Also, which branch are you trying to build?

Note that the easiest way to compile yourself is to use Docker. For example with the 4.3 branch:
docker run --rm -p 5000:5000/tcp -v %cd%:/project -w /project -it sle118/squeezelite-esp32-idfv4-master

sle118, 2022-06-01:
closing because of no activity

closed 2022-06-01 by sle118

kine90, 2022-10-04:
@sle118 can you check this command? I´ve tried the procedure but it gives me errors....

> Please provide more details. The code will build on a very specific version of the ESP idf. Also, which branch are you trying to build?
>
> Note that the easiest way to compile yourself is to use [...]

```
C:\Users\kine9\squeezelite-esp32>docker run --rm -p 5000:5000/tcp -v %cd%:/project -w /project -it sle118/squeezelite-esp32-idfv4-master
Unable to find image 'sle118/squeezelite-esp32-idfv4-master:latest' locally
latest: Pulling from sle118/squeezelite-esp32-idfv4-master
08c01a0ec47e: Pull complete
260eee8bf99e: Pull complete
cce429d594ef: Pull complete
b6a5643f3c16: Pull complete
2e0be29b33e9: Pull complete
2912da10312b: Pull complete
ade1466089da: Pull complete
7a92ed1084bb: Pull complete
Digest: sha256:c1002f3ef84d3006b7f6eac7724588c5652ec2d1f65869f12d020ddfa4840aab
Status: Downloaded newer image for sle118/squeezelite-esp32-idfv4-master:latest
Detecting the Python interpreter
Checking "python" ...
Python 3.8.10
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /opt/esp/python_env/idf4.3_py3.8_env/bin/python
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /opt/esp/tools/cmake/3.16.4/bin
  /opt/esp/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin
  /opt/esp/python_env/idf4.3_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

npm version is 8.5.2
node version is v16.14.0
To build the web application, run:
pushd components/wifi-manager/webapp/ && npm rebuild node-sass && npm run-script build && popd

To run size statistics, run:
puncover/runner.py --gcc_tools_base /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf- --elf ./build/recovery.elf --build_dir build --src_root .

root@03ba21d857ba:/project# idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /project"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /opt/esp/idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- Building RECOVERY
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
[... 43 lines cut ...]
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /project/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /opt/esp/idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /opt/esp/idf/components/esp32/ld/esp32.peripherals.ld
******************* local.500.cmake-master
******************* local.500.cmake-master
-- Components: app_recovery app_squeezelite app_trace app_update asio audio bootloader bootloader_support bt cbor cmock coap codecs console cxx display driver driver_bt efuse esp-dsp esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon platform_config platform_console protobuf-c protocomm pthread raop sdmmc services soc spi_flash spiffs squeezelite squeezelite-ota targets tcp_transport tcpip_adapter telnet tinyusb tjpgd tools ulp unity vfs wear_levelling wifi-manager wifi_provisioning wpa_supplicant xtensa
-- Component paths: /project/components/platform_console/app_recovery /project/components/platform_console/app_squeezelite /opt/esp/idf/components/app_trace /opt/esp/idf/components/app_update /opt/esp/idf/components/asio /project/components/audio /opt/esp/idf/components/bootloader /opt/esp/idf/components/bootloader_support /opt/esp/idf/components/bt /opt/esp/idf/components/cbor /opt/esp/idf/components/cmock /opt/esp/idf/components/coap /project/components/codecs /opt/esp/idf/components/console /opt/esp [... cut ...]
-- Adding new build target (from build folder): ninja squeezelite-flash
CMake Error at squeezelite.cmake:7 (esptool_py_flash_project_args):
  Unknown CMake command "esptool_py_flash_project_args".
Call Stack (most recent call first):
  squeezelite.cmake:124 (___register_flash)
  CMakeLists.txt:11 (include)

-- Configuring incomplete, errors occurred!
See also "/project/build/CMakeFiles/CMakeOutput.log".
See also "/project/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
root@03ba21d857ba:/project#
```