sle118/squeezelite-esp32 #149: LMS remote

ykutovoy, 2022-05-26:
Hello!
Could You provide a build without audio codecs?
I would like to write this build to ESP32 whithout memory needed for codecs

And use this device as an remote for LMS with Screen, encoder, buttons (may be NFC reader) (some kind of replica for [this device](https://www.amazon.com/Logitech-Squeezebox-Controller-Discontinued-Manufacturer/dp/B001413GT6))

I understand that it is possible to get another WROWER but I've already have some WROOM modules and would like to give them additional life :)

sle118, 2022-05-26:
We currently have very little bandwidth to take new features requests, and even if "removing stuff" may look simple to achieve, the reality is very different. The whole architecture is built around having the audio buffers and removing them would have consequences everywhere. Plus, audio buffers are just the tip of the iceberg when looking at memory usage. For example, attempting to run even the recovery alone on a wroom (no audio there) is more than likely going to fail. Not enough memory, and the fact that we explicitly allocate several chunks of memory in the PSRAM area.

closed 2022-05-26 by sle118

ykutovoy, 2022-05-27:
Thank You for Your reply.
Will be glad to help You if I can :) (I'm not a programmer but could provide any user experience if needed)