sle118/squeezelite-esp32 #151: MCLK Pin Option

AxelJust, 2022-06-22:
Thanks For your great Work!
By now I had great sucses flashing all kind of ESP32 Audio boards and got enouf motivation to start my own Desing.
One Amp based on the ADAU1701 (JAB5) would be Ideal for one of my Projects. Unfortunatly it isn't posible to use it without MCLK.

An option to set a Pin as MCLK output would be all we need for that.

wizmo2, 2022-06-22:
I think this is availble now

"The option "mck" is used for some codecs that require a master clock (although they should not). Only GPIO0 can be used as MCLK and be aware"

AxelJust, 2022-06-22:
With a TTGO T9 on hand I'm on `fw: I2S-4MFlash.16.1023.master-cmak, mode: Squeezelite-ESP32 `
I coudn't find "mck" anywehre. not in the UI or repository.
Would you simply put `model=I2S,bck=33,ws=25,do=32,mck=0` in dac_config?
Can you tell us what you are quoting @wizmo2 ?

AxelJust, 2022-06-22:
oh, sorry. Google found that frase in the README.md . wierd that github coudn't find itself.
I'l try the dac_config from above and giv an report.

AxelJust, 2022-06-22:
I folowed your documentation before to setup my T9 as well, thank you!
in That case I just made a tipo.
By now, pin 0 works to avoid horible noise the ADAU1701 throws out when confused.
But Audio isn't puut out either.
Now I need to find out if SigmaStudio is needet to change something in the DSP.
Or dose sombody els have an idea?
Greatings

AxelJust, 2022-06-26:
### GOOD NEWS
After resting my mind for some days and corecting a falls Pin Conection I made, I Got it to work this time:

**Parts:**

- AMP with DSP  - [WONDOM JAB5](https://store.sure-electronics.com/product/AA-JA33286)
- ESP32 Board     - [TTGO T8](https://de.aliexpress.com/item/32877829189.html?spm=a2g0o.productlist.0.0.7e4a4282xNyzEe&algo_pvid=8df66882-502f-401a-8d8d-db15a7cb3230&algo_exp_id=8df66882-502f-401a-8d8d-db15a7cb3230-8&pdp_ext_f=%7B%22sku_id%22%3A%2212000022835912435%22%7D&pdp_npi=2%40dis%21EUR%21%217.61%217.61%21%21%21%21%402100bdde16562785159217422e2be8%2112000022835912435%21sea)
- Test Speaker     - [Visaton EX 80 S
](https://www.thomann.de/de/visaton_ex_80_s.htm)

**Pinout:**

JAB5________ESP32
1-LRCLK____25
2-BCLK_____33
3-DATA1___32
4-GND_____GND
5-+5V______3V3 or 5V ( JAB5 Pin actualy is at 3,5V; VIN Pin on T8 not posible )
6-MCLK____0

dac_config: `model=I2S,bck=33,ws=25,do=32,mck=0`

closed 2022-06-26 by AxelJust

hgoldwire, 2025-09-04:
I tried the same thing using this [ESP32-S3](https://www.amazon.com/dp/B0DB1WK3CW)
and
[Dayton Audio KABD-4100](https://www.parts-express.com/pedocs/manuals/325-434--dayton-audio-kabd-4100-user-manual-v1.pdf?srsltid=AfmBOoo8hhkfqa_EC1eLL4KX4g2XDrKW6tbP8IEOB_ezmHDTDxbxz1w3) which is a rebranded JAB5

I used a homemade JST-PH 6-pin connector to individual duponts. Really needed to pay attention to the amp's pinout diagram and the ESPs pinout.  Got it working as well  ( YES !!!!!!!!! )-- kindof.

I get great, clear sound via airplay or cspot as long as the volume is low to moderate.  If I turn volume up pretty loud, I get glitchy machine-gun pops/clicks.

This with a homemade 32-bit build of `master-v4.3` using docker image `sle118/squeezelite-esp32-idfv446` 
and this NVS
```
dac_config: model=I2S,bck=11,ws=9,do=10,mck=1
autoexec1: squeezelite -o I2S -b 500:2000 -d all=info -C 30 -n i2s-test -a 24  -W -r 48000
```

ChatGPT tells me squeezlite is maybe putting out 32-bit samples, or maybe wrongly-formatted samples, and it only matters when volume is high enough for < hand wavey > significant bits things </ hand wavey>

Can anyone shed some light on this?