sle118/squeezelite-esp32 #152: Background noise on ESP32 A1S with AC101 dac

just-jason, 2022-06-25:
I have noticed a issue with background noise with an ESP32 A1S dev board using the AC101 dac. It is noticeable on standby as a "crackle"that get progressively worse the longer the board stays in standby. By standby, I mean powered on , but not playing any tracks. It is also observable on reboot where there is a classic pop as you might expect if no mute was active. 

For the sake of tests , I tried out [this](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/ESP32-A1S/ESP32_A1S.ino) example and I found that there was no background noise nor startup "pop"

I am not sure what the exact cause of this is , but I found some info that may be a starting point [here](https://github.com/schreibfaul1/AC101/issues/3)

The readout from the registers on the board ( from the "Advanced config" tab) are as follows at the moment:
```
00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 0123456789abcdef
00: 01 01 ff ff 80 0c ff ff 00 00 ff ff 00 00 ff ff ??..??..........
10: 88 50 ff ff a0 a0 ff ff 00 00 ff ff 00 00 ff ff ?P..??..........
20: 00 00 ff ff a0 a0 ff ff 00 00 ff ff 00 00 ff ff ....??..........
30: 00 00 ff ff 00 00 ff ff 00 00 ff ff 00 00 ff ff ................
40: 00 00 ff ff 00 00 ff ff 80 00 ff ff 88 00 ff ff ........?...?...
50: 33 00 ff ff 05 0a ff ff 00 1f ff ff d1 09 ff ff 3...??...?..??..
60: 00 00 ff ff 00 00 ff ff 00 00 ff ff 00 00 ff ff ................
70: 00 00 ff ff 00 00 ff ff 00 00 ff ff 00 00 ff ff ................
```

sle118, 2022-06-27:
Thank you for reporting this; this board/chip is fairly popular, given that it's both cheap and easy to procure.

I don't have a lot of bandwidth right now for this project, but I'll prioritize when I can.

If Philippe surfaces, then perhaps he'll also get a chance to take a look.

AxelJust, 2022-07-01:
As I've got the same bord on hand, I'm Happy to try both senarios as well to report if I get the same result start of next week, when I'm back home.

AxelJust, 2022-07-10:
[image]
It dose no ocure here.
Do you have some more details if and what speakers you use or is it the headphones out?
Where is the power from.
I highly discourage to use Battery as the connection goes up to 4.5v when connected to usb.
With the new update I recommend you backup your Config and update your firmware.

sle118, 2022-08-31:
@philippe44 this is something you'll be more familiar with!

sle118, 2023-04-01:
@just-jason I'm revisiting older issues and cleaning up. Is this a problem for you or did you find a solution?

just-jason, 2023-04-01:
Yes, the issue still remains. If there is any more info needed , let me know.

ei23fxg, 2023-05-08:
I had the same issue when amp and esp32 are on the same power source. 
A ground loop isolator or 5V/5V DC/DC converter did the trick for me.

just-jason, 2023-05-08:
> I had the same issue when amp and esp32 are on the same power source. A ground loop isolator or 5V/5V DC/DC converter did the trick for me.

Did you have noise all the time , or only starting several hours after standing in standby

ei23fxg, 2023-05-08:
> > I had the same issue when amp and esp32 are on the same power source. A ground loop isolator or 5V/5V DC/DC converter did the trick for me.
> 
> Did you have noise all the time , or only starting several hours after standing in standby

On boot / power on there was crackle and when in standby.
When playing i could not hear it.
I did not test if there is crackle noise after several hours standby, since my speakers are battery powered.

just-jason, 2023-05-09:
The noise is not present on boot, its only stars after the esp sits for a few hours on standby/ pause. These seems to to be different issues

philippe44, 2023-08-27:
That board is usually a pain, so I'll close this one

closed 2023-08-27 by philippe44