sle118/squeezelite-esp32 #154: Adafruit I2S Stereo Decoder - UDA1334A

ykutovoy, 2022-06-27:
Hello guys! 
I'm using ESP32-WROVER + I2S DAC UDA1334A
everything works fine, sound perfect
but I'm faced with issue with popping/cracking not often but very annoying 
On the "adafruit" site I've found tweak for same issue 

> The workaround is to use a software mixer to output a fixed sample rate to the I2S device so the bit clock does not change.
[Original page](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/pi-i2s-tweaks)

Is it possible to set fixed sample rate in my setup?
Thank You for help.

philippe44, 2022-06-27:
There is a command line of squeezelite to set sample rate, it's honored by this embedded version. Don't know it by heart, it'd -R or -u -r, look at the doc. Only for 26 bits version

sle118, 2022-07-01:
I'll go ahead and close this one.  If anything, feel free to open a discussion or to meet in the logitech forums.

closed 2022-07-01 by sle118

ykutovoy, 2022-07-01:
Thank You for help!
I have no more questions on this topic
Only may be to add linc to documettation?

> Squeezelite v1.9.x, Copyright 2012-2015 Adrian Smith, 2015-2021 Ralph Irving.
> See the squeezelite manpage for usage details.
> https://ralph-irving.github.io/squeezelite.html