sle118/squeezelite-esp32 #156: Feature request: Multiple I2S Outputs

AxelJust, 2022-07-01:
Inspierd by [Multisqueeze](https://www.max2play.com/en/portfolio/plugin-multisqueeze/) I Was wondering if it could be posible to use tow I2S outputs to adress diferent DACs.
Next step would be to show up as tow diferent devises within Airplay or Squeezelite. 
I'm aware it would be Propably simpler to put tow esp's next to each other...

sle118, 2022-07-02:
What's the end goal? Playing the same stream over 2 devices, or different streams?

We are already pushing the esp32 to its limits, so having multiple streams wouldn't be possible.

AxelJust, 2022-07-02:
What would have the moast use cases would be to choose diferent outputs, not tu supy diferent streams.
To chose output A, B or A+B
That would alow for more flexible board desings, or expansion of curent ones to adress diferent rooms if analog instalations are already present.
I would find an echo dot like setup intresting. Muse Proto in litle Speaker with conetion to HiFi Sterio.

sle118, 2022-07-02:
I will think about that. If shouldn't be too difficult to store multiple configurations and toggle between them then reboot

sle118, 2022-07-04:
The more I think about it, the bigger this is becoming. If the idea is to expose two player names, and toggle between DAC outputs when playing a different zone, it would be quite challenging as it would require some server side logic to be implemented in the plugin.

Given this is an edge case, with cheap hardware options available (this is the point of this project), and given the little time we can allocate to this project for new features, I'll go ahead and close that one.

Maybe a raspberry Pi would be more appropriate for this.

Thank you!

closed 2022-07-04 by sle118