sle118/squeezelite-esp32 #159: Feature Request | BAcklight invert toggle

docgalaxyblock, 2022-07-09:
I successfully wired up today a ESP32 and a [Pimoroni Pirate Audio HAT (DAC + AMP)](https://shop.pimoroni.com/products/pirate-audio-3w-stereo-amp)
This ST7789 based LCDs backlight gets disabled by a high signal and so please add a backlight pin invert checkbox

Greetings from 🇩🇪 

[image]

AxelJust, 2022-07-09:
Very Cool.
I'd like to use the same display as well. Would you mind to share the config file?
Greatings from Erlangen

docgalaxyblock, 2022-07-09:
@AxelJust No idea what you found but that is not my work 🙈 Link?
I could share the conf later, will update the initial post

docgalaxyblock, 2022-07-09:
Thanks, no I did not followed it
Had the same problem as he regarding the 3.3V - if I found this earlier I would not had waste yesterday debugging the hat with a Pi and esp...

docgalaxyblock, 2022-07-09:
@AxelJust Are you at the lms/squeeze forum active?
Could you share your profile there pls, then wedo not mess up this issue
And if you order a pcb i may be interested ^^

AxelJust, 2022-07-09:
I'm not very active there. Coldn't find you either @docgalaxyblock . Send me your phone number at axel@axeljust.eu and we can have a chat.
I took a look at the PCB. to me it is missing some features like Battery Carging and Protection. In your case  you propably only want to step your battery power down, use the display and use AUX on the Boombox.
Greatings

sle118, 2022-07-23:
Philippe and myself have little time right now allocated on the project. Anyone feel free to submit a pull request and I'll do my best to review it. I'll keep the issue open so it stays on the radar. Maybe we'll get to this later this year

sle118, 2022-07-28:
@docgalaxyblock have you been able to resolve your problem? 
It is not clear from the discussion. If so, I'll close the issue.

docgalaxyblock, 2022-07-28:
Let me check it again
It may was a wirering issue...

docgalaxyblock, 2022-07-28:
@sle118 Year it seems that it was a wirering/user issue ...

Should this issue stay open as a feature request for displays that have a inverted backlight?

sle118, 2022-08-02:
I think I'll go ahead and close it for now as we don't have a lot of bandwidth right now to implement new features. There has not been requests from other users as well in the past.

closed 2022-08-02 by sle118