sle118/squeezelite-esp32 #160: How tight is the time sync

StuartIanNaylor, 2022-07-09:
On multiple devices whats the time sync tolerance of squeezelite-esp32?

Or should I ask does it?

sle118, 2022-07-12:
Not sure I understand what you mean by tolerance?

The sync mechanism is required when some devices process samples/play slower than others. In an ideal world, all clocks would have the same frequency and samples would be fed at the same rate to the DAC. This is not the case, however, even for devices from the "same batch of components". Each device has a clock that has a small drift that cause samples to be consumed at more or less the same rate, with variations that aren't noticeable. 

In short, each device sends a status message back to the server that includes the current sample position and a time stamp. The server then sends "skip X samples" back to players that are lagging beyond a certain threshold.  There's a bit more to it than this, but it should help you get the point.

This means devices from Logitech should play along fine, synchronized with ESP32 based devices. Someone from Logitech said that in the early days, they tested successfully with "a large number of players piled on top of each other".  

You should pay attention to the network bandwidth, though, as it has been the bottleneck for several users. This is important if you either live in a larger home where wifi signal needs to travel through walls, as signal strength might become an issue. It is also important if you live in a building with lots of access points and users since your devices will be competing with others for a given wifi channel bandwidth.

Let me know if this helps

Joshfindit, 2022-08-02:
@StuartIanNaylor If I understand correctly, the core of your question is:

When playing multi-room audio, what's the maximum that one device's output could be out of sync with the others: <1ms, 10ms-30ms, >30ms?

Is that right?

StuartIanNaylor, 2022-08-22:
Yeah is there any control where on different carriers such as wifi, ethernet and infrastructure routing, is there anything that resyncs on NTP or other methods?
And what is the tolerance of the resync <1ms, 10ms-30ms, >30ms?

philippe44, 2022-08-24:
There is no use of NTP or such with, LMS whose target is 10 ms for resync which as you know gives you 3m which is fine with me for multiroom because, by definition you're likely to be more than 3m away from ideal position. When using AirPlay, it's using NTP-type sync timestamps.

sle118, 2022-09-09:
I guess I'll go ahead and close the issue for now. If anything, please use a discussion instead