sle118/squeezelite-esp32 #161: AC101 only 16bit

AxelJust, 2022-07-11:
ESP32-A1S AC101 only compatible to 16bit (tested on v2.2 2748 / fw: I2S-4MFlash.32.1036.master-cmak, mode: Squeezelite-ESP32)
ESP32-A1S ES8388 compatible with both.
AMP Gain with AC101 sems to be higher. or might that be a Settings diference?
that might be to mention in the readme

philippe44, 2022-08-24:
What do you mean by "only compatible" ?

AxelJust, 2022-08-27:
I tried Both and only one worked

philippe44, 2022-08-27:
I'll have a look. As @sle118 days, I don't recommend 32 bits version anyway, but there is probably a setting for word length that I did not add in AC101, or word length is only up to 16 or 24 bits

philippe44, 2022-08-28:
Well, unfortunately it seems that the issue is with the 24 bits driver of the esp32. Not 100% sure but it seems to be that data is not send correctly over the I2S line when bits_per_sample is 24 which is needed for ac101 as it cannot handle 32 bits samples

AxelJust, 2022-08-28:
A note in the docs would be helpful for others 👍

closed 2022-08-28 by AxelJust

philippe44, 2022-08-30:
Found a ways to make it work - fixed now