sle118/squeezelite-esp32 #166: Cannot get BT configured on squeezelite

suldobrasil70, 2022-08-28:
Kindly asking the community for help. I have an ESP-32-WROVER-E, installed the fw with the web installer and upgraded to Muse.16.1036. I'd like to have another squeezelite player at home, and I want to stream this one to a BT speaker.
Based on the instructions I have edited the squeezelite command line to:
**squeezelite -o "BT -n 'B13'" -b 500:2000 -d all=info -C 30 -W -n esp-32-wrover -R -u m -Z 192000 -r "44100-44100"**
...where B13 is my BT speaker.
I can sync this player with my other player, but it seems to disturb the LMS server, every 10 seconds or so the sound interrupts, keeps coming back after a few secs and then again.
For now, I have set the output to I2S, and all works fine:
**squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n esp-32-wrover -R -u m -Z 192000 -r "44100-44100"**
Any idea what I am doing wrong? Most only radio channels I listen to come at 64kps, in case that is of any importance.
thanks in advance

philippe44, 2022-08-30:
Why using Muse build?

suldobrasil70, 2022-08-30:
I don't know, just picked it. Any better suggestions?

philippe44, 2022-08-30:
Use the generic i2s version. All others are platform specific and have pre-configured I/O. Please go through the whole README for explanationS

sle118, 2022-09-01:
@suldobrasil70 output to Bluetooth was never the long term goal of this project, as it is very demanding for the platform for several reasons. We're pushing the limits of the hardware, including the radio bandwidth, which means your problem may very well be cause by something in your environment (did someone use the micro wave?)

I advise steering away from Bluetooth as target output except for experimenting.

closed 2022-09-01 by sle118

daduke, 2022-09-01:
that said, with decent wifi and the BT speaker nearby I run several WROVERS in BT output mode 24/7 with no problems at all..

philippe44, 2022-09-01:
@sle118 says, BT remains experimental. Now, can you provide the log ?