sle118/squeezelite-esp32 #168: Wrover with LAN8720 no ip address

pwcutajar, 2022-09-20:
Hi 
I've setup a wrover with fw: I2S-4MFlash.32.1180.master-cmak, mode: Squeezelite-ESP32

and have wired up a PCM5102 DAC (working)

I've also added a Microchip LAN8720 with a RMII interface as per the instruction (adding 3.3v and ground). 

The Lan interface flashes and my switch reports a 100mbps connection but no ip is assigned to the interface.

I an see my WIFI config in the gui but no sign of a any lan config other than in the NVS editor.

Any help would be most welcome

cheers

sle118, 2022-09-20:
Ethernet is only supported in the master-v4.3 branch. 

Please flash that version and try again.

pwcutajar, 2022-09-20:
HI

still no joy with fw: I2S-4MFlash.32.1188.master-v4.3, mode: Squeezelite-ESP32

pwcutajar, 2022-09-20:
also wifi will now not release / disconnect

pwcutajar, 2022-09-20:
Status shows the following 

Configuring Ethernet failed: ESP_FAIL

rst | 32 | Configuration
mdio | 34 | Configuration
tx_en | 21 | Fixed
tx0 | 19 | Fixed
tx1 | 22 | Fixed
rx0 | 25 | Fixed
rx1 | 26 | Fixed
crs_dv | 27 | Fixed

sle118, 2022-09-21:
As with anything, please provide some meaningful logs

pwcutajar, 2022-09-21:
I cant seem to find a log file? I tried to telnet but the connection is refused. even though telnet only is set.

pwcutajar, 2022-09-21:
[image]

pwcutajar, 2022-09-21:
[image]

pwcutajar, 2022-09-21:
[image]

pwcutajar, 2022-09-21:
[image]

pwcutajar, 2022-09-21:
[image]

sle118, 2022-09-22:
Please go ahead and grab the logs using either a serial adapter or a USB connection, depending on your board

pwcutajar, 2022-09-22:
[putty.log](https://github.com/sle118/squeezelite-esp32/files/9624062/putty.log)

pwcutajar, 2022-09-22:
This might be better
[putty.log](https://github.com/sle118/squeezelite-esp32/files/9625306/putty.log)

sle118, 2022-09-24:
Two of the GPIO you chose cannot be used as output.

network_ethernet: Ethernet config => model: lan8720, valid: YES, type: RMII, **mdc:35, mdio:34**, rst:32, intr:-1, cs:0, speed:0, host:1
(3167) gpio: io_num=35 **can only be input**
(3167) gpio: io_num=34 **can only be input**

sle118, 2022-09-24:
you can have a look at espressif's documentation
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html

sle118, 2022-09-26:
I think this was answered. Cleaning up the issues back log right now, so closing that one

closed 2022-09-26 by sle118

pwcutajar, 2022-09-26:
> Two of the GPIO you chose cannot be used as output.
> 
> network_ethernet: Ethernet config => model: lan8720, valid: YES, type: RMII, **mdc:35, mdio:34**, rst:32, intr:-1, cs:0, speed:0, host:1 (3167) gpio: io_num=35 **can only be input** (3167) gpio: io_num=34 **can only be input**

Im asking for help here, i already know I have it wrong.

I dont know what I need these for, as i dont have connections for them on the ethernet board but if i just put model=lan8720 into the config i still get 
Configuring Ethernet failed: ESP_FAIL

any other set of gpio i pick results in a boot loop with the following in the log

I (3098) network_handlers: Network manager configuration: polling max 10176, polling min 15000, ap delay 20000, dhcp timeout 8000, eth timeout 8000
I (3128) http_server: Initializing HTTP Server
I (3128) network_ethernet: Detected driver lan8720
I (3128) network_ethernet: Attempting to initialize Ethernet
I (3128) network_ethernet: Detected driver lan8720
I (3138) network_ethernet: Ethernet config => model: lan8720, valid: YES, type: RMII, mdc:16, mdio:4, rst:32, intr:-1, cs:0, speed:0, host:1
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009b696  PS      : 0x00060c33  A0      : 0x8009bf69  A1      : 0x3ffbcc50
A2      : 0x00000000  A3      : 0x00000028  A4      : 0x3ffc6678  A5      : 0x3f41b8fd
A6      : 0x400f7eb8  A7      : 0x00000000  A8      : 0x8009bf3a  A9      : 0x3ffbcc20
A10     : 0x3ffc58b0  A11     : 0x3ffc58b8  A12     : 0x3ffc58b0  A13     : 0x00060c20
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000018  LBEG    : 0x40093730  LEND    : 0x4009374c  LCOUNT  : 0x00000000