sle118/squeezelite-esp32 #170: Bootup stuck - endless Wifi discovery

c-OOL-io, 2022-09-22:
Check here for more details and logs: https://forums.slimdevices.com/showthread.php?116788-ANNOUNCE-Squeezelite-ESP32-version-2-x&p=1061279&viewfull=1#post1061279

c-OOL-io, 2022-09-22:
New log ...

I (7475) network_wifi: Initiating wifi network scan
I (7475) dns_server: DNS Server listening on 53/udp
[00:14:52.497650] discover_server:828 sending discovery
[00:14:57.497608] discover_server:828 sending discovery
**### E (18425) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_TIMER]**
[00:15:02.497575] discover_server:828 sending discovery
[00:15:07.497581] discover_server:828 sending discovery
[00:15:12.497581] discover_server:828 sending discovery
[00:15:17.497581] discover_server:828 sending discovery
[00:15:22.497572] discover_server:828 sending discovery

c-OOL-io, 2022-09-22:
And here a log when the device has started already successfully, then hit reboot with another successful startup.

error related to wifi

I (3375) wifi:E (3375) gpio: gpio_set_level(237): GPIO output gpio_num error
enable tsf
[00:22:45.6I (3385) network_handlers: Existing wifi config found. Attempting to connect.

here the full log:

W (1321385) messaging: System reboot requested.
I (1322135) wifi:state: run -> init (0)
I (1322135) wifi:pm stop, total sleep time: 871158662 us / 1316184186 us

I (1322135) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
W (1322145) wifi:age timer already deinit
W (1322145) wifi:hmac tx: ifx0 stop, discard
W (1322145) wifi:hmac tx: ifx0 stop, discard
W (1322145) wifi:hmac tx: ifx0 stop, discard
W (1322155) wifi:hmac tx: ifx0 stop, discard
W (1322165) wifi:hmac tx: ifx0 stop, discard[00
:22:37W (1322165) wifi:.22hmac tx: ifx0 stop, discard934
5] slimprotoW (1322175) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 2c:91:ab:db:23:88, reason code: 8 (WIFI_REASON_ASSOC_L_rEAVE)
un:609 error reading from socket: Software caused connection abort
W (1322155) httpd_txrx: httpd_sock_err: error in recv : 113
W (1322215) httpd_txrx: httpd_sock_err: error in recv : 113
W (1322225) httpd_txrx: httpd_sock_err: error in recv : 113
[00:22:37.354059] slimproto:967 unable to connect to server 0
W (1327155) wifi:TX Q not empty: 500, TXQ_BLOCK=0
W (1327155) wifi:force witi stop
I (1327155) wifi:flush txq
I (1327155) wifi:stop sw txq
I (1327155) wifi:lmac stop hw txq
W (1327165) wifi:sw tx 0 state not idle, potential error!
W (1327165) wifi:force sw tx 0 state to idle, ebuf flag=0
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6480
load:0x40078000,len:15868
load:0x40080400,len:4384
entry 0x40080684
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 07:17:25
I (27) boot: chip revision: 1
I (30) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) boot_comm: chip revision: 1, min. application chip revision: 0
I (122) esp_image: segment 0: paddr=00150020 vaddr=3f400020 size=aec8ch (715916) map
I (343) esp_image: segment 1: paddr=001fecb4 vaddr=3ffbdb60 size=01364h (  4964) load
I (345) esp_image: segment 2: paddr=00200020 vaddr=400d0020 size=1c4a88h (1854088) map
I (898) esp_image: segment 3: paddr=003c4ab0 vaddr=3ffbeec4 size=05c64h ( 23652) load
I (906) esp_image: segment 4: paddr=003ca71c vaddr=40080000 size=1de48h (122440) load
I (948) esp_image: segment 5: paddr=003e856c vaddr=400c0000 size=00068h (   104) load
I (949) esp_image: segment 6: paddr=003e85dc vaddr=50000000 size=00010h (    16) load
[... 181 lines cut ...]
[00:22:48.208122] raop_sink_start:173 Starting Airplay for ip 192.168.178.166 with servicename MAKT32
[00:22:48.223046] raop_create:201 starting mDNS with 8CCE4EBB4814@MAKT32
I (5995) cspot: Starting Spotify (CSpot) servicename MAKT32 with id 8cce4ebb4814
I (5995) cspot: Started CSpot with MAKT32 (bitrate 320)
I ZeroconfAuthenticator.cpp:55: Zeroconf registering handlers
I (6295) httpd_handlers: serving /status.json to peer 192.168.178.34 port 19690
I (7535) httpd_handlers: serving /commands.json to peer 192.168.178.34 port 52454
[00:22:50.776386] discover_server:828 sending discovery
[00:22:50.782870] discover_server:840 got response from: 192.168.178.200:3483
[00:22:50.785879] slimproto:942 connecting to 192.168.178.200:3483
[00:22:50.811650] slimproto:989 connected
[00:22:50.813429] sendHELO:153 mac: 8c:ce:4e:bb:48:14
[00:22:50.815222] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=16,Firmware=v1.0-1188-16,ModelName=SqueezeESP32,MaxSampleRate=192000,alc,aac,ogg,ops,ogf,flc,wav,aif,pcm,mp3
[00:22:50.848036] sendSETD:431 sending dimension 128x64
[00:22:50.859486] notify:223 notified server 192.168.178.200 hport 9999 cport 9090
I (8625) network_status: LMS IP: 192.168.178.200, hport: 9999, cport: 9090
[00:22:50.865646] decode_flush:251 decode flush
[00:22:50.877915] output_flush:446 flush output buffer
[00:22:50.881104] grfb_handler:743 brightness 5
[00:22:50.912928] visu_handler:1232 Stopping visualizer
[00:22:50.917402] handler:54 got equalizer 14
E (8685) gpio: gpio_set_level(237): GPIO output gpio_num error
[00:22:50.924125] visu_handler:1232 Stopping visualizer
I (11515) httpd_handlers: serving /messages.json to peer 192.168.178.34 port 53478
I (14175) httpd_handlers: serving /status.json to peer 192.168.178.34 port 47846

c-OOL-io, 2022-09-24:
Another log ... (never saw before) maybe this new warning message helps to find the reason!? After few RESET it started normally.  

**W (6854) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 2c:91:ab:db:23:88, reason code: 2 (WIFI_REASON_AUTH_EXPIRE)**
W (6864) wifi:Haven't to connect to a suitable AP now!
I (6864) network_wifi: Configuring Access Point.
I (6884) network: Stopping DHCP server
I (6884) network: Starting DHCP server
I (6894) network_wifi: AP SSID: MAKT32
I (6894) network_wifi: AP Password: squeezelite
I (6894) wifi:mode : sta (8c:ce:4e:bb:48:14) + softAP (8c:ce:4e:bb:48:15)
I (6904) wifi:Total power save buffer number: 6
I (6904) wifi:Init max length of beacon: 752/752
I (6904) wifi:Init max length of beacon: 752/752
I (7464) wifi:Total power save buffer number: 6
I (7474) wifi:Total power save buffer number: 6
I (7474) wifi:Set ps type: 1

I (7474) network_wifi: Initiating wifi network scan
I (7474) dns_server: DNS Server listening on 53/udp
[00:00:07.044284] discover_server:828 sending discovery
[00:00:12.044269] discover_server:828 sending discovery
E (18424) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_TIMER]

sle118, 2022-09-26:
I'm going to assume my fix last change fixed this issue, from the following post
https://forums.slimdevices.com/showthread.php?116788-ANNOUNCE-Squeezelite-ESP32-version-2-x&p=1061872#post1061851

And I'll close this one

closed 2022-09-26 by sle118