sle118/squeezelite-esp32 #171: Can't disconnect from router, to renew the Wifi config.

c-OOL-io, 2022-09-22:
fw: I2S-4MFlash.16.1188.master-v4.3

Since i'm still looking on the boot-up stuck endless wifi discovery issue, it came to my mind that it's maybe somehow the same as with the cspot config issue that changed from the 1st v4.3 to the next one, where I had to delete the whole cspot config from NVS. Therfore I tried to disconnet my the device from the AP, to then rejoin my Wifi from the build in AP. But I can't disconnect from my Router, even when the message says it's discconecte, it's still connected and the internal AP didn't sow up. 

I'm currently also not able to turn of my router, to see if this forces the device to start the build in AP, (casue I think this act in the same way as Tasmota did!?).

c-OOL-io, 2022-09-22:
BTW ... no logs can be provided, since nothing change there  when clicking on Disconnect,

sle118, 2022-09-24:
Il try to have a look as soon as life permits

sle118, 2022-09-24:
Please try to keep related issues together. This looks related to #170 so I'll close it. Too many issues just ends up diluting the attention

closed 2022-09-24 by sle118

c-OOL-io, 2022-09-24:
ok got it!