sle118/squeezelite-esp32 #172: Spotify (cSpot) Options > on save reboot and no change to config

c-OOL-io, 2022-09-22:
fw: I2S-4MFlash.16.1188.master-v4.3

In the Web GUI > System > Spotify (cSpot) Options

Change Bitrate from 160 > 320
Save > device reboots already but no change to the bitrate.
On Save I would not except a reboot as there is an additional Apply button to really activate the new saved settings.

c-OOL-io, 2022-09-22:
Error logs:

I (1359085) httpd_handlers: serving /status.json to peer 192.168.178.34 port 55532
cfg-syst-cspot --deviceName MAKT32 --bitrate 320 

I (1365965) config: Setting changed flag config [cspot_config]
I (1365965) config: Updating config [cspot_config]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40094d91  PS      : 0x00060630  A0      : 0x8026fb40  A1      : 0x3ffd6570  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd6530  
A10     : 0x3ffd6550  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x00000000  
A14     : 0x3ffb69a8  A15     : 0x3ffb6954  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40094d91  LEND    : 0x40094da1  LCOUNT  : 0xffffffff  

Backtrace:0x40094d8e:0x3ffd6570 0x4026fb3d:0x3ffd6580 0x40264147:0x3ffd6890 0x400f6e3d:0x3ffd68e0 0x401c3dc1:0x3ffd6910 0x400f5acf:0x3ffd6930 0x400f6182:0x3ffd6970 0x400d56ac:0x3ffd6990

ELF file SHA256: 5acde0f78329534e

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6480
load:0x40078000,len:15868
load:0x40080400,len:4384
entry 0x40080684
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 07:17:25
I (27) boot: chip revision: 1

c-OOL-io, 2022-09-24:
Same issue with: Hardware > I2C Bus Parameters > Save

c-OOL-io, 2022-09-24:
Error log for I2C

I (27494) httpd_handlers: serving /messages.json to peer 192.168.178.34 port 28863
cfg-hw-i2c --port 1 --speed 400000 --sda 4 --scl 5 

E (28674) i2c: i2c_master_cmd_begin(1147): i2c driver not installed
I (28674) messaging: cfg-hw-i2c
Initializing i2c driver configuration.
   mode = I2C_MODE_MASTER, 
   scl_pullup_en = GPIO_PULLUP_ENABLE, 
   i2c port = 1, 
   sda_io_num = 4, 
   sda_pullup_en = GPIO_PULLUP_ENABLE, 
   scl_io_num = 5, 
   scl_pullup_en = GPIO_PULLUP_ENABLE, 
   master.clk_speed = 400000

I (28704) messaging: cfg-hw-i2c
Installing i2c driver on port 1

I (28714) messaging: Updating I2C configuration to scl=5,sda=4,speed=400000,port=1
I (28724) config: Setting changed flag config [i2c_config]
I (28734) config: Updating config [i2c_config]
I (28734) messaging: cfg-hw-i2c
Uninstalling i2c driver from port 1 if needed
i2c driver successfully started.
Storing i2c parameters.
Done.

squeezelite-esp32> I (28954) config: configuration has some uncommitted entries
I (28954) config: Committing configuration to nvs. Locking config object.
I (28974) nvs_utilities: Value stored under key 'i2c_config'
I (28974) config: Done Committing configuration to nvs.
E (29724) I2CDefaultWri
***ERROR*** A stack overflow in task sb_displayer has been detected.

Backtrace:0x40081a9d:0x3f802b30 0x40097c45:0x3f802b50 0x400981fd:0x3f802b70 0x40098f7b:0x3f802bf0 0x40097d3c:0x3f802c30 0x40097cee:0x00000000 |<-CORRUPTED

ELF file SHA256: 5acde0f78329534e

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6480
load:0x40078000,len:15868
load:0x40080400,len:4384
entry 0x40080684
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader

sle118, 2022-11-25:
This should now be resolved with https://github.com/sle118/squeezelite-esp32/commit/8462de1c1665da9d8316831c30ab546a224da785

closed 2022-11-25 by sle118