sle118/squeezelite-esp32 #173: [GUI add request] Player name all the time visible in the header menu

c-OOL-io, 2022-09-22:
Please add the name of the Player which is also used in LMS to easily see all the time on which player you are operating on.
Maybe directly below the Squeezelite-ESP32 "branding"?

sle118, 2022-09-24:
Wouldn't the host name in the web browser provide the same info?

c-OOL-io, 2022-09-25:
Honestly, I never changed the player name in LMS. So I don't know if those could be different, or will be pushed back to the player!? I Have only seen that the name will be pushed/read from the player to LMS. Anyhow, sure the hostname (variable), which is currently only visible @Audio > Player Name (= host_name NVS) was exactly what i meant. Very easy to implement, but with a lot of benefit for those with several players. ;)

sle118, 2022-09-26:
When you go to the system tab and change the device name, this updates **everything**. Host name, LMS player name, airplay, etc. 

Have a look at your browser tag title and you'll see that it's there too. I think that pretty much addresses the requirement. Yes it's easy to implement, but I have limited bandwidth right now and have to decline.

closed 2022-09-26 by sle118

c-OOL-io, 2022-09-26:
THX for the hint. Didn't realize that, cause I'm mostly using FF @ kiosk/full screen mode. So all good!