sle118/squeezelite-esp32 #174: [Feature Request] Add more Bluetooth codes

c-OOL-io, 2022-09-22:
As already possible in piCorePlayer, add more BT codes, like aptX, aptX HD, AAC, etc to the build in BT.

sle118, 2022-09-24:
I don't mean to disappoint you, but this is very unlikely to happen, unless expressif releases some optimized codec as part of the BT audio support. We are also running very low on available room in flash memory.

PiCorePlayer runs on typically more powerful hardware when compared to the esp32.

sle118, 2022-09-24:
I'm going to flash @philippe44 to see what he thinks

philippe44, 2022-09-24:
No, it won't happen

c-OOL-io, 2022-09-24:
ok, wifi is even better for audio quality then BT. I don't need it on my own, only intereset in home use with multiroom. but thought if possible, interesting for others who are using ESP32 devices also on the go.  ;)

closed 2022-09-24 by sle118