sle118/squeezelite-esp32 #178: Missing driver for I2S DAC:  UDA1334ATS (Makerfabs MPAUDIO Audio Extension board) + Mute GPIO34 issue

c-OOL-io, 2022-09-22:
Have to mention that the GPIO34 error of the I2S DAC, shown also in the log, is to the used Boards [ MAKT32 (MakePython ESP32 V1.2)] (https://www.makerfabs.com/makepython-esp32.html) + [ MPAUDIO (MakePython Audio Expansion V2.0)](https://www.makerfabs.com/makepython-audio.html).

To get rid of the extreme boosted Bass I had to add mute values to the dac_config:  model=I2S,bck=26,ws=25,do=27,**mute=34:0**,sda=4,scl=5,i2c=106

I had never expected that GPIO34 of Mute does the trick: https://forums.slimdevices.com/showthread.php?116799-Squeezelite-ESP32-gt-3-NEW-OpenHardware-Boards-(MAKT32-MAKT32CL-ESP32-ADF)&p=1061223&viewfull=1#post1061223

Because with FW 1.x I was never able to set the Mute GPIO34 via the hardware web interface. Because of the same error message in the logs.

error log:

I (3325) DAC core: DAC uses I2C port:0, sda:4, scl:5
E (3335) i2c: i2c driver install error
I (3335) DAC external: DAC on I2C @106
W (3335) DAC external: no i2c controlset found
I (3355) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=12
I (3355) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199I (3365) wifi:.625000, diva: 1, divb: 0
mode : sta (8c:ce:4e:bb:48:14)E (3365) gpio: io_num=34 can only be input
...
[00:22:50.917402] handler:54 got equalizer 14
E (8685) gpio: gpio_set_level(237): GPIO output gpio_num error

Audio and EQ is working without any problems!

Full log:
I (1311925) httpd_handlers: serving /messages.json to peer 192.168.178.34 port 28313
W (1321385) messaging: System reboot requested.
W (1321385) messaging: System reboot requested.
I (1322135) wifi:state: run -> init (0)
I (1322135) wifi:pm stop, total sleep time: 871158662 us / 1316184186 us

I (1322135) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
W (1322145) wifi:age timer already deinit
W (1322145) wifi:hmac tx: ifx0 stop, discard
W (1322145) wifi:hmac tx: ifx0 stop, discard
W (1322145) wifi:hmac tx: ifx0 stop, discard
W (1322155) wifi:hmac tx: ifx0 stop, discard
W (1322165) wifi:hmac tx: ifx0 stop, discard[00
:22:37W (1322165) wifi:.22hmac tx: ifx0 stop, discard934
5] slimprotoW (1322175) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 2c:91:ab:db:23:88, reason code: 8 (WIFI_REASON_ASSOC_L_rEAVE)
un:609 error reading from socket: Software caused connection abort
W (1322155) httpd_txrx: httpd_sock_err: error in recv : 113
W (1322215) httpd_txrx: httpd_sock_err: error in recv : 113
W (1322225) httpd_txrx: httpd_sock_err: error in recv : 113
[00:22:37.354059] slimproto:967 unable to connect to server 0
W (1327155) wifi:TX Q not empty: 500, TXQ_BLOCK=0
W (1327155) wifi:force witi stop
I (1327155) wifi:flush txq
I (1327155) wifi:stop sw txq
I (1327155) wifi:lmac stop hw txq
W (1327165) wifi:sw tx 0 state not idle, potential error!
W (1327165) wifi:force sw tx 0 state to idle, ebuf flag=0
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6480
load:0x40078000,len:15868
load:0x40080400,len:4384
entry 0x40080684
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 07:17:25
I (27) boot: chip revision: 1
I (30) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
[... 199 lines cut ...]
I (5995) cspot: Starting Spotify (CSpot) servicename MAKT32 with id 8cce4ebb4814
I (5995) cspot: Started CSpot with MAKT32 (bitrate 320)
I ZeroconfAuthenticator.cpp:55: Zeroconf registering handlers
I (6295) httpd_handlers: serving /status.json to peer 192.168.178.34 port 19690
I (7535) httpd_handlers: serving /commands.json to peer 192.168.178.34 port 52454
[00:22:50.776386] discover_server:828 sending discovery
[00:22:50.782870] discover_server:840 got response from: 192.168.178.200:3483
[00:22:50.785879] slimproto:942 connecting to 192.168.178.200:3483
[00:22:50.811650] slimproto:989 connected
[00:22:50.813429] sendHELO:153 mac: 8c:ce:4e:bb:48:14
[00:22:50.815222] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=16,Firmware=v1.0-1188-16,ModelName=SqueezeESP32,MaxSampleRate=192000,alc,aac,ogg,ops,ogf,flc,wav,aif,pcm,mp3
[00:22:50.848036] sendSETD:431 sending dimension 128x64
[00:22:50.859486] notify:223 notified server 192.168.178.200 hport 9999 cport 9090
I (8625) network_status: LMS IP: 192.168.178.200, hport: 9999, cport: 9090
[00:22:50.865646] decode_flush:251 decode flush
[00:22:50.877915] output_flush:446 flush output buffer
[00:22:50.881104] grfb_handler:743 brightness 5
[00:22:50.912928] visu_handler:1232 Stopping visualizer
[00:22:50.917402] handler:54 got equalizer 14
E (8685) gpio: gpio_set_level(237): GPIO output gpio_num error
[00:22:50.924125] visu_handler:1232 Stopping visualizer
I (11515) httpd_handlers: serving /messages.json to peer 192.168.178.34 port 53478
I (14175) httpd_handlers: serving /status.json to peer 192.168.178.34 port 47846
I (26045) httpd_handlers: serving /status.json to peer 192.168.178.34 port 51430
[00:23:20.973955] grfb_handler:743 brightness 3

c-OOL-io, 2022-09-24:
I have now noticed that no matter what I have tried, this extreme bass I had since the beginning can no longer set/activated. I think it was the video mode of the DAC [UDA1334ATS](https://www.nxp.com/docs/en/data-sheet/UDA1334ATS.pdf) ,  Therefore, I now think that it is enough to set the Mute GPIO 34 via set_GPIO [ 34=mute:0 ] to get rid of it, when flashing a device from factory the 1st time with Squeezelite ESP32.

sle118, 2022-12-23:
Just wondering. Did you figure this out? If so, then I'll close the issue

sle118, 2023-03-06:
allrigth. this has gone stale so I'm going to just close for now.

closed 2023-03-06 by sle118