sle118/squeezelite-esp32 #179: BT output does not work - long discovery then crash/reboot

c-OOL-io, 2022-09-23:
fw: I2S-4MFlash.16.1188.master-v4.3

The BT output does not work at all. No pairing, only nearly endless discovery until a crash & reboot.

To test this I mainly I was using an external BT Receiver/Transmitter (Elegiant BTI-066) with staus LED to see the used BT Audio Codec (aptX, aptX LL, aptX HD, SBC, AAC). It has a 3,5mm Audio jack (IN or OUT) and 2 OPT (IN/OUT). The device was found, but not connected. (tested in addition also with some JBL and S2G active BT speakers with the same negative result)
  

Error Log:
...
I (163265) bt_app_source: Device discovery failed, continue to discover...
I (163265) bt_app_source: Discovery started.
I (173655) bt_app_source: Found BT Sink Device: ELEGIANT BTI-066 rssi is -26
I (176095) bt_app_source: Device discovery failed, continue to discover...
I (176095) bt_app_source: Discovery started.
I (188935) bt_app_source: Device discovery failed, continue to discover...
...
 (16965) bt_app_source: Device discovery failed, continue to discover...
I (16975) bt_app_source: Discovery started.
ASSERT_PARAM(1024 0), in rwbt.c at line 381
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400853e8: 0000f01d 00004136 f01d0000
Core  0 register dump:
PC      : 0x400853ef  PS      : 0x00060031  A0      : 0x800857a5  A1      : 0x3ffbeaa0  
A2      : 0x00000000  A3      : 0x00000400  A4      : 0x00000000  A5      : 0x3f44dde8  
A6      : 0x0000017d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffbea10  
A10     : 0x00000000  A11     : 0x3ffbea33  A12     : 0x3ffbe9df  A13     : 0x00000031  
A14     : 0x00000000  A15     : 0x3ffbe9e4  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400856ad  LEND    : 0x400856b5  LCOUNT  : 0x00000000  

Backtrace:0x400853ec:0x3ffbeaa0 0x400857a2:0x3ffbeac0 0x4008bb39:0x3ffbeae0 0x4008bf8b:0x3ffbeb00 0x400842f1:0x3ffbeb20 0x402843b7:0x3ffbd5c0 0x400dc9e9:0x3ffbd5e0 0x402932bf:0x3ffbd600

ELF file SHA256: 5acde0f78329534e

Rebooting...

sle118, 2022-09-24:
Please refer to the readme
`Note that streaming to a Bluetooth speaker is not the main purpose and remains experimental, so your mileage will vary. We will not work on improving or fixing that feature, please don't open issues about that.`

closed 2022-09-24 by sle118

c-OOL-io, 2022-09-25:
I'm totally fine with that, I would never need it on my own. For myself BT is only needed when on the go. I was just testing all kind of features and the Web GUI itself and reported my findings. Just that, nothing else.

But maybe if you have the time, you can show an experimental TAG (or something like that) in the web GUI for those features that have no or very low priority. Then (hopefully) everyone knows that it is not worth to open up a new issue. This would be a win win situation for everyone. Because the users don't open unnecessary tickets, which you have to comment unnecessarily. Because at the moment I don't know what is important to you and what is not. And I think other users feels the same way. Just as food for thought. ;)