sle118/squeezelite-esp32 #182: Boot fails on ESP32-WROVER-E board after flashing I2S 1.1023

kine90, 2022-10-01:
Hello, I am trying to flash the version I2S 1.1023 using the web installer.
I am using an ESP32-WROVER-DEV V1.6 board based on the ESP32-WROVER-E chip.
After successfully flashing the firmware, the board does not boot.
From the console output it looks like a problem with the partition table.
Thanks in advance for any advice.

Console output:

```
boot: ESP-IDF v4.0.2-206-g5630b17e1 2nd stage bootloader␛[0m
␛[0;32mI (32) boot: compile time 07:30:49␛[0m
␛[0;32mI (32) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (38) qio_mode: Enabling default flash chip QIO␛[0m
␛[0;32mI (43) boot: SPI Speed      : 80MHz␛[0m
␛[0;32mI (47) boot: SPI Mode       : QIO␛[0m
␛[0;32mI (51) boot: SPI Flash Size : 4MB␛[0m
␛[0;31mE (55) flash_parts: partition 0 invalid magic number 0x88aa␛[0m
␛[0;31mE (61) boot: Failed to verify partition table␛[0m
␛[0;31mE (66) boot: load partition table error!␛[0m
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6948
load:0x40078000,len:5580
load:0x03660138,len:646002720
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6948
load:0x40078000,len:1284
ho 0 tail 12 room 4
load:0x09800020,len:-1591703312
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6948
load:0x40078000,len:13772
load:0x40080400,len:4532
csum err:0x36!=0xfe
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6948
load:0x40058000,len:13772
load:0x40080400,len:4532
csum err:0x48!=0xfe
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6948
load:0x40058000,len:13644
load:0xf04631f3,len:-2013257536
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
1162 mmu set 00130000, pos 00130000
1162 mmu set 00140000, pos 00140000
1162 mmu set 00150000, pos 00150000
1162 mmu set 00160000, pos 00160000
1162 mmu set 00170000, pos 00170000
1162 mmu set 00180000, pos 00180000
1162 mmu set 00190000, pos 00190000
1162 mmu set 001a0000, pos 001a0000
ets Jun  8 2016 00:22:57
```

sle118, 2022-10-02:
Which web installer?

kine90, 2022-10-02:
This one: https://raspiaudio.github.io/

c-OOL-io, 2022-10-07:
Here are more logs.with  same or similar issue and an ESPRESSIF ESP32 WROVER E (4MB Flash 8MB PSRAM) dev kit V1.6.
I manually flashed with these partitions as recovery only.
    ota_data_initial.bin @ 0xd000
    bootloader.bin @ 0x1000
    partition-table.bin @ 0x8000
    recovery.bin @ 0x10000

In Both cases the 1st boot ended in the same sate of ready to flash firmware (gain).
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

Her are the detailed error logs:
 
This 1st error log came after I successfully flashed latest FW 2.x (16 bit I2S) as recovery only.
squeezelite-esp32-master-v4.3-I2S-4MFlash-16-2.1223

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6480
load:0x40078000,len:15868
load:0x40080400,len:4384
entry 0x40080684
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 06:34:40
I (27) boot: chip revision: 1
I (30) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
I (95) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (119) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=55760h (350048) map
I (232) esp_image: segment 1: paddr=00065788 vaddr=3ffbdb60 size=04520h ( 17696) load
I (238) esp_image: segment 2: paddr=00069cb0 vaddr=40080000 size=06368h ( 25448) load
I (247) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=cde24h (843300) map
I (497) esp_image: segment 4: paddr=0013de4c vaddr=40086368 size=0ea18h ( 59928) load
I (518) esp_image: segment 5: paddr=0014c86c vaddr=400c0000 size=00068h (   104) load
I (518) esp_image: segment 6: paddr=0014c8dc vaddr=50000000 size=00010h (    16) load
I (534) boot: Loaded app from partition at offset 0x10000
I (534) boot: Disabling RNG early entropy source...
I (546) psram: This chip is ESP32-D0WD
ets Jun  8 2016 00:22:57
[... 63 lines cut ...]
Backtrace: 0x40101fd5:0x3ffe3bb0 0x400d52eb:0x3ffe3c20 0x40081753:0x3ffe3c40 0x400791de:0x3ffe3c80 |<-CORRUPTED

I (650) esp_core_dump_uart: Press Enter to print core dump to UART...
I (657) esp_core_dump_uart: Print core dump to uart...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400921ab  PS      : 0x00060c33  A0      : 0x80091b3f  A1      : 0x3ffe3690  
A2      : 0x3ffe3700  A3      : 0x3ffe36b0  A4      : 0x00000040  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffbec30  A8      : 0x00000000  A9      : 0x3ffe3ad0  
A10     : 0x3ffe3af0  A11     : 0x40101fd4  A12     : 0x3ffc0f4a  A13     : 0x40083f78  
A14     : 0x3ff44574  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008bb42  LEND    : 0x4008bb4d  LCOUNT  : 0x00000000  

ELF file SHA256: bbffeabbefabf7b7

Backtrace: 0x400921a8:0x3ffe3690 0x40091b3c:0x3ffe36b0 0x4008eaa5:0x3ffe36e0 0x4008e6fb:0x3ffe3700 0x4008e915:0x3ffe3a50 0x4008ed10:0x3ffe3a70 0x4008d33d:0x3ffe3ab0 0x4008d616:0x3ffe3ad0 0x40083fc2:0x3ffe3af0 0x40101fd5:0x3ffe3bb0 0x400d52eb:0x3ffe3c20 0x40081753:0x3ffe3c40 0x400791de:0x3ffe3c80 |<-CORRUPTED

Re-entered core dump! Exception happened during core dump!
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

c-OOL-io, 2022-10-07:
THX for mention the issue with a bad flash chip 
https://github.com/espressif/esp-idf/issues/7994

So this means our boards (flash chip) are already bricked, since we flashed them already which leads to a bad flash? Or could they be some how recovered?

Because in the workaround they mentioned a modified bootloader which must be compiled. But since I'm not able (missing time and knowledge) on how to compile a new bootloader, I can only test if someone who knows what he is doing, could create such a workaround!? But then I'm absolutely wilful to test and find a solution for all other effected users on this issue.

sle118, 2022-10-07:
you should try compiling their "hello world" project. This will create all the binaries you need including the bootloader. I don't have one of these board to test the fix or to troubleshoot the issue. If you confirm the nature of the issue, then I'll update the readme with a warning. Moving to a new ESP-IDF release (in this case to get their modified bootloader code) is no easy task as it often leads to unexpected and hard to identify problems.

Given that these new chips are starting to show up in the wild, we'll have to bite the bullet at some point of time but I have a few things to finalize before I spend time on that kind of mini project. 

If needed, I might go and create a feature branch at some point, but keeping the code in sync between branches is almost a full time job and I don't have that bandwidth right now :)

kine90, 2022-10-07:
Hi!
I found a work around which seems to be working in my case, but I haven't extensively tested yet.
I am on a windows machine.
Download the ESP-IDF 4.3 installer from their website. It should make everything easily ready for you to compile amd flash. Using the IDF command prompt try the following:
1- Erase completely the flash (you can find the commands online)
2- Move to the Hello World project folder (it is inside the IDF folder, under Examples)
3- Build the project.
3- At the end of the build process, you get a recommendation about the command to use to flash the Hello World project. Do it and check that everything worked using the serial monitor.
4-At this point, move back to the Squeezelite release folder and, using the same parameters in the flash command, flash all the partition as instructions EXCEPT the bootloader.

With a bit of luck you should finally get Squeezelite running!

sle118, 2022-10-07:
This will actually provide a bootloader with the latest code from Espressif. This is probably a good around for now. 

one way to easily get all the dependencies you need for compiling an espressif project is to install docker and, from the root folder of a project, call the following line (in a windows host).
`docker run --rm -v %cd%:/project -w /project -it espressif/idf:release-v4.3`

from there, you can do
`idf.py build`

c-OOL-io, 2022-10-10:
And I get these error logs when I flash all partitions in the same way (80Mhz - 460800)

ts Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6560
load:0x40078000,len:15448
load:0x40080400,len:4324
entry 0x40080684
I (27) boot: ESP-IDF v4.3.4-73-g9e863ecdb8 2nd stage bootloader
I (27) boot: compile time 19:15:50
I (27) boot: chip revision: 1
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (104) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (116) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=55760h (350048) map
I (225) esp_image: segment 1: paddr=00065788 vaddr=3ffbdb60 size=04520h ( 17696) load
I (231) esp_image: segment 2: paddr=00069cb0 vaddr=40080000 size=06368h ( 25448) load
I (239) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=cde24h (843300) map
I (472) esp_image: segment 4: paddr=0013de4c vaddr=40086368 size=0ea18h ( 59928) load
I (492) esp_image: segment 5: paddr=0014c86c vaddr=400c0000 size=00068h (   104) load
I (492) esp_image: segment 6: paddr=0014c8dc vaddr=50000000 size=00010h (    16) load
I (508) boot: Loaded app from partition at offset 0x10000
I (508) boot: Disabling RNG early entropy source...
I (520) psram: This chip is ESP32-D0WD
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6560
load:0x5557d555,len:1431666013
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
1162 mmu set 00130000, pos 00130000
1162 mmu set 00140000, pos 00140000
1162 mmu set 00150000, pos 00150000
1162 mmu set 00160000, pos 00160000
1162 mmu set 00170000, pos 00170000
1162 mmu set 00180000, pos 00180000
1162 mmu set 00190000, pos 00190000
1162 mmu set 001a0000, pos 001a0000
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6560
load:0x5557d555,len:1431666013
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
1162 mmu set 00130000, pos 00130000
1162 mmu set 00140000, pos 00140000
1162 mmu set 00150000, pos 00150000
1162 mmu set 00160000, pos 00160000
1162 mmu set 00170000, pos 00170000
1162 mmu set 00180000, pos 00180000
1162 mmu set 00190000, pos 00190000
1162 mmu set 001a0000, pos 001a0000
ets Jun  8 2016 00:22:57

...

sle118, 2022-10-10:
Ok. Your problem doesn't seem to be caused by a recent version of the chip, as the bootloader reports `ESP32-D0WD`. The boot process looks like it dies when initializing the psram. Since I don't have one of these, I can't help much.

If you manage to figure out the issue and a solution, please report here.

sle118, 2022-10-11:
Also. Please confirm that there's nothing connected to the gpio's

sle118, 2022-10-11:
That makes sense, especially since some logs are crashing right before the PSRAM init has a chance to compete

sle118, 2022-11-01:
> Ok understood! Since this Wrover-E Dev Kit is by far the cheapest I've bought so far, I was very suspicious if I'd even get what's being sold there. Because in the meantime I noticed that on the SoC cap it says WROVER-E, but the code number XX0H32, which is also on the cap, does not match at all. The code XX0H32 actually belongs to WROOM-32E. With a WROVER-E, it would actually have to be N4R8, if it really is a WROVER-E from Expressife. Therefore, I now rather assume that this is a fake or very strange custom build. Meanwhile I think, because of the very low price of 3 EUR that I paid, that it is a fake chip.

I have finally received the boards, which I ordered from AliExpress.  I flashed the I2S build on one of them with our new web installer (see below) and the board booted flawlessly. Since this seems to be a hardware issue, I'll go ahead and close the issue.

https://sle118.github.io/squeezelite-esp32-installer/

closed 2022-11-01 by sle118

stephiegu, 2022-11-03:
Hello @sle118 
I am trying to get this thing working. I got it working with flash installer on one ESP32-WROVER board. From the pictures it seems to be the same likes yours. I have two batches of devices. One from Ali and one from Amazon. Sadly all other (about 10 devices) are not flashing. Every time after flash it goes to endless boot-loop showing:

Even pressing reset is not helping here.
Did you something/anything as preparation for the devices? Are you holding any buttons while flashing it? 
I somehow can't believe that all devices are factory broken. Especially because the blink example works on all of them.

Things I see while web-browser flashing is that it hangs at 44% and then jumps to 68% and then hangs from 88% and jumps to 99%.
I tried different cables. Different machines and different OS (mac and windows). 
Can I help with any information? It would be really cool to be able using squeezelite.
The boards are all 4MB Versions.

After a flash-test (that always ends in endless boot-loop) I have to unplug the device, hold boot and then plug it in. Only then re-flashing is possible.

Would be really nice if we can solve this out to have a reliable installable version.
I am really curious how you were able to install it on the board :)

Best,
Stephie

kine90, 2022-12-18:
Hello @sle118 , I have been able to build squeezlite using docker and flash my two boards (those instructions helped a lot: https://github.com/sle118/squeezelite-esp32/issues/196#issuecomment-1296846638).
Everything works this way, the boards flashed and booted without any issue. So it looks like something is wrong with the webinstaller or the precompiled packages, not with the hardware.

Vanixxx, 2023-09-01:
Hi @kine90 ,
can you give more details what exactly you did? 
Thx

acca84, 2024-01-01:
> > Ok understood! Since this Wrover-E Dev Kit is by far the cheapest I've bought so far, I was very suspicious if I'd even get what's being sold there. Because in the meantime I noticed that on the SoC cap it says WROVER-E, but the code number XX0H32, which is also on the cap, does not match at all. The code XX0H32 actually belongs to WROOM-32E. With a WROVER-E, it would actually have to be N4R8, if it really is a WROVER-E from Expressife. Therefore, I now rather assume that this is a fake or very strange custom build. Meanwhile I think, because of the very low price of 3 EUR that I paid, that it is a fake chip.
> 
> I have finally received the boards, which I ordered from AliExpress. I flashed the I2S build on one of them with our new web installer (see below) and the board booted flawlessly. Since this seems to be a hardware issue, I'll go ahead and close the issue.
> 
> https://sle118.github.io/squeezelite-esp32-installer/
> 
> for the record, here's the boot log
> 
> ```
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:6480
> load:0x40078000,len:15868
> load:0x40080400,len:4384
> entry 0x40080684
> I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
> I (27) boot: compile time 06:34:40
> I (27) boot: chip revision: 1
> I (30) qio_mode: Enabling default flash chip QIO
> I (36) boot.esp32: SPI Speed      : 80MHz
> I (40) boot.esp32: SPI Mode       : QIO
> I (45) boot.esp32: SPI Flash Size : 4MB
> I (49) boot: Enabling RNG early entropy source...
> I (55) boot: Partition Table:
> I (58) boot: ## Label            Usage          Type ST Offset   Length
> I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
> I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
> I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
> I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
> I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
> I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
> I (111) boot: End of partition table
> I (115) boot: Defaulting to factory image
> I (120) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=55760h (350048) map
> I (232) esp_image: segment 1: paddr=00065788 vaddr=3ffbdb60 size=04520h ( 17696) load
> I (238) esp_image: segment 2: paddr=00069cb0 vaddr=40080000 size=06368h ( 25448) load
> I (247) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=cde24h (843300) map
> I (497) esp_image: segment 4: paddr=0013de4c vaddr=40086368 size=0ea18h ( 59928) load
> I (518) esp_image: segment 5: paddr=0014c86c vaddr=400c0000 size=00068h (   104) load
> I (518) esp_image: segment 6: paddr=0014c8dc vaddr=50000000 size=00010h (    16) load
> I (534) boot: Loaded app from partition at offset 0x10000
> I (534) boot: Disabling RNG early entropy source...
> I (546) psram: This chip is ESP32-D0WD
> I (547) spiram: Found 64MBit SPI RAM device
> I (547) spiram: SPI RAM mode: flash 80m sram 80m
> I (550) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
> I (557) cpu_start: Pro cpu up.
> I (561) cpu_start: Starting app cpu, entry point is 0x4008130c
> I (554) cpu_start: App cpu up.
> I (1064) spiram: SPI SRAM memory test OK
> I (1073) cpu_start: Pro cpu start user code
> I (1073) cpu_start: cpu freq: 240000000
> I (1073) cpu_start: Application information:
> I (1076) cpu_start: Project name:     Squeezelite-ESP32
> I (1082) cpu_start: App version:      I2S-4MFlash.16.1223.master-v4.3
> I (1089) cpu_start: Compile time:     Sep 25 2022 06:36:16
> I (1095) cpu_start: ELF file SHA256:  42e4dfd835460693...
> I (1101) cpu_start: ESP-IDF:          v4.3.2-dirty
> ```
> 
> And a picture of the board [image]

Hello and happy new year, i have bought the same board and flashed it with your web installer, but it boot loops with this message : 
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```
I can't flash it again now, i tried two boards, same result
Any idea please ?

kine90, 2024-01-03:
Sorry for the late. My two boards never worked reliably in this year, finally I found some time to look again in this project.
I found this conversation which helped: https://www.esp32.com/viewtopic.php?t=3452
In my design, I used PIN14 for the DAC. As pointed out in that conversation, and confirmed from the datasheet, IO12 aka PIN14 is MTDI and should be pulled low at boot to properly set the SRAM module voltage.

acca84, 2024-01-03:
Hello, I have the same problem with the same board
If I erase it with flashdownloadtool then flash it with WE installer, it works and boots squeezebox as long as I don't perform a hard reset
Could the wrong bootloader be flashed by we installer ?
Kine90, could you share your bin image please ?

kine90, 2024-01-03:
Sure, here is a link to a zip with the full repo and my build files. I think I did anything special while bulding with the docker image though. https://drive.google.com/file/d/1UsrLi7lae5IL47-zW2U0KGBWFocX07bv/view?usp=sharing

acca84, 2024-01-03:
Thanks
I flashed your files and it works but I have to push EN button in order to boot (like you)
I also built the files using docker and I2S-4MFlash-sdkconfig.defaults, but the esp32 didn't boot at all
Strange !
I think I'll try another device
Thanks again

ses1er, 2024-01-03:
I was unable to get whatever has been suggested here to work.

I did however figured out a way to get mine working, the exact module pictured above.

1. Flash using the squeezelite-esp2 web installer: https://sle118.github.io/squeezelite-esp32-installer/. I've found it mostly doesn't boot correctly as desribed in this thread. However, it does boot correctly 1 out of 50 times with no changes, which is weird.
2. Download the ESP32-AT firmware from Espressif and extract: https://dl.espressif.com/esp-at/firmwares/esp32/ESP32-WROVER-32/ESP32-WROVER-32-AT-V2.4.0.0.zip
3. Flash the bootloader from ESP32-AT over the bootloader from squeezelite-esp32 web installer: (Mac) ```python3 -m esptool write_flash 0x1000 ./ESP32-WROVER-32-AT-V2.4.0.0/ESP32-WROVER-V2.4.0.0/bootloader/bootloader.bin```

It now appears to boot correctly every single time.

ses1er, 2024-01-03:
Interestingly, I just flashed the bootlaoder from @kine90 's  dump and it does boot correctly with the following output:

```
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 14:22:53
I (27) boot: chip revision: v1.0
I (30) qio_mode: Enabling default flash chip QIO
I (35) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
```

It is also 40MHz. The most logical conclusion to me at the moment is the module I have does not support 80Mhz correctly.

ses1er, 2024-01-04:
> Thanks I flashed your files and it works but I have to push EN button in order to boot (like you) I also built the files using docker and I2S-4MFlash-sdkconfig.defaults, but the esp32 didn't boot at all Strange ! I think I'll try another device Thanks again

if you jump 3v3 with GPIO0, it'll boot automatically.

acca84, 2024-01-04:
Thanks, it does, but bootloops after exiting recovery
After that, I have to flash again

ses1er, 2024-01-04:
> Thanks, it does, but bootloops after exiting recovery After that, I have to flash again

Can you try using the bootloader from ESP-AT as I had onlined? I ran into this issue with @kine90 's bootloader.

All said, I agree these modules are to be avoided. I have a WROVER bought from Amazon that has none of these issues.

acca84, 2024-01-04:
I tried Esp32-at bootloader, it didn't bootloop during 5 minutes then bootlooped
I think I'll just avoid this module 
Thanks

DonKracho, 2024-02-22:
I buyed two of these modules from amazon (a FREENOVE clone), because they were dirty cheap and other WROVER dev boards are hardly to find, or you have to wait to get them from China. The camera applicalion, compiled with VS-Code and PlatformIO, did work. But once being in a wired state you need several approaches to program this device. The bootloop mentioned is definetly caused by the bootloader.

And indeed, to boot up properly on power on it needs a 10k pull-up resistor to 3.3V at the 0 pin.  Do not connect it  directly to 3.3V, because the boot switch will tie this pin to ground. By default pin 0 pin is floating for some reason and causing the not booting issue when powered on.

I also failed using the Web installers. Just find a propper bootloader.bin or just do not program the bootloader when it is booting properly and use the rest of the squeezelite binaries only. Once you have got this 'clone of shit' working you can use the OTA updates flawlessly. I2S-4MFLash-1670 was a great improvement using the simple S/P-Dif interface for me! Thanks a lot to the developers! 

But I agree to the conclusion, avoid this WROVER-E Cam-Module, it is to tricky to handle.

sle118, 2024-02-24:
I guess you get what you pay for, and here we're talking a couple of bucks cheaper than the legitimate options.

Stay clear of these cheap boards or they will cost you way more in time than actually money "saved"

polygon242, 2024-08-20:
Gor the same problem (no boot after flash) with this cheap board, sending it back. 
Flashing Tasmota beforehand worked, surprisingly. Using Wrover-cam firmware and right template even the cam worked as expected.

nehasjains, 2026-08-24:
How to use pcm5102 dac?

What configurations do I have to do now ?

I have using the same board esp32-e wrover

I have able to see the player in LMS. I have don't connected the pcm5102 dac yet.

Please guide