sle118/squeezelite-esp32 #183: Point LMS or Squeezelite-esp32 to a different subnet?

NonaSuomy, 2022-10-01:
Does anyone know how to point Logitech Media Server (docker) 10.42.42.* at a different IP range (wifi) 10.13.37.* where the squeezelite-esp32 exists?

Or Squeezelite-esp32 to where the LMS is?

Thank you.

sle118, 2022-10-02:
I recommend you check either in the LMS forums, or that you ask the question to the owner of the docket image you are running. I am not using such image, so I can't help.

https://forums.slimdevices.com/index.php

closed 2022-10-02 by sle118

NonaSuomy, 2022-10-02:
What do you use? The owner of the docker image is Logitech I believe https://github.com/Logitech/slimserver-platforms/tree/public/8.3/Docker

Is there anyway to tell squeezelite-esp32 where the server is?

daduke, 2022-10-02:
you mean like `-s 192.168.0.1`?

NonaSuomy, 2022-10-02:
@daduke thanks for commenting. Where would you enter this into squeezelite-esp32 webui?

daduke, 2022-10-02:
yes autoexec1 is correct. If it still won't connect, something's iffy in your network.

NonaSuomy, 2022-10-02:
Thank you for confirming that is the port correct I noticed there are other ports listed what are they for?

```
  logitech-media-server:
    container_name: lms
    hostname: lms-docker
    image: lmscommunity/logitechmediaserver:latest
    #restart: unless-stopped
    ports:
      - "9000:9000"
      - "9090:9090"
      - "3483:3483"
      - "3483:3483/udp"
    volumes:
      # Folder where LMS stores its data (cache, logs, prefs)
      - /var/lib/libvirt/images/usb001/lms/config:/config
      # Folder where LMS should look for music
      - /var/lib/libvirt/images/usb001/audio/music:/music
      # use time of host
      - /etc/localtime:/etc/localtime:ro
      #- /etc/timezone:/etc/timezone:ro
    mac_address: 9e:57:59:8e:5e:28
    networks:
      - dhcp
    environment:
      - PUID=1000
      - PGID=1003

networks:
  dhcp:
    name: dbrv700
    external: true
```

daduke, 2022-10-02:
LMS default is 9000, as for the others: https://superuser.com/questions/1442385/which-ports-must-be-open-on-the-server-to-use-logitech-media-server-lms-for-sq

NonaSuomy, 2022-10-02:
So should I be pointing the player device at "3483 TCP: Physical Squeezebox" instead of "9000 TCP: Web GUI" or maybe just leave the port blank as it seems to notation that it's optional?

squeezelite -b 500:2000 -d all=info -C 30 -W -n "squeezelite-face01" -o I2S -s 10.42.42.64:3483

Elkropac, 2022-11-30:
Thanks, adding `-s ip` helped with discovery in home assistant. It seems it's not needed for later usage. I noticed, it keeps dissapearing from the input box after reboot