sle118/squeezelite-esp32 #184: [FEATURE] Support ESP32-ADF from Olimex

zeridon, 2022-10-03:
I would like to have ESP32-ADF from Olimex supported as a targed (sdkconfig)

The board itself is ESP AudioKit very similar to A1S from AI Thinkerer.

Product Page: https://www.olimex.com/Products/IoT/ESP32/ESP32-ADF/open-source-hardware
Schematic: https://github.com/OLIMEX/ESP32-ADF/tree/master/HARDWARE/ESP32-ADF%20Rev.C (kicad + schematic in PDF)

I have a board available for testing and am not afraid to do light meddling with code.

Alternatively point me where to start playing with config and will try to do it myself

AxelJust, 2022-10-04:
I have a Olimex running.
It is the identical configuration as with a Lyrat

zeridon, 2022-10-05:
@AxelJust Do you mind sharing your configs

c-OOL-io, 2022-10-07:
Here you can find 2 fully working configs for the OLiMEX ESP32-ADF board.
[https://squeezeesp32.blogspot.com/p/supported-hardware.html](https://squeezeesp32.blogspot.com/p/supported-hardware.html  )

sle118, 2022-10-07:
I'll try to include these configurations to the 4.3 version drop down list of known hardware

sle118, 2022-10-23:
Since you have access to these boards, could you share your config for them?

There's the format that the presets uses
https://gist.github.com/sle118/dae585e157b733a639c12dc70f0910c5

I believe you can simply create the config section by exporting your configuration to a file from the NVS menu, and removing all NVS entries that aren't specific to a given board (e.g. only keeping gpio related entries, d'ac control set, etc)

philippe44, 2023-04-02:
Can we close this one?

sle118, 2023-04-02:
I think I'll move that one to discussions, since there seems to be useful info here