sle118/squeezelite-esp32 #187: Implement improv for wifi provisioning

sle118, 2022-10-08:
In order to offer an easier way to provision new devices or to easily switch wifi networks, it would make sense to implement improv. 

#157

sle118, 2022-10-20:
Well well. Bluetooth Improv is definitely not happening in the short term; the extra Bluetooth libraries needed to support that just aren't going to fit in the current partition scheme.

I've made progress on the serial implementation, though

anabolyc, 2024-07-23:
Hey @sle118 . I know this one pretty old. But I was hoping this can be done at some point. This would simplify wifi-provisioning a lot, since many people use web-installer. We [integrated improv](https://github.com/CarlosDerSeher/snapclient/commit/a619587c03ced4284383fef2e6ba2343bb119bb2) in the snaplient and it works like a treat. I was hoping this example might simplify porting the same to the squeezelite-esp32