sle118/squeezelite-esp32 #191: [Feature request] Emit events when playing through any method (including Bluetooth)

Joshfindit, 2022-10-19:
I'm looking for the end goal of being able to automate lighting/smart outlets based on whether the squeezelite is playing audio (device connected but not playing is fine too).

I do have access to the SLIM event subscription so it seems like a handy method. I can also make any type of network API work (push is desired over poll, but whatever)

sle118, 2022-10-20:
Not sure about how this could be implemented. We're pushing the limits of the platform already, and adding yet another socket is going to cost some memory.

If you look at the network traffic between a browser and a player, you'll see that the status message contains two Bluetooth related fields.

Perhaps you could build your automation from two different sources: the Logitech media server for playback originating from there, and the player status JSON for Bluetooth.

https://github.com/sle118/squeezelite-esp32/blob/a2c16decee56126419d46f74e62be60261d8c683/components/wifi-manager/webapp/src/js/custom.js#L467

Joshfindit, 2022-10-20:
I’ll take a look at that network request the next chance I get, but I also wonder:

Could Bluetooth events be emitted to the SLIM server? That way the memory could be saved

sle118, 2022-10-22:
I doubt that the slimproto has any provision for that sort of event, unfortunately. If you think of something specific, let me know

sle118, 2022-10-23:
I'm converting this to a discussion for now