sle118/squeezelite-esp32 #193: Compiling Problems with VScode

Gerhardt-Tooling, 2022-10-21:
Hello for a while i try to compile the squeezelite on my windows10 machine with VS code.
I'm facing some troubles. I'm more the HW guy you have to know and i built my own board i want to run the SW on.
1. Getting VS code with the ESP32 get to run even for the blink example is inherently dificultIt. Tried to install the ESP first then VS code with ESP extension then i tried it vice versa... Does anyone know a good tutorial to do it right?
2. On my laptop (Windows11) i got it to work with Version IDE 4.3 and the blink example by accident. So i tried to compile the squeezelite as it is comming from git. I was facing some errors from some functions in the libs. Which versions of ESP IDE, VSCode and Extensions do i have to use?
Appreciate your advice.
Kind regards Martin

sle118, 2022-10-22:
Is there any specific reason for you to compile the project yourself? There are many specific dependencies needed to build the system and its you're not familiar with the environment, there is a learning curve. We have very limited bandwidth, as this is a hobby, so we can't be of much help here I'm afraid.

You could definitely get some help on other forums and get back here on squeezelite topics.

Four a quick start with your board, you can try the new installer I put online recently here
https://sle118.github.io/squeezelite-esp32-installer/

sle118, 2022-10-22:
For the record, here is how you would start a Docker environment for compilation, after installing Docker... Of course

```
To run the image interactive (windows): 
docker run --rm -v %cd%:/project -w /project -it sle118/squeezelite-esp32-idfv43
To run the image interactive (linux):
docker run --rm -v `pwd`:/project -w /project -it sle118/squeezelite-esp32-idfv43
```
This is for the master-v4.3 branch

closed 2022-10-22 by sle118

Gerhardt-Tooling, 2022-10-23:
Hello sle118, 
thanks for the answer. I read now a bit about docker and set it up as you recommended.
The steps what i did:
Installed docker on windows and pulled with:
`sle118/squeezelite-esp32-idfv43`
Clone the repository in git bash with:
`$ git clone -b master-v4.3 https://github.com/sle118/squeezelite-esp32.git`
Open a CMD consolse in the respective code folder and run
`docker run --rm -v %cd%:/project -w /project -it sle118/squeezelite-esp32-idfv43`
Then i wanted to start the config with:
`idf.py menuconfig`
I get the error:
`root@7277e4485bb6:/project# idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /project"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /opt/esp/idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- Building RECOVERY
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /opt/esp/idf/tools/cmake/build.cmake:191 (message):
  Failed to resolve component 'esp-dsp'.
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/build.cmake:222 (__build_resolve_and_add_req)
  /opt/esp/idf/tools/cmake/build.cmake:223 (__build_expand_requirements)
  /opt/esp/idf/tools/cmake/build.cmake:441 (__build_expand_requirements)
  /opt/esp/idf/tools/cmake/project.cmake:399 (idf_build_process)
  CMakeLists.txt:25 (project)

-- Configuring incomplete, errors occurred!
See also "/project/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1`

What did i wrong?
kind regards Martin.