sle118/squeezelite-esp32 #194: esp32-s3

StuartIanNaylor, 2022-10-22:
Sébastien just curious but with the vector instructions and optimisation of the esp32-s3 have you ever tried any benchmarks on the difference it could make?

sle118, 2022-10-22:
I haven't tried that chip. I still have many wrover modules lying around. I would be surprised if we spent time on that chip either; the S3 does not support classic Bluetooth, which means no Bluetooth audio streaming!

StuartIanNaylor, 2022-10-22:
That is ok if you have not. I am not sure it matters that much as the LMS Server only needs a BT5 dongle not the client. I think its a weird decision by Espressif not to support Classic as via the SIG 5.0 is backward compatible but you don't need it at a client if you get what I mean, but maybe they are edging all their bets on AuraCast broadcast audio and much better audio quality codecs than SBR.

I was just curious with with https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-benchmarks.html if the optimised libs where of any use.

closed 2022-10-22 by StuartIanNaylor