sle118/squeezelite-esp32 #196: Compile error with docker image

Gerhardt-Tooling, 2022-10-26:
Hello sle118,
 I read now a bit about docker and set it up as you recommended.
The steps what i did:

Installed docker on windows and pulled with:
sle118/squeezelite-esp32-idfv43
Clone the repository in git bash with:
$ git clone -b master-v4.3 https://github.com/sle118/squeezelite-esp32.git
Copied the I2S-4MFlash-sdkconfig.defaults file from build-scrips folder to the root folder and renamed it to sdkconfig thus overwriting the existing file.
Open a CMD consolse in the respective code folder and run
docker run --rm -v %cd%:/project -w /project -it sle118/squeezelite-esp32-idfv43
Then i wanted to start the config with:
idf.py menuconfig
I get the error:

`root@7277e4485bb6:/project# idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /project"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /opt/esp/idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- Building RECOVERY
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /opt/esp/idf/tools/cmake/build.cmake:191 (message):
Failed to resolve component 'esp-dsp'.
Call Stack (most recent call first):
/opt/esp/idf/tools/cmake/build.cmake:222 (__build_resolve_and_add_req)
/opt/esp/idf/tools/cmake/build.cmake:223 (__build_expand_requirements)
/opt/esp/idf/tools/cmake/build.cmake:441 (__build_expand_requirements)
/opt/esp/idf/tools/cmake/project.cmake:399 (idf_build_process)
CMakeLists.txt:25 (project)

-- Configuring incomplete, errors occurred!
See also "/project/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1`

What did i wrong? 
kind regards Martin.

sle118, 2022-10-29:
Could you please try to clone recursively?
git clone --recursive https://github.com/sle118/squeezelite-esp32.git

https://github.com/sle118/squeezelite-esp32#building-squeezelite-esp32

Gerhardt-Tooling, 2022-10-30:
Hi sle118,
did the same procedure now making sure i got it recursively.
Now it get this error:
`root@3c8a16142fea:/project# idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /project"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /opt/esp/idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- Building RECOVERY
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /project/sdkconfig
Loading defaults file /project/sdkconfig.defaults...
/project/sdkconfig.defaults:8 CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE was replaced with CONFIG_COMPILER_OPTIMIZATION_SIZE
/project/sdkconfig.defaults:14 CONFIG_TIMER_TASK_STACK_DEPTH was replaced with CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
/project/sdkconfig.defaults:15 CONFIG_USE_ONLY_LWIP_SELECT was replaced with CONFIG_LWIP_USE_ONLY_LWIP_SELECT
/project/sdkconfig.defaults:25 CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST was replaced with CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP
/project/sdkconfig.defaults:528 CONFIG_ESP32_ENABLE_COREDUMP_TO_NONE was replaced with CONFIG_ESP_COREDUMP_ENABLE_TO_NONE
/project/sdkconfig.defaults:785 CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS was replaced with CONFIG_VFS_SEMIHOSTFS_MAX_MOUNT_POINTS
/project/sdkconfig.defaults:786 CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN was replaced with CONFIG_VFS_SEMIHOSTFS_HOST_PATH_MAX_LEN
/project/sdkconfig:136 CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE was replaced with CONFIG_COMPILER_OPTIMIZATION_Swarning: LOGGING_SLIMPROTO (defined at /project/main/Kconfig.projbuild:3) has leading or trailing whitespace in its prompt
warning: LOGGING_STREAM (defined at /project/main/Kconfig.projbuild:8) has leading or trailing whitespace in its prompt
warning: I2S_NUM (defined at /project/main/Kconfig.projbuild:148) has leading or trailing whitespace in its prompt
warning: I2S_BCK_IO (defined at /project/main/Kconfig.projbuild:153) has leading or trailing whitespace in its prompt
warning: I2S_WS_IO (defined at /project/main/Kconfig.projbuild:158) has leading or trailing whitespace in its prompt
warning: I2S_DO_IO (defined at /project/main/Kconfig.projbuild:163) has leading or trailing whitespace in its prompt
warning: I2S_DI_IO (defined at /project/main/Kconfig.projbuild:168) has leading or trailing whitespace in its prompt
warning: I2C_SDA (defined at /project/main/Kconfig.projbuild:175) has leading or trailing whitespace in its prompt
warning: I2C_SCL (defined at /project/main/Kconfig.projbuild:180) has leading or tIZE
/project/sdkconfig:150 CONFIG_ESP32_APPTRACE_DEST_NONE was replaced with CONFIG_APPTRACE_DEST_NONE
/project/sdkconfig:152 CONFIG_ESP32_APPTRACE_LOCK_ENABLE was replaced with CONFIG_APPTRACE_LOCK_ENABLE
/project/sdkconfig:267 CONFIG_ESP32_PANIC_PRINT_REBOOT was replaced with CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT
/project/sdkconfig:361 CONFIG_ESP32_ENABLE_COREDUMP_TO_UART was replaced with CONFIG_ESP_COREDUMP_ENABLE_TO_UART
/project/sdkconfig:363 CONFIG_ESP32_ENABLE_COREDUMP was replaced with CONFIG_ESP_COREDUMP_ENABLE
/project/sdkconfig:364 CONFIG_ESP32_CORE_DUMP_MAX_TASKS_NUM was replaced with CONFIG_ESP_COREDUMP_MAX_TASKS_NUM
/project/sdkconfig:365 CONFIG_ESP32_CORE_DUMP_UART_DELAY was replaced with CONFIG_ESP_COREDUMP_UART_DELAY
/project/sdkconfig:640 CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS was replaced with CONFIG_VFS_SEMIHOSTFS_MAX_MOUNT_POINTS
/project/sdkconfig:641 CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN was replaced with CONFIG_VFS_SEMIHOSTFS_HOST_PATH_MAX_LEN
railing whitespace in its prompt
warning: A2DP_CONTROL_DELAY_MS (defined at /project/main/Kconfig.projbuild:232) has leading or trailing whitespace in its prompt
warning: style: quotes recommended around default value for string symbol AIRPLAY_PORT (defined at /project/main/Kconfig.projbuild:270)
/tmp/confgen_tmpxmlttjof:42: warning: ignoring malformed line 'CONFIG_SQUEEZELITE_ESP32_RELEASE_URL "https://github.com/sle118/squeezelite-esp32/releases"'
[... cut ...]
-- Adding new build target (from build folder): ninja size-components-squeezelite
-- Adding new build target (from build folder): ninja squeezelite-flash
CMake Error at squeezelite.cmake:7 (esptool_py_flash_project_args):
  Unknown CMake command "esptool_py_flash_project_args".
Call Stack (most recent call first):
  squeezelite.cmake:124 (___register_flash)
  CMakeLists.txt:11 (include)

-- Configuring incomplete, errors occurred!
See also "/project/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1`

Any hints?

sle118, 2022-10-31:
Yes. Please checkout branch master-v4.3, as it is the one being prepared for being the default branch some time soon

Gerhardt-Tooling, 2022-10-31:
Hi sle118,

it got the build!
So the procedure is:

Installed docker on windows and pulled with:
`sle118/squeezelite-esp32-idfv43`
Clone the repository in git bash with:
`$ git clone --recursive -b master-v4.3 https://github.com/sle118/squeezelite-esp32.git`
Copied the I2S-4MFlash-sdkconfig.defaults file from build-scrips folder to the root folder and renamed it to sdkconfig thus overwriting the existing file.
Open a CMD consolse in the respective code folder and run
`docker run --rm -v %cd%:/project -w /project -it sle118/squeezelite-esp32-idfv43`
Then start the config with:
`idf.py menuconfig`
Then i can build with:
`idf.py build`

**The last thing missing is flashing the device. This does not work out of the Docker container on windows. 
Do you have a hint for that?**

PS: Why am i doing this? I want to flash several devices so i want a preconfigured flashable file. The way via the online installer worked only half way. The online installer did not install the version "I2S" which i selected, it installed the "MUSE". Then i tried to update it within the muse system, which gave me a https error. So i got back to build it my own....

sle118, 2022-11-01:
Could you please tell me which web installer you have tried and didn't install the right firmware? 

If this is mine, then I'd like to know so I can take the necessary steps to correct.

Also, I'd be curious to know about the HTTPS errors you got. Perhaps you could log an issue for that so I can see if I can reproduce.

By the way, I'm working on implementing a wifi provisioning solution that will work hand in hand with the web installer (the solution is improv-wifi). That should help with the initial config for several devices

Thank you!

sle118, 2022-11-01:
I have used our new installer (see link below) and choosing the I2S build worked perfectly well.

https://sle118.github.io/squeezelite-esp32-installer/

sle118, 2022-11-02:
> **The last thing missing is flashing the device. This does not work out of the Docker container on windows. Do you have a hint for that?**
> 
I will see if I can include a button for a local build in the web flasher. If that would, then you could load a build folder and flash it from the web.

Gerhardt-Tooling, 2022-11-02:
Hi sle118,
the programmer which installed the wrong version was this one:
https://paulusschoutsen.nl/squeezelite-esp32-install/
I tried yours and it works.
Also updating via the webinterface works. I cant reproduce the issue I had some days ago.

To flash the build i tried the ESP32 flash tool. However i dont know how to set it up correctly.
[image]

I will try your link in the post before on the weekend. I will post if it works.
To flash a custom folder via the online installer is also a nice idea.

sle118, 2022-11-03:
You can find the list of files to flash as well as their address in any of the build artifacts
https://github.com/sle118/squeezelite-esp32/suites/8906681937/artifacts/407743503

Where you will find a couple of files with that info. Can't remember the names, and I'm writing from a small screen so not able to check that right now.

You can also see the addresses in one of the web installer manifest
https://github.com/sle118/squeezelite-esp32-installer/blob/main/docs/artifacts/manifest-master-v4.3-I2S-4MFlash-2.1186.json

sle118, 2022-11-12:
Stalled topic, closing

closed 2022-11-12 by sle118