sle118/squeezelite-esp32 #198: ES8388 stereo downmix (register 29)

pmannk, 2022-11-18:
Hi,
Firstly, thanks for this great project. Absolutely love it!

Could you please consider exposing an option to control Register 29 on the ES8388?
The datasheet indicates register 29, bit 5 can control the stereo/mono behaviour of the DAC.
[image]
If I'm understanding the data sheet correctly this looks like a great way to downmix a mono output, in hardware.

[Linking this old issue for reference in case anyone searches for it.](https://github.com/sle118/squeezelite-esp32/issues/36)
Thankyou.

sle118, 2022-11-18:
You should try to configure speakeron event of dac_controlset

https://github.com/sle118/squeezelite-esp32#daci2s

pmannk, 2022-11-23:
Thanks. Have had a couple of unsuccessful attempts. Will keep trying to work it out.

sle118, 2022-11-25:
For now, I'm concentrating on some stabilizing of release 4.3 and, given we're already short on flash space, is unlikely that we'll be able to implement new features anytime soon.

Hopefully the events should accommodate this. I'll convert this to a conversation instead of an issue