sle118/squeezelite-esp32 #207: 1231 and 1245 won't connect to BT speaker

daduke, 2022-11-26:
Hi,

I use BT output to Divoom speakers [1] (yes I know it's not the preferred output, but it's what they have and it's always been working great). With 1231 (and 1234), squeezelite won't connect to those speakers any more. 1230 still works. Hardware: a naked WROVER module.
Log output at the end. Let me know if I can test anything else.

thanks,
-Christian

[1] https://www.divoom-gz.com/product/tivoo.html

log:
```
I (1875) esp_app_main: Setting up config subsystem.
W (1945) nvs_utilities: Configuration memory usage.  Heap internal:224535 (min:224407) (used:200) external:4058503 (min:4051051) (used:12916)
I (1945) esp_app_main: Registering default values
I (1945) system_api: Base MAC address is not set
I (1945) system_api: read default base MAC address from EFUSE
W (1955) config: Waiting for config commit ...
I (2945) config: configuration has some uncommitted entries
I (2945) config: Committing configuration to nvs. Locking config object.
I (2945) config: Done Committing configuration to nvs.
I (2945) config: Config committed!
I (2945) esp_app_main: Configuring services
I (2945) services: Configuring I2C sda:-1 scl:-1 port:1 speed:400000
W (2945) services: no I2C configured
I (2955) services: Configuring SPI mosi:-1 miso:-1 clk:-1 host:1 dc:-1
W (2955) services: no SPI configured
W (2955) led: LED GPIO -1 ignored
W (2955) led: LED GPIO -1 ignored
I (2955) led: Configuring LEDs green:-1 (active:0 -1%), red:-1 (active:0 -1%)
I (2955) battery: No battery
I (2955) monitor: Initializing monitoring
I (2955) monitor: Heap internal:225327 (min:224315) external:4042023 (min:4042003) dma:217131 (min:216131)
I (2955) esp_app_main: Initializing display
I (2955) display: Trying to configure display with N/A
W (2955) display: No display driver
I (2955) esp_app_main: Checking if certificates need to be updated
I (2955) esp_app_main: About to check if certificates need to be updated in flash
I (2955) esp_app_main: Running partition [ota_0] type 0 subtype 16 (offset 0x00150000)
I (2955) esp_app_main: Running version: local.500.cmake-master
I (2955) esp_app_main: Certificate version: I2S-4MFlash.16.2.1230.master-v4
W (2955) esp_app_main: Found a different software version. Updating certificates
I (3005) messaging: HTTPS Certificates were updated with version: local.500.cmake-master
I (3005) esp_app_main: Checking for update url
W (3005) config: Value not found for key fwurl
W (3005) config: Value not found for key
I (3005) esp_app_main: Starting Network Manager
I (3005) uart: queue free spaces: 3
W (3005) config: Value not found for key apdelay
I (3005) network_handlers: Network manager configuration: polling max 10176, polling min 15000, ap delay 20000, dhcp timeout 8000, eth timeout 8000
W (3005) network_wifi: No known access point found
I (3005) network_wifi: Adding legacy/active wifi connection to the known list
I (3005) http_server: Initializing HTTP Server
I (3015) network_ethernet: Attempting to initialize Ethernet
I (3015) network_ethernet: No Ethernet configuration, or configuration invalid
I (3025) wifi:wifi driver task: 3ffdfd04, prio:23, stack:6656, core=0
I (3035) wifi:wifi firmware version: eb52264
I (3035) wifi:wifi certification version: v7.0
I (3035) wifi:config NVS flash: enabled
I (3035) wifi:config nano formating: disabled
I (3045) wifi:Init data frame dynamic rx buffer num: 40
I (3045) wifi:Init management frame dynamic rx buffer num: 40
I (3045) wifi:Init management short buffer num: 32
I (3045) wifi:Init static tx buffer num: 12
I (3045) wifi:Init tx cache buffer num: 32
I (3045) wifi:Init static rx buffer size: 1600
I (3045) wifi:Init static rx buffer num: 12
I (3045) wifi:Init dynamic rx buffer num: 40
I (3045) wifi_init: tcpip mbox: 32
I (3045) wifi_init: udp mbox: 32
[... 78 lines cut ...]

[00:08:03.035466] handler:54 got equalizer 14
[00:08:03.127968] stream_thread:264 headers: len: 491
HTTP/1.1 200 OK
Server: Logitech Media Server (8.3.1 - 1669201751)
Connection: close
Content-Type: audio/mpeg
Set-Cookie: Squeezebox-albumView=; path=/
Set-Cookie: Squeezebox-expandPlayerControl=true; path=/
Set-Cookie: Squeezebox-expanded-MY_MUSIC=1; path=/
Set-Cookie: Squeezebox-expanded-FAVORITES=1; path=/
Set-Cookie: Squeezebox-expanded-PLUGINS=0; path=/
Set-Cookie: Squeezebox-expanded-PLUGIN_MY_APPS_MODULE_NAME=1; path=/
Set-Cookie: Squeezebox-expanded-RADIO=0; path=/

[00:08:03.163256] mad_decode:247 setting track_start
[00:08:03.163943] resample_newstream:125 disable resampling - rates match
[00:08:03.164101] process_newstream:124 processing: inactive
I (22915) httpd_handlers: serving /status.json to peer 192.168.2.4 port 48278
I (115845) httpd_handlers: serving /status.json to peer 192.168.2.4 port 49891
I (115865) httpd_handlers: serving /config.json to peer 192.168.2.4 port 43235
I (117525) network_wifi: Initiating wifi network scan
I (120015) httpd_handlers: serving /messages.json to peer 192.168.2.4 port 47331
I (151295) telnet: We have a new client connection 57
```

wizmo2, 2022-11-26:
@sle118, I'll look at adding a switch for the legacy pairing support

daduke, 2022-11-26:
do I have to pair in a different way?

wizmo2, 2022-11-26:
I had issues with BT recently, and added a PR with a couple of improvements that got merged with 1234.  I cant say its very stable, as I still have to reboot / repair

sle118, 2022-11-26:
We were in the readme that we wouldn't support Bluetooth output anymore. But that doesn't apply to @daduke 🥇 

I will try to take a look and make sure we align with the latest espressif demo. As for UI option, that is a possibility and it's actually done via the configuration commands of the console, which expose all their metadata to the UI.

daduke, 2022-11-26:
> We were in the readme that we wouldn't support Bluetooth output anymore. But that doesn't apply to @daduke 1st_place_medal
awwwwwwwwwww 😍

don't wreck yourself over it. I can stay on 1230.

wizmo2, 2022-11-26:
@sle118 I think there's a sdkconfig change need on the Bluetooth controller configuration  for BTDM mode  (8462de1c1665da9d8316831c30ab546a224da785)

sle118, 2022-11-27:
@wizmo2 could you be more specific please? I haven't had a chance to dig much into this today

wizmo2, 2022-11-27:
[image]

wizmo2, 2022-11-27:
@philippe44 changed the  bbt controller mode from ESP_BT_MODE_CLASSIC_BT to ESP_BT_MODE_BTDM.  The latter requires the IDF option above.  

Haven't looked at how you create your images, but I guess you need to update all the build-scripts.

sle118, 2022-11-27:
I will look into this today. hoping this doesn't inflate the binaries too much, as we're really short on Flash right now

philippe44, 2022-11-27:
I did not change anything - that was somebody's else PR

wizmo2, 2022-11-27:
> I did not change anything - that was somebody's else PR

Apologies!

philippe44, 2022-11-27:
There is like 15kB (of flash!) or less left and it took me extraordinary measures to make cspot fit so this might be challenging

sle118, 2022-11-27:
Too bad it's not simple to create dynamic linked libraries. We could compile out some features and then download/run them on demand.
I think I just had an idea. @philippe44  I'll ping you later

wizmo2, 2022-11-29:
I've made a some progress on bt source capabilities, 
1.  I'm not convinced of the need for BTDM.  If we are not going to use BLE, then I don't think its needed.  Saying that, it does not look like there is much of an overhead in flash or memory use when using BTDM without actually implementing BLE.
2. I have code that is able to successfully allow a device to reconnect at boot. This actually helps significantly, as it makes bt output a lot more practical, plus some of my devices connect with choppy sound on discovery connection, but then is fine when reconnecting from a ESP reboot. 

As part of the investigation, I have made a lot of changes to optimize the code (mostly to get a grip of whats going on).  So I'm not sure if its ready for a PR.  It's also raised more questions than answers, making the project a little more intriguing!.  For example, I  have a couple of JBL devices that are always choppy.  I vaguely remember there are some bit-rate configurations in Squeezelite core that may help, but would think that there is probably a way to make this automatic. 
  
@sle118 and @philippe44 Do you think I should submit a draft PR as-is, or do I just keep at it for a while?

philippe44, 2022-11-29:
You should submit a PR if you want our opinions :smile:

philippe44, 2022-11-29:
re BTDM, I finally got it so yes @sle118 I think that part in https://github.com/sle118/squeezelite-esp32/commit/8462de1c1665da9d8316831c30ab546a224da785 is a typo that should be removed.

philippe44, 2022-11-29:
I've pushed some Spotify changes at brought rollback the BTDM change (just the one variable at init)

sle118, 2022-11-30:
@daduke could you please confirm that BT Source works now for you?

wizmo2, 2022-12-04:
I think this is now closed with #210 ?  

I'll push a draft PR with an optimized bt_source hopefully before EOY.  Have made a number of improvements, but want to work on a few compatibility issues.

@daduke, but now I've have  Divoom timebox-evo and now need to integrate bt path-through and HA integration for the built in display!

daduke, 2022-12-05:
> @daduke, but now I've have Divoom timebox-evo and now need to integrate bt path-through and HA integration for the built in display!

nice. The displays of the Tivoo and Tivoo Max are not readily accessible (one would have to reverse engineer the BT protocol I guess)

sle118, 2022-12-05:
I'll go ahead and close this one then.

closed 2022-12-05 by sle118