sle118/squeezelite-esp32 #208: Feature Request: Show battery voltage level on screen during boot

heinerolli, 2022-11-27:
I see in the serial log an entry "battery: Voltage = ...."
Is that something that you could show i.e. during boot on an attached display?
(i.e while "SqueezeAmp" is showing on the display)
That would help to get an idea of the current battery charging level

philippe44, 2022-12-05:
The battery level is available on LMS, in the player's information

heinerolli, 2022-12-05:
Thanks, for the hint, philippe44. I use that as well - but still think to have this information right on the attatched screen during boot up would be beneficial.
At least in my usecase: Go to the player, switch it on and check battery voltage ( -> charge level) . 
One stop shopping; no additional buttons to press, apps to check etc.
As the batteries are good for several hours music streaming at moderate volume setting, checking battery level once atmboot up would be sufficient.
As always: Thanks to both you guys fordoing this project! Such a great firmware - such  a great player (SqueezeAmp)

wizmo2, 2023-08-28:
FYI, I have a working modification of MusicInfoSCR with wifi and battery icons.  I've asked @mherger if he is willing to add the changes to his original, or if I need create a separate 
[image]

mherger, 2023-08-28:
That's a lot of information around that artwork 😀!

wizmo2, 2023-09-04:
MusicInfoSCR 4.5 is now released (Thank you @mherger).

For those that are not aware, Michaels' excellent Music Information Screen LMS plugin allows for customization of the Now Playing screen with extra information.  The latest version adds icons for Hardware Status to display the **Wifi signal strength and battery voltage**.  

[image]

There are two battery scales that are selected automatically;  A standard single cell Lipo (3.3-3.9), and a 3-cell NiCad (11.4-12.6) tested on my SqueezeAmp +3*18650.  _NOTE: The latter is configured to show empty at 11.4V with @philippe44 standard settings, which is a long way from processor failure, but at the point when the amplifier quality is affected. You can use the NVS parameter "bat_config' `scale` parameter to tweak the reported voltage if required._ 

@heinerolli et-al. Please give it a go and provide feedback where necessary.

P.S. @sle118 do you think we should add something to the README?

heinerolli, 2023-09-04:
Thanks!
I am currently away from home for a couple of weeks, but when I am back, I will try it
Thanks again,
Heiner

[image]<https://user-images.githubusercontent.com/11602094/265454791-b234855d-042c-4b8f-966a-6db60a311643.png>

sciurius, 2023-09-09:
I don't see the battery level... What am I missing?

[image]

philippe44, 2023-09-09:
You need to use music info scr plugin and enable it in player's settings (basic or display o don't remember)

sciurius, 2023-09-09:
It is not in "Basic" and apparently the squeezeesp32 does not have "Display".

sciurius, 2023-09-09:
When I issue a "status" via the CLI it doesn't seem to report the battery status. Maybe I need some settings on the board itself but I can't find any info.

philippe44, 2023-09-09:
oh, that only works if you have a display so I guess this is not what you want

wizmo2, 2023-09-09:
Do you have your battery configured in the NVS settings?
(https://github.com/sle118/squeezelite-esp32#battery--adc)

as @philippe44 mentions, you then need to have the plugin installed and configured in the Basic settings
[image]

Of course, you do need a Display installed and configured to see it!

sciurius, 2023-09-09:
I've seen that link but I have no clue what to enter in the `bat_config`. It needs a channel (no idea), a scale (?) and the choice between two and three cells. I have a single-cell LiPo..

wizmo2, 2023-09-09:
You need to check your hardware schematic and see if there is a battery voltage sensor input.

philippe44, 2023-09-09:
the README https://github.com/sle118/squeezelite-esp32#battery--adc is your friend

sciurius, 2023-09-09:
@wizmo2 I have an ESP32 Audio Kit 
[image]

wizmo2, 2023-09-09:
You maybe out of luck.  I was using one of these and I don't think VBAT is connected to any inputs.  You could double check on the [forum](https://forums.slimdevices.com/forum/user-forums/3rd-party-hardware/109412-announce-squeezelite-esp32-dedicated-thread) and see if anyone else has figured it out.  These a few different versions of these boards

philippe44, 2023-09-09:
Yeah that board is a mess. Look at schematics but if in the example of the README I've not included the battery information it's probably because there is no connection. You'd have to jump one

sciurius, 2023-09-09:
When I set bat_config to `channel=2,scale=1,cells=1` I get readings in the log file of around 0.10V. It's a start ☺.

wizmo2, 2023-09-09:
(https://docs.ai-thinker.com/_media/esp32-audio-kit_v2.2_sch.pdf)
have a look at the schematic above.  IO34/Channel 6 may be a good start. 
[image]
Why we call these DIY projects ;)  Good luck

sciurius, 2023-09-10:
Thanks for the suggestions. This is really not my field of expertise...

With channel 6 I only get `1`.

BTWE I'm surprised to find out it is still running after 12 hours. I would have expected just a couple of hours on such a small battery.

sciurius, 2023-09-11:
After studying the schemata et al a couple of conclusions (so far).

* Squeezelite-ESP32 can monitor battery on ADC1 only.
* The chip on the board seems to have three channels on ADC1: Channel 0 (GPIO36), 3 (GPIO39) and 6 (GPIO34).
GPIO39 (SEN_VN) is used for headphone detection, GPIO36 and GPIO34 are not connected according to the schema but **they are connected** on the board. I can't trace the connection since it disappears under the chip. And, unfortunately, neither of them is exposed on a header.

I have on intentions to solder wires directly to the chip so unless anyone comes up with new ideas it stops here.

sle118, 2023-09-11:
The ADC channel limitation isn't specific to our project; it is a limitation of the ESP32 itself, which requires one of the ADC channel to be allocated to the wifi circuitry when in use.  As for the Audio Kit, you are out of luck unless you are willing to solder 2 resistor in a voltage divider arrangement since the design of the board doesn't include one for measuring voltage. Setting up the battery nvs option here will result in measuring from a floating pin or from a pin that is pulled up/down on the circuit board itself.

sciurius, 2023-09-11:
Thanks for the clear explanation. 
I've added a voltage divider ( 2 x 100K ) to BAT+ and BAT-, and connected the mid point to pin5 GPIO34. After configuring bat_config to `channel=6` I get solid `1` reading.
I tried pin4 GPIO36 and channel 0, same result - a solid `1` reading.

According to the specs the input for ADC1 should be between 0 and 3.3V. Dividing the battery voltage by two should be safe.

Do you know what I have to enter for `cells`?

philippe44, 2023-09-11:
It is used to convert voltage in % (roughly). So it's the number of LiIon cells in serie this board has, which is one I assume.

sciurius, 2023-09-11:
It's a small battery so I assume 1 cell (3.7V). But the docs say the mandatory and the only choices are 2 and 3.

    channel=0..7,scale=<scale>,cells=<2|3>[,atten=<0|1|2|3>]

philippe44, 2023-09-11:
Indeed, it could be 1. I'll correct that. Bear in mind that battery % evaluation is not great, to say the leasy

sle118, 2023-09-12:
> Indeed, it could be 1. I'll correct that. Bear in mind that battery % evaluation is not great, to say the least

and this is mainly due to the voltage discharge curve, which isn't constant from battery to battery. 
[image]

The only precise way to measure battery percentage is to have a coulomb counter that measures the charge in and charge out. Measuring voltage is a "good enough" proxy but far from being ideal.

sciurius, 2023-09-12:
Reminds me of my old [Psion 3a](https://en.wikipedia.org/wiki/Psion_Series_3). It did not measure the batteries voltage, but kept track of the amount of energy used.

Currently I would already be glad if I had *any* reading... It stays solid `1` (same as if not connected). Strange.

philippe44, 2023-09-12:
How much do you have on that input using a voltmeter? Reading a 1 would seem to indicate that it is still configured a a binary gpio.

sciurius, 2023-09-12:
Interesting. Between the resistors of the divider (each 100K) I measure approx 1.8V.
Pin 4 and pin 5 on the ESP both measure approx 3.4V (Vcc, I assume). Connecting the mid point of the divider to either pin does not change that.
Maybe the pins are pulled up harder than my divider? They are not supposed to have an internal pull-up/down.

philippe44, 2023-09-12:
Are you sure that in your config, nothing uses io34 as a gpio?
 And so has configured as such?

philippe44, 2023-09-12:
Did you check you have a v2.2 of the board? Even there, these are full of crap and publish wrong schematics.

wizmo2, 2023-09-12:
Do you have the attenuation configured?

_ADC sampling voltage attenuation configuration. Modification of attenuation affects the range of measurements. 0: measurement range 0 - 800mV, 1: measurement range 0 - 1100mV, 2: measurement range 0 - 1350mV, 3: measurement range 0 - 2600mV._

sciurius, 2023-09-12:
> Are you sure that in your config, nothing uses io34 as a gpio? And so has configured as such?

[image]

sciurius, 2023-09-12:
> Did you check you have a v2.2 of the board? Even there, these are full of crap and publish wrong schematics.

[image]

philippe44, 2023-09-12:
But then you should be able to st worse remove the pull-up of sd-detect. Still, that if you choose low enough (not too low of course ...) for the divisor, you should be able to make than "win" against the pull-up and integrate its influence.

sciurius, 2023-09-12:
I'll probably consider to desolder pin4 or pin5 using the 'thin wire' approach. I'm not sure yet.