sle118/squeezelite-esp32 #209: How to build Squeezelite for ESP32-S3 ?

RASPIAUDIO, 2022-11-30:
It would be interesting for us to run squeezelite on S3 (for the cost among others...)
But we are faced with a problem:
Squeezelite compiles well only with IDF4.0 but this version of IDF does not support S3...
Does anyone have any thoughts on this?...
Thanks!

sle118, 2022-12-02:
adding up to this. I tried compiling, hoping it would be as easy as setting the target with `idf.py set-target esp32s3` but no. Support will require some extensive work to tweak the sdkconfig as well as to compile out some bluetooth specific code, UI elements, etc. Then the build system will have to be tweaked to put out some more targets. I am likely to combine the I2S-4MBFlash (generic) target release to generate 2 builds, one for the ESP32 and one for the ESP32-S3.  The Web installer will be able to choose the right set of binaries based on the device processor signature. 

But I (and probably @philippe44)  don't have much bandwidth or desire at this time to undertake this sort of thing, which is likely going to be a rabbit hole, so I'll go ahead and close this for now

RASPIAUDIO, 2022-12-02:
Our motivation to go with the S3 is to avoid buying those USB to serial chipset (like the CP2104) that is becoming as expensive as the ESP32 module, as the S3 could emulate this USB to serial profile natively.

But indeed it does not make sense if classic A2DP Bluetooth audio is not supported. As an alternative I'm in contact with Espressif to understand if the Bluetooth 5.0 in the S3 could support the new BLE audio profiles that could improve quality. I will let you know if this is possible when I get an answer.

Conditional on the above validation we could try to help to integrate S3 in Squeezelite.

Thanks for your amazing work!

philippe44, 2022-12-03:
I've compiled for S3 last year and it ran (with a lot of handcrafted stuff) but as @sle118 said, the lack of BT classic stopped the effort.

dlbogdan, 2023-04-18:
philippe44, can I have a look at your handcrafted stuff to make it compile on the S3 (do you have a test branch somewhere?)
A2DP BT sink is not a necessity for a lot of users, given that an alternative exist (such as Airplay). 
I wish to investigate some potential performance improvements with the S3.

Many thanks!

philippe44, 2023-05-13:
I now have a successful compilation on esp32s3. Need to see how it works :smile:

philippe44, 2023-05-15:
It seems that the load on CPUs is around half of what it is on esp32, which an octal SPIRAM but still a quad Flash.

dlbogdan, 2023-05-16:
I’m very interested in your code. I could even help clean it up if you want.

philippe44, 2023-05-16:
I'm just fixing a strange issue with cspot and I'll do a branch

philippe44, 2023-05-16:
The branch is here https://github.com/sle118/squeezelite-esp32/tree/4.4-s3

Now, this is highly experimental 😄 and you need to

- use esp-idf 4.4 
- for once, I've committed the sdkconfig which is certainly not well checked but is likely needed
- delete _override repositoy
- delete (just to be sure) driver_bt repository

... and probably other things that I'm forgetting to mention

sle118, 2023-05-16:
We should probably keep the driver_bt component and only compile and link for targets that support this. That way we can keep our build system common to all target chips.

philippe44, 2023-05-17:
Yes I know but for now this is just a test. The intent is to have a common set of sources st the end

dlbogdan, 2023-05-17:
Can’t wait to test this. It’s like christmas!

philippe44, 2023-05-17:
You can try it from my branch directly

dlbogdan, 2023-05-17:
I will. Just as soon as I’m done with meetings and get my babies to sleep. Thx!

FHFS, 2023-06-05:
I was able to build for esp32s3, and it runs! Thanks!

closed 2023-06-07 by sle118

przemeklagod, 2024-01-02:
@philippe44 I compiled Your branch but got CPU halt after programming: 

`
I (3456) network_wifi: Initiating wifi network scan
I (3456) dns_server: DNS Server listening on 53/udp
****************************************************************
RECOVERY APPLICATION
This mode is used to flash Squeezelite into the OTA partition
****

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

assert failed: prvNotifyQueueSetContainer queue.c:3092 (pxQueueSetContainer->uxM                                                     essagesWaiting < pxQueueSetContainer->uxLength)

Backtrace: 0x4037623a:0x3fc98660 0x40380375:0x3fc98680 0x40385a75:0x3fc986a0 0x4                                                     0380c9e:0x3fc987c0 0x40380e6b:0x3fc987f0 0x4202ed89:0x3fc98820 0x403774fd:0x3fc9                                                     8880 0x420b8e17:0x3fcf5230 0x42006025:0x3fcf5250 0x420bf6a3:0x3fcf5270

ELF file SHA256: eca75eb2618f55c2

CPU halted.

`
What can be wrong ?

philippe44, 2024-01-03:
What do you mean my branch?

przemeklagod, 2024-01-03:
Sorry I mean 4.4-s3. I'm trying to run it on https://espressif-docs.readthedocs-hosted.com/projects/esp-adf/en/latest/design-guide/dev-boards/user-guide-esp32-s3-korvo-2.html Can It be related to memory ? I have N16R8 version (16MB Flash 8MB PSRAM)

przemeklagod, 2024-01-03:
Can You help me ? I tried to compile it with idf 4.4.6 and 4.4.4 both has same issue.

philippe44, 2024-01-03:
I really don't know. It's probably something wrong with your config file. If you look at the backtrace, try to find where, in squeezelite.map, is the crash. You have a set of pairs of addresses like 0x420b8e17:0x3fcf5230

Search for 420b8e17 if not found 
search for 420b8e1, then
search for 420b8e

You can use each backtrace elements to find where the crash really happened

przemeklagod, 2024-01-03:
I'm using your branch and config without any changes :-( can I ask You for compiled files just to be sure that it is not related to mg environment/board ?

philippe44, 2024-01-04:
You have to change some config items. Are you sure you have the proper memory config in term of flash bus length (quad-SPI) and SPIRAM bus length (Octal SPI). There are many versions: https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf

chagoguila, 2024-01-09:
Hi everyone, this project is awesome, I have already a wrover working, but I received a s3 cam (the one with ble, 2x usb c and mic) by error. Would like to try and make squeezelite work on this but haven't been able to build correctly. Does any one have a build for this board?
[image]

philippe44, 2024-01-09:
Does it have spiram

chagoguila, 2024-01-09:
> Does it have spiram
Yes it does (16mb flash 8mb psram). Here is the link:
https://es.aliexpress.com/item/1005006099376239.html?spm=a2g0o.productlist.main.1.66545edeYndgQf&algo_pvid=39a937b9-195a-479d-9904-b1ccbd4796b5&algo_exp_id=39a937b9-195a-479d-9904-b1ccbd4796b5-0&pdp_npi=4%40dis%21CHF%2112.46%2111.71%21%21%2114.44%21%21%40210321dc17047929321584071eab56%2112000035736382712%21sea%21CH%21811199566%21&curPageLogUid=ioXPsH4Qm6iq&utparam-url=scene%3Asearch%7Cquery_from%3A

chagoguila, 2024-01-09:
I was able to advance a little bit, but got stuck with an error regarding spotify nanopb when building:
FAILED: esp-idf/spotify/cspot/nanopb/generator/proto/nanopb_pb2.py

chagoguila, 2024-01-10:
Any chance you could help me guys?

philippe44, 2024-01-10:
I think mentioned in the README the additions you need to install to compile protobuf. If not, look at what is recommended in cspot

chagoguila, 2024-01-10:
Ok, shall check it again, thanks a lot...looking at other closed issues I found some stuff regarding proto, hope I can make it work.

chagoguila, 2024-01-11:
So, got idf to run ok, can flash examples, but I have pinpointed my problem:
If I build with master 4.3 without changing anything, it build ok.
If I set-target esp32s3, it gives me error: FAILED: esp-idf/spotify/cspot/CMakeFiles/cspot.dir/src/TrackQueue.cpp.obj
And if I use the sdkconfig from 4.4-S3: FAILED: esp-idf/tools/CMakeFiles/__idf_tools.dir/tools.c.obj

philippe44, 2024-01-11:
What is the compiler error?

chagoguila, 2024-01-11:
CMake version 3.23.1
I'm using ESP-IDF 4.4.6
When using 4.4-s3 sdkconfig:
C:\Espressif\frameworks\esp-idf-v4.4.6\squeezelite-esp32>idf.py all
Executing action: all (aliases: build)
Running cmake in directory C:\Espressif\frameworks\esp-idf-v4.4.6\squeezelite-esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 C:\Espressif\frameworks\esp-idf-v4.4.6\squeezelite-esp32"...
-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- ==> NO OVERRIDE <==
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/sdkconfig
Loading defaults file C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/sdkconfig.defaults...
C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/sdkconfig.defaults:622 CONFIG_ESP32_REDUCE_PHY_TX_POWER was replaced with CONFIG_ESP_PHY_REDUCE_TX_POWER
C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/sdkconfig.defaults:822 CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE was replaced with CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE
C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/sdkconfig.defaults:824 CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER was replaced with CONFIG_ESP_PHY_MAX_WIFI_TX_POWER
C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/sdkconfig.defaults:825 CONFIG_ESP32_PHY_MAX_TX_POWER was replaced with CONFIG_ESP_PHY_MAX_TX_POWER
-- Found Python3: C:/Espressif/python_env/idf4.4_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "recovery" version: local.500.cmake-master
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib-time.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Websocket transport is disabled so the esp_websocket_client component will not be built
-- Bell options:
--     Disable all codecs: ON
--     Disable built-in audio sinks: ON
--     Use Vorbis float version: OFF
--     Use cJSON only: ON
--     Disable Fmt: ON
--     Disable Mqtt: ON
--     Disable Regex: ON
--     Disable Web server: ON
-- Found Nanopb: C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/components/spotify/cspot/bell/external/nanopb

C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/components/spotify/cspot/bell/external/nanopb
-- Using external Vorbis codec idf::codecs
******************* local.500.cmake-master
******************* local.500.cmake-master
-- Components: app_recovery app_squeezelite app_trace app_update asio audio bootloader bootloader_support bt cbor cmock coap codecs console cxx display driver efuse esp-dsp esp-tls esp32s3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json led_strip libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon platform_config platform_console protobuf-c protocomm pthread raop sdmmc services soc spi_flash spiffs spotify squeezelite squeezelite-ota targets tcp_transport tcpip_adapter telnet tinyusb tjpgd tools ulp unity usb vfs wear_levelling wifi-manager wifi_provisioning wpa_supplicant xtensa
[... cut ...]
[106/106] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.6\squeezelite-esp32\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.6/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/build/bootloader/bootloader.bin"
Bootloader binary size 0x5820 bytes. 0x27e0 bytes (31%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

chagoguila, 2024-01-11:
:And when seting target to esp32-s3:
C:\Espressif\frameworks\esp-idf-v4.4.6\squeezelite-esp32>idf.py build
Executing action: all (aliases: build)
Running ninja in directory C:\Espressif\frameworks\esp-idf-v4.4.6\squeezelite-esp32\build
Executing "ninja all"...
[23/1352] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-id...idf-v4.4.6/squeezelite-esp32/generate_debug_scripts.cmake"
Generating debug script files
[69/1352] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[672/1352] Running C++ protocol buffer compiler using nano...p32/components/spotify/cspot/protobuf/authentication.proto
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
W0000 00:00:1705010547.034078    1252 parser.cc:678] No syntax specified for the proto file: protobuf/authentication.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[673/1352] Running C++ protocol buffer compiler using nano...-esp32/components/spotify/cspot/protobuf/keyexchange.proto
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
W0000 00:00:1705010547.066628   21160 parser.cc:678] No syntax specified for the proto file: protobuf/keyexchange.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[675/1352] Running C++ protocol buffer compiler using nano...elite-esp32/components/spotify/cspot/protobuf/login5.proto
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
W0000 00:00:1705010547.091505   22508 parser.cc:678] No syntax specified for the proto file: protobuf/login5.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[676/1352] Running C++ protocol buffer compiler using nano...zelite-esp32/components/spotify/cspot/protobuf/spirc.proto
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
W0000 00:00:1705010547.085280    9416 parser.cc:678] No syntax specified for the proto file: protobuf/spirc.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[678/1352] Running C++ protocol buffer compiler using nano...lite-esp32/components/spotify/cspot/protobuf/mercury.proto
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
W0000 00:00:1705010547.107495   21276 parser.cc:678] No syntax specified for the proto file: protobuf/mercury.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[736/1352] Performing configure step for 'bootloader'
-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/sdkconfig
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.6/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32s3 esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v4.4.6/components/bootloader C:/Espressif/frameworks/esp-idf-v4.4.6/components/bootloader_support C:/Espressif/frameworks/esp-idf-v4.4.6/components/efuse C:/Espressif/frameworks/esp-idf-v4.4.6/components/esp32s3 C:/Espressif/frameworks [... cut ...]
C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/components/spotify/cspot/bell/main/io/include/URLParser.h: In static member function 'static bell::URLParser bell::URLParser::parse(const string&)':
C:/Espressif/frameworks/esp-idf-v4.4.6/squeezelite-esp32/components/spotify/cspot/bell/main/io/include/URLParser.h:80:48: error: exception handling disabled, use -fexceptions to enable
       throw std::invalid_argument("Invalid URL");
                                                ^
[1159/1352] Building CXX object esp-idf/spotify/cspot/CMakeFiles/cspot.dir/src/Session.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

philippe44, 2024-01-11:
Your have 2 errors that requires changes in your config, per messages
- CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS must be 2
- you have not enabled exception handling
So, change these with idf.py menuconfig

chagoguila, 2024-01-12:
Thnx! Got it working this way:
1. Cloned the master git.
2. Copied over the sdkconfig from the 4.4-s3 git.
3.  In Menuconfig had to set freertos threads to 2, but also had to set up flash size to 16MB and spram as quad (no octal).
With the above now compiles and flashes. But only works with spotify connect. If I enable airplay it will hang and I had to erase the whole flash to get it working again.

jochen-rath, 2024-04-03:
Hello [chagoguila](https://github.com/chagoguila)
I could compile this code with the sdconfig from the 4.4-s3 git, but I couldn't flash it. Maybe, because I can not find the spram to quad setting. Can you please explain, where I can find this setting in the Menuconfig?

I used esp-idf-v4.4.7 with the sdconfig from 4.4-S3 and with freertos of 2 and a flash size of 16MB.

martymcbro, 2024-07-17:
I have a problem getting the 4.4-s3 branch to run on a Seeed Studio XIAO [ESP32S3](https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/). I just cloned the branch as it is and kept the default sdkconfig. I did not change anything in menuconfig and after fixing some errors regarding protobuf I got it to build successfully with esp-idf 4.4. However after flashing it gets stuck and I can see no Wifi AP.

I don't know if there is still a big issue left or if it's something small. Maybe someone can help me.

Thank you

Here is the log:

idf.py -p /dev/ttyACM0 monitor
/home/martin/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: monitor
Running idf_monitor in directory /home/martin/squeezelite-esp32
Executing "/home/martin/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/martin/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32s3 /home/martin/squeezelite-esp32/build/recovery.elf -m '/home/martin/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/home/martin/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42006324
0x42006324: esp_panic_handler at /home/martin/esp/esp-idf/components/esp_system/panic.c:382 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x178c
load:0x403b6000,len:0xe28
load:0x403ba000,len:0x30b8
entry 0x403b6274
I (24) boot: ESP-IDF v4.4 2nd stage bootloader
I (25) boot: compile time 17:14:53
I (25) boot: chip revision: 0
I (26) qio_mode: Enabling default flash chip QIO
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : QIO
I (41) boot.esp32s3: SPI Flash Size : 8MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 recovery         factory app      00 00 00010000 00140000
I (92) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (99) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (107) boot: End of partition table
I (111) boot: Defaulting to factory image
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=57ce4h (359652) map
I (179) esp_image: segment 1: paddr=00067d0c vaddr=3fc95a70 size=04cd0h ( 19664) load
I (183) esp_image: segment 2: paddr=0006c9e4 vaddr=40374000 size=03634h ( 13876) load
I (187) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=bd3fch (775164) map
I (310) esp_image: segment 4: paddr=0012d424 vaddr=40377634 size=0e434h ( 58420) load
I (322) esp_image: segment 5: paddr=0013b860 vaddr=50000000 size=00010h (    16) load
I (322) esp_image: segment 6: paddr=0013b878 vaddr=600fe000 size=00028h (    40) load
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (351) opi psram: vendor id : 0x0d (AP)
I (351) opi psram: dev id    : 0x02 (generation 3)
I (351) opi psram: density   : 0x03 (64 Mbit)
I (355) opi psram: good-die  : 0x01 (Pass)
I (359) opi psram: Latency   : 0x01 (Fixed)
I (364) opi psram: VCC       : 0x01 (3V)
I (369) opi psram: SRF       : 0x01 (Fast Refresh)
I (374) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (380) opi psram: BurstLen  : 0x01 (32 Byte)
I (385) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (391) opi psram: DriveStrength: 0x00 (1/1)
W (396) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (406) spiram: Found 64MBit SPI RAM device
I (411) spiram: SPI RAM mode: sram 80m
I (415) spiram: PSRAM initialized, cache is in normal (1-core) mode.
[... 45 lines cut ...]
I (2350) esp_app_main: Configuring services
I (2350) services: Configuring I2C sda:-1 scl:-1 port:1 speed:400000
W (2360) services: no I2C configured
I (2360) services: Configuring SPI mosi:-1 miso:-1 clk:-1 host:1 dc:-1
W (2370) services: no SPI configured
E (2380) ledc: ledc_timer_config(322): speed_mode argument is invalid
W (2380) led: LED GPIO -1 ignored
W (2390) led: LED GPIO -1 ignored
I (2390) led: Configuring LEDs green:-1 (active:0 -1%), red:-1 (active:0 -1%)
I (2400) battery: No battery
I (2400) monitor: Initializing monitoring
I (2410) monitor: Heap internal:293059 (min:289875) external:8367587 (min:8367571) dma:287335 (min:284151)
I (2420) esp_app_main: Initializing display
I (2420) display: Trying to configure display with N/A
W (2430) display: No display driver
I (2430) esp_app_main: Checking for update url
W (2440) config: Value not found for key fwurl
I (2440) esp_app_main: Starting Network Manager
W (2450) config: Value not found for key apdelay
I (2450) uart: queue free spaces: 3
I (2450) network_handlers: Network manager configuration: polling max 10176, polling min 15000, ap delay 20000, dhcp timeout 8000, eth timeout 8000
W (2460) network_wifi: No known access point found
E (2470) nvs_utilities: Could not open the nvs storage.
W (2470) network_wifi: network manager has no previous configuration. ESP_FAIL
I (2480) http_server: Initializing HTTP Server

alsaadi3, 2024-12-28:
Can anyone upload a compiled binary for esp32 s3 N16R8? I have tried to get it working for a couple of evenings but no progress. I have no experience with esp-idf. 

I'm in to for a cup of coffee at least.

martymcbro, 2024-12-28:
> Can anyone upload a compiled binary for esp32 s3 N16R8? I have tried to get it working for a couple of evenings but no progress. I have no experience with esp-idf.
> 
> I'm in to for a cup of coffee at least.

Did you already try this installer? 

https://sonocotta.github.io/esp32-audio-dock/

You probably have to disable telnet while in recovery.

alsaadi3, 2024-12-28:
> Did you already try this installer?
>
> https://sonocotta.github.io/esp32-audio-dock/
>
> You probably have to disable telnet while in recovery.

Nice one, thank you so much. I was using a different installer that didn't support esp32s3. If you send me your paypal I will send you that coffee.

martymcbro, 2024-12-28:
> Nice one, thank you so much. I was using a different installer that didn't support esp32s3. If you send me your paypal I will send you that coffee.

You're welcome. No coffee required.

jochen-rath, 2024-12-29:
> Can anyone upload a compiled binary for esp32 s3 N16R8? I have tried to get it working for a couple of evenings but no progress. I have no experience with esp-idf.
> 
> I'm in to for a cup of coffee at least.

If the other way doesn't work, you can find a compiled version for the N16R8 here:
https://github.com/jochen-rath/rfidLmsSteuerung/tree/main/firmware/esp32S3-N16R8_16MB_squeezelite

you can flash the esp32 with following commands:
`pip install esptool`
`esptool.py --chip esp32s3 --port /dev/ttyACM0 erase_flash
esptool.py -p /dev/ttyACM0  -b 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 esp32s3_16MB_squeeze_bootloader.bin 0x8000 esp32s3_16MB_squeeze_partition-table.bin 0xd000 esp32s3_16MB_squeeze_ota_data_initial.bin 0x10000 esp32s3_16MB_squeeze_recovery.bin 0x150000 esp32s3_16MB_squeeze_squeezelite.bin`

If you are using Windows, you need to change the ports accordingly

RASPIAUDIO, 2024-12-29:
I'm not sure to understand why you need a version for N16R8, a N8R8 or N4R8 will probably work as well

alsaadi3, 2024-12-29:
The webflasher didn't end up working. It flashed but then when I changed the I2S pins it went into a bootloop. I tried your firmware and the same thing happened. I changed the pins and it worked with; clock 16, word select 17, data 18. Thanks @jochen-rath for your help.

> I'm not sure to understand why you need a version for N16R8, a N8R8 or N4R8 will probably work as well

I'm not sure, but I would think that you can flash a binary meant for N8R8 or N4R8 to the N16R8, but potentially not the other way around. N16R8 have larger flash size, so I would think it would depend on how the partitions are set out.

RASPIAUDIO, 2024-12-30:
> The webflasher didn't end up working. It flashed but then when I changed the I2S pins it went into a bootloop. I tried your firmware and the same thing happened. I changed the pins and it worked [...]

yes this is what I was saying

Git2tony, 2025-01-14:
> If the other way doesn't work, you can find a compiled version for the N16R8 here: https://github.com/jochen-rath/rfidLmsSteuerung/tree/main/firmware/esp32S3-N16R8_16MB_squeezelite
>
> you can flash [...]

I tried this, and it seemed to work, except for the Bluetooth part. I have no clue how to check, but it seems not to function at all.

I have this board, to me should be up to the task....?!

ARCELI 3pcs ESP32-S3 WROOM-1-N16R8 ESP32-S3-DevKitC-1 Development Boards with WiFi, Bluetooth 5.0

    8M PSRAM: The motherboard supports high-speed OctalPlflash and higher capacity of off-chip RAM and allows user configuration of data and instruction cache.
    RF performance: ESP32-s3-devkitc-1 module of high-performance mode (20 dBm) shares PA with WiFi and ensures a strong signal strength.
    ESP32-S3 multi-mode: it supports infrastructure BSS, simultaneous station mode, SoftAP mode and station + SoftAP mode.
    Low power consumption: The motherboard has a 32-bit LX7 dual-core processor with a main frequency of up to 240 MHz, 512 KB SRAM (TCM), 45 programmable GPIO pins and extensive communication interfaces.
    MCU system on chip: The ESP32-S3 chip offers built-in 2.4 GHz WiFi and Bluetooth 5.0 (LE) with support for long range mode.
    
If anyone could point me in the right direction, that would really be appreciated! I'm trying to create a bridge between HA and my bluetooth speaker. (I got to the part where it was detected by the "Squeezebox (Lyrion Music Server)" -integration.)

sle118, 2025-01-15:
The S3 simply doesn't have the capability to support Bluetooth audio, at least not natively using the esp-idf.

Git2tony, 2025-01-15:
Ok, thank you. If I understand correctly....the board does have Bluetooth, but not the right kind for streaming. Is there any board you can recommend for my purpose?

matthew-mclaren, 2025-01-24:
> Ok, thank you. If I understand correctly....the board does have Bluetooth, but not the right kind for streaming. Is there any board you can recommend for my purpose?

https://github.com/sle118/squeezelite-esp32?tab=readme-ov-file#supported-hardware

Unlikeall3, 2025-04-28:
I was trying to flash 32 bit version with esp idf 4.4.5 without success, I get only errors, when I've managed to flash it there was no audio.

00:02:46.315] output_init_i2s:394 I2S DAC using I2S bck:39, ws:16, do:38, mute:-1:0 (res:0)
[00:02:46.317] output_init_i2s:414 Initializing I2S mode normal with rate: 44100, bits per sample: 32, buffer frames: 512, number of buffers: 12
[00:02:46.329] output_thread_i2s:536 Output state is -1
[00:02:46.329] output_visu_init:75 Initialize VISUEXPORT 512 32 bits samples
[00:02:46.341] output_thread_i2s:540 switching off amp GPIO -1
[00:02:46.352] output_init_embedded:108 init completed.
[00:02:46.354] decode_init:153 init decode
[00:02:46.354] register_alac:547 using alac to decode alc
[00:02:46.366] register_helixaac:686 using helix-aac to decode aac
[00:02:46.367] register_vorbis:526 using vorbis to decode ogg
[00:02:46.378] register_opus:458 using opus to decode ops
[00:02:46.379] register_flac:338 using flac to decode ogf,flc
[00:02:46.390] register_pcm:468 using pcm to decode wav,aif,pcm
[00:02:46.391] register_mad:417 using mad to decode mp3
E (3608) pthread: Failed to create task!
[00:02:46.405] discover_server:844 sending discovery 25
[00:02:51.407] discover_server:844 sending discovery 24
[00:02:56.407] discover_server:844 sending discovery 23
[00:03:01.407] discover_server:844 sending discovery 22
[00:03:06.407] discover_server:844 sending discovery 21
[00:03:11.407] discover_server:844 sending discovery 20
I (28798) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (28798) wifi:station: 2a:b1:21:25:f1:c2 join, AID=1, bgn, 20
W (28798) wi

philippe44, 2025-04-28:
Your sdkcondig is probably incorrect, compared it with our references

philippe44, 2025-04-29:
I’ll upload a working one

I've used i2s sdkconfig from project build,tons of errors,
Can't program it properly, ESP32-S3-WROOM-1-N16R8,any clues what could be wrong,I can upload logs sdkconfig or even screencapture if necessary.

Unlikeall3, 2025-04-29:
> I’ll upload a working one
> 
> Envoyé de mon iPhone
> 
> 
> 
> [https://avatars.githubusercontent.com/u/190084118?s=20&v=4]Unlikeall3 left a comment ([sle118/squeezelite-esp32#209](https://github.com/sle118/squeezelite-esp32/issues/209))<[#209 (comment)](https://github.com/sle118/squeezelite-esp32/issues/209#issuecomment-2837543166)>
> 
> Your sdkcondig is probably incorrect, compared it with our references
> 
> I've used i2s sdkconfig from project build,tons of errors,
> Can't program it properly, ESP32-S3-WROOM-1-N16R8,any clues what could be wrong,I can upload logs sdkconfig or even screencapture if necessary.

Oh,thanks,I am using it with my jab5 amplifier: https://youtube.com/shorts/DYBJ-F49zzo
The 16 bit version works,but I want 32 bit one,then I'll be waiting for proper sdk config file.

Unlikeall3, 2025-05-03:
> I’ll upload a working one
> 
> Envoyé de mon iPhone
> 
> 
> 
> [https://avatars.githubusercontent.com/u/190084118?s=20&v=4]Unlikeall3 left a comment ([sle118/squeezelite-esp32#209](https://github.com/sle118/squeezelite-esp32/issues/209))<[#209 (comment)](https://github.com/sle118/squeezelite-esp32/issues/209#issuecomment-2837543166)>
> 
> Your sdkcondig is probably incorrect, compared it with our references
> 
> I've used i2s sdkconfig from project build,tons of errors,
> Can't program it properly, ESP32-S3-WROOM-1-N16R8,any clues what could be wrong,I can upload logs sdkconfig or even screencapture if necessary.

Okay so,would you upload sdk file?

ElFishi, 2025-12-27:
I am still struggling with building for the S3 in a reproducible manner. Sometimes it works and sometimes I can't switch from recovery to to squeezlite
```
W (69785) messaging: restart_ota
Booting to Squeezelite
I (69785) messaging: Rebooting to ota_0
I (69795) esp_image: segment 0: paddr=00150020 vaddr=3c160020 size=aa37ch (697212) map
E (69865) esp_image: invalid segment length 0x3e000000
E (69865) messaging: Unable to select partition for reboot: ESP_ERR_OTA_VALIDATE_FAILED
E (69885) messaging: Unable to select partition for reboot: ESP_ERR_OTA_VALIDATE_FAILED
W (69885) messaging: Rebooting.
```
The problem only seems to occur in flash_mode QIO and seems to correlate with the size of segment 0.
Sizes ending in hex 0, 4, and 8 work, a size ending in c fails. It seems the 8 byte size header of the following segment can't be read when it spans a 16 byte window. A segment length ending with c leaves only 4 bytes in one window and needs 4 bytes of the next. I can introduce dummy arrays in main/esp_app_main.c to nudge the size and make it work, but that is very tedious. Surely there must be a better way?

prydie, 2026-01-12:
> I was able to flash on my ESP32-S3 N16R8.
> 
> 1. Git cloned this repo
> 2. Deleted the sdkconfig.default
> 3. Replaced the sdkconfig with https://github.com/sonocotta/esp32-audio-dock/blob/main/firmware/squeezelite-esp32/sdkconfig.esp32s3
> 4. Created docker container for esp-idf 4.4.8
> 
> ```
> docker pull espressif/idf:v4.4.8
> docker run -it -v /tmp/squeezelite-esp32:/workspace/squeezelite-esp32 \
> 	 -w /workspace/squeezelite-esp32 \
> 	 espressif/idf:v4.4.8 \
> 	 bash
> pip3 install protobuf grpcio-tools
> ```
> 
> 5. I also increased the partition size, but not sure if needed:
> 
> ```
> # Name,   Type, SubType, Offset,  Size, Flags
> # Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
> nvs,  data,  nvs,  0x9000,  0x4000,
> otadata,  data,  ota,  0xD000,  0x2000,
> phy_init,  data,  phy,  0xF000,  0x1000,
> recovery,  app,  factory,  0x10000,  0x240000,
> ota_0,  app,  ota_0,  ,  0x2A0000,
> settings,  data,  nvs,  ,  0x10000,
> ```
> 
> 6. From the container: `idf.py build`
> 7. From my Mac: `esp_rfc2217_server -p 4000 /dev/tty.wchusbserial5A790608361`
> 8. From the container: `idf.py --port 'rfc2217://host.docker.internal:4000?ign_set_control' flash --baud 921600`
> 9. Then wifi AP appears, connect to your wifi, open the right page, etc... Then press `exit recovery` from the web ui to start the process.
> 10. By running `dns-sd -B _raop._tcp local` on your mac, you can see when the AirPlay device becomes available, just for monitoring.
> 
> Currently my airplay device disappears after some time though...
> 
> ```
> [00:01:36.830] discover_server:844 sending discovery 6
> [00:01:41.830] discover_server:844 sending discovery 5
> [00:01:46.830] discover_server:844 sending discovery 4
> [00:01:51.830] discover_server:844 sending discovery 3
> [00:01:56.830] discover_server:844 sending discovery 2
> [00:02:01.830] discover_server:844 sending discovery 1
> [00:02:06.831] decode_close:233 close decode
> [00:02:06.931] deregister_external:495 Stopping AirPlay sink
> [00:02:07.031] deregister_external:502 Stopping CSpot sink
> [00:02:07.032] stream_close:593 close stream
> [00:02:07.131] output_close_embedded:112 close output
> ␛
> ```
> 
> I don't use LMS, just AirPlay, maybe that's the issue.

https://github.com/sonocotta/esp32-audio-dock/blob/main/firmware/squeezelite-esp32/sdkconfig.esp32s3 appears to be a dead link; interested as to where I can find this.

prydie, 2026-01-13:
I think we're talking at crossed purposes. Either way I've finally got a working build with the help of my _trusty_ AI companion.

I'm mainly confused as to how others are getting it built with the includes of `esp_eth.h` without doing something similar to the below.

```
$ docker run -it -v $(pwd):/workspace/squeezelite-esp32 \
         -w /workspace/squeezelite-esp32 \
         espressif/idf:v4.4.8 \
         bash
```

```
$ pip3 install protobuf grpcio-tools
```

```bash
#!/bin/bash
# -----------------------------------------------------------------------------
# Bristol ESP32-S3 N16R8 Build Script
# Target Hardware: ESP32-S3 (16MB Quad Flash, 8MB Octal PSRAM)
# Strategy: Disable Ethernet drivers to prevent S3 crash, fix linker dependencies.
# -----------------------------------------------------------------------------

set -e # Exit immediately if a command exits with a non-zero status

LOG_FILE="build_log.txt"

# 1. Clean Environment
echo "--- 🧹 Cleaning Build Environment ---"
idf.py fullclean
rm -rf build sdkconfig $LOG_FILE

# 2. Configure for ESP32-S3
echo "--- ⚙️ Applying Configuration ---"
cp build-scripts/I2S-S3-sdkconfig sdkconfig

# 3. Apply N16R8 Hardware Specifics (Hybrid Memory Config)
# We use Octal (OPI) for the fast RAM, but Dual/Quad (DIO) for the Flash.
cat <<EOF >> sdkconfig
# --- Bristol N16R8 Overrides ---
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_ESPTOOLPY_FLASHMODE_DIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_OCT_FLASH=n
# Fix missing font symbol
CONFIG_GDS_FONT_DROID_SANS_FALLBACK_15X17=y
EOF

# 4. Stub Low-Level Ethernet Drivers
# These drivers access ESP32 hardware registers that don't map 1:1 to S3, causing crashes.
echo "--- 🚫 Stubbing Incompatible Ethernet Drivers ---"
STUB_TEXT="/* S3 Stub: Ethernet disabled for stability */"
for file in components/wifi-manager/network_driver_DM9051.c \
            components/wifi-manager/network_driver_LAN8720.c \
            components/wifi-manager/network_driver_W5500.c; do
    echo "$STUB_TEXT" > "$file"
done

# 5. Patch Network Ethernet Logic
# We remove conflicting system calls but preserve application-layer stubs.
echo "--- 🩹 Patching Network Ethernet Layer ---"
cat <<EOF > components/wifi-manager/network_ethernet.c
#include <stdio.h>
#include <stdbool.h>
#include "esp_eth.h"
#include "esp_netif.h"
#include "network_ethernet.h"

// --- Application Layer Stubs ---
// These satisfy linker references from other components (accessors.c, network_status.c)
// without actually invoking the Ethernet hardware.

void init_network_ethernet() { }
bool network_ethernet_enabled() { return false; }
esp_netif_t* network_ethernet_get_interface() { return NULL; }
network_ethernet_driver_t* network_ethernet_driver_autodetect(const char* driver) { return NULL; }
bool network_ethernet_is_up() { return false; }
bool network_ethernet_wait_for_link(uint16_t max_wait_ms) { return false; }
void destroy_network_ethernet() { }
void network_ethernet_start_timer() { }
EOF

# 6. Fix Linker Dependencies
echo "--- 🔗 Fixing CMake Dependencies ---"
# Add missing 'audio' and 'wifi-manager' deps to platform_console
sed -i 's/PRIV_REQUIRES /PRIV_REQUIRES audio wifi-manager /' components/platform_console/CMakeLists.txt
# Add missing 'display' and 'services' deps to squeezelite-ota
sed -i 's/PRIV_REQUIRES /PRIV_REQUIRES display services wifi-manager /' components/squeezelite-ota/CMakeLists.txt
# Fix Squeezelite include path
sed -i 's|INCLUDE_DIRS .|INCLUDE_DIRS . ../../main|' components/squeezelite/CMakeLists.txt

# 7. Stub Bluetooth (Optional, to save flash/IRAM if needed)
echo "--- 🦷 Stubbing Bluetooth Helpers ---"
sed -i '/bt_app_core.h/d' components/wifi-manager/network_status.c
sed -i '1i int bt_app_source_get_a2d_state() { return -1; }' components/wifi-manager/network_status.c
sed -i '1i int bt_app_source_get_media_state() { return -1; }' components/wifi-manager/network_status.c

# 8. Build
echo "--- 🔨 Starting Build ---"
export TARGET_BUILD_NAME="I2S-S3-Bristol"
export BUILD_NUMBER="500"
export DEPTH="16"

idf.py build
```

prydie, 2026-01-13:
Ahh, no. I couldn't get it compiled at all! I don't need AirPlay - just using the LMS functionality but on S3-N16R8.

kruzio1985, 2026-01-14:
i donknow yet why my 4.4 code works good and code where i was upgraded drivers all to new one and settings to 5.3 then nvs write after save complety stop working

ElFishi, 2026-01-14:
@prydie : do you switch flashmode to DIO? 
I [can't build](https://github.com/sle118/squeezelite-esp32/issues/533) reliably with QIO. 

[Using 4.4.7](https://github.com/sle118/squeezelite-esp32/discussions/522) I don't have problems building without tampering with esp_eth.h.
In addition, I just tried 4.4.8, and it builds, too (haven't flashed it though).

[5 more replies]