sle118/squeezelite-esp32 #212: MKS-MINI12864-V3

bench2012, 2022-12-02:
Hi,

Any chance of adding support for ST7567 128x64 mono lcd display
, which the MKS-MINI12864-V3 is based on. It make a nice display package with rotary encoder and rgb backlight.

https://github.com/makerbase-mks/MKS-MINI12864-V3

Many thank,

bench

sle118, 2022-12-02:
Hi Bench,

This is a nice display and certainly an interesting addition, but difficult without having a unit to test against.  Most of it should be supported via configuration changes, except for the display. You might want to try some of the existing drivers for the display and see if the ST7567 works out of the box. If not, then I suggest you try adding it yourself (by copying an existing driver that is close to working and changing it accordingly). 

At this point, we could review a pull request and integrate the changes.

For now, I will have to close this one. 

Thank you!

closed 2022-12-02 by sle118

bench2012, 2022-12-05:
Many thanks. Will give it a go, once I find some time.