sle118/squeezelite-esp32 #216: LMS playback stuttering

DL6ER, 2022-12-31:
Thank you for this amazing project.

I set up a bare ESP32-WROVER. I only enabled SPDIF and want to play music from my LMS server. For SPDIF, I connected an LED to the `DO` output and attached this to the SPDIF fiber:

[image]

This all works amazingly well for up to about 10 minutes where the audio is getting intermittent and eventually completely breaks. The web interface shows some `timeout` status messages at the same time. WiFi signal is rather good, the ESP32 is located about 2 meters away from the access point. It also happens when I don't have a browser tab opened for the web interface.
A restart of the ESP32 resolves the issue for the next 10 minutes.

**Do you have any suggestions for me what I can try?**

Below is my current configuration:

[image]

[image]

[image]

sle118, 2023-01-02:
First thing to do is to connect your device to a USB port and get some logs that you can attach here.

andiling, 2023-03-04:
Hi, i have the same setup (SPDIF Transmitter on PIN 32) but i have no playback problems! I Use a Toslink transmitter from a ASUS Mainboard. 

fw: I2S-4MFlash.16.2.1254.master-v4, mode: Squeezelite-ESP32

sle118, 2023-03-06:
@DL6ER given that I don't have logs to figure out what is going on, I cannot help. I'll close this for now. Feel free to reopen to add more details

closed 2023-03-06 by sle118