sle118/squeezelite-esp32 #219: Muse Luxe Visualiser support

dekoenpi, 2023-01-31:
Hi all,

Please add visualiser support in squeezelite.

command line working:
squeezelite -b 500:2000 -d all=info -C 30 -W -o I2S -Z 96000 -n "MuseLuxe"
command line not working on muse luxe :
squeezelite -b 500:2000 -d all=info -C 30 -W -o I2S -Z 96000 -n "MuseLuxe" -v

-v | Visualiser support

Best regard
P-Kay

sle118, 2023-02-03:
What outcome would you expect from activating this flag? I'm not sure I understand

sle118, 2023-02-06:
This is not going to happen. We're already pushing the ESP32 to its limits and there's no way it will be able to support that sort of heavyweight task on top of it all. 

This sort of visualizer is likely better suited for a raspberry Pi or a similar platform.

closed 2023-02-06 by sle118