sle118/squeezelite-esp32 #221: SPDIF algorithm

masterxq, 2023-02-08:
Hey bois*,
Tonight I had time to learn spdif, and I wrote an optimized implementation of the spdif convertion algorithm. As it not matches all your requirements (only 4 byte samples) I would like to post it here, so you have the option to use it.

I measured the speed of multiple algorithm, currently this on is the winner.
This was the **test conditions**:

- 1024 byte input buffer (256 stereo samples).
- Test environment: ESP32
- Optimization: -O2

**Results:**
- Philippe44 algorithm needs: `88 us`
- Another algorithm I found on the net and optimized slightly, before I understood spdif: `73 us`
- MasterQ (this one) algorithm: `60 us`

Additionally, this algorithm **fixes some problems**:
- passes the validation
- no noise on the last 4 bits
- much more straight (handles sample by sample)
- really sets the validation bit instead of using

It brings one **new potential problem**:
- Only for one instance as local static variables are used
- 8 byte samples not implemented

Both points should be easy to change or implement.

Phillippe44 algorithm did not parse the verification, what most likely was An mistake of me by reimplementing it to my test application. But maybe this should be checked, I will attach my validation code.
```
E (5080) TestProject: Failed dword3 or dword4, halfsample: 511 bit: 0
I (5090) TestProject: V_U_C_P_P_P_P_P_ A_A_A_A_S_S_S_S_ S_S_S_S_S_S_S_S_ S_S_S_S_S_S_S_S_ V_U_C_P_P_P_P_P_ A_A_A_A_S_S_S_S_ S_S_S_S_S_S_S_S_ S_S_S_S_S_S_S_S_
I (5110) TestProject: 1100110011100100 1100110011010101 1100110011001100 1100110011001100 1100110011100010 1100110011001100 1100110011001100 1100110011001100
```

What was tested:
- Tested to play sound on a real device (passed)
- Tested the data with a very simple spdif validator (passed)
- Tested the spdif validator with other algorithm (2 of 3 passed)
- Chunk test with real device and validator

What was not tested:
- Decode the data and compare with the original data
- Long time tests

Fell free to use this code as and where you want. If you have a good
day, then mention where it comes from :)

I will post here if i find issues or have optimizations.

**Here is the code**
```
#define VUCP_PREAMBLE_B_16_32 0xCCE80000
#define VUCP_PREAMBLE_M_16_32 0xCCE20000
#define VUCP_PREAMBLE_W_16_32 0xCCE40000

//For having a pretictable state the entries allways starts
//with first bit set. This is ok as the phase does not matter
//in the bmc encoding!
uint16_t spdif_bmclookup_1first[256] =
{
	0xCCCC, 0xB333, 0xD333, 0xACCC, 0xCB33, 0xB4CC, 0xD4CC, 0xAB33,
	0xCD33, 0xB2CC, 0xD2CC, 0xAD33, 0xCACC, 0xB533, 0xD533, 0xAACC,
	0xCCB3, 0xB34C, 0xD34C, 0xACB3, 0xCB4C, 0xB4B3, 0xD4B3, 0xAB4C,
	0xCD4C, 0xB2B3, 0xD2B3, 0xAD4C, 0xCAB3, 0xB54C, 0xD54C, 0xAAB3,
	0xCCD3, 0xB32C, 0xD32C, 0xACD3, 0xCB2C, 0xB4D3, 0xD4D3, 0xAB2C,
	0xCD2C, 0xB2D3, 0xD2D3, 0xAD2C, 0xCAD3, 0xB52C, 0xD52C, 0xAAD3,
	0xCCAC, 0xB353, 0xD353, 0xACAC, 0xCB53, 0xB4AC, 0xD4AC, 0xAB53,
	0xCD53, 0xB2AC, 0xD2AC, 0xAD53, 0xCAAC, 0xB553, 0xD553, 0xAAAC,
	0xCCCB, 0xB334, 0xD334, 0xACCB, 0xCB34, 0xB4CB, 0xD4CB, 0xAB34,
	0xCD34, 0xB2CB, 0xD2CB, 0xAD34, 0xCACB, 0xB534, 0xD534, 0xAACB,
[... 246 lines cut ...]
		}
		
		if(!expected)
		{
			ESP_LOGE(TAG, "wrong expection for next bit");
			return false;
		}
		
		//TODO: remove this
		expected = true;
	}
	if(pw_rest < 0 || pb_rest < 0 || pm_rest < 0)
	{
		//This dont has to be an error for small datasets!
		ESP_LOGW(TAG, "Did not see all preamble, problem is possible");
		return false;
	}
	ESP_LOGW(TAG, "Did not expect to reach this");
	return true;
}
```

Hopefully this helps somebody!

Best regards and happy hacking!

sle118, 2023-02-08:
Thank you for your contribution. Hopefully this will be something that Philippe will take the time to consider, when he has chance

philippe44, 2023-02-08:
Thanks, seems great - have you tried it with the complete squeezelite-esp32 to measure the global cpu gain? I've not checked how much this loops weights in the cpu load vs the fact that we have to 4x i2s rate

masterxq, 2023-02-08:
I'm very sorry, I have no environment for testing it and have very limited knowledge about your project, just read the i2s/spdif part, what really was helpful by understanding what I have to do...
I have written my own libs and using some adf libs. What maybe is partly a bit sad, because I think there is high quality code in your project available, and I did fix the same issues as you did. But the problem remains the same, I have to know what issues exist before they can fix them, after understanding the issues is often easier to write the solution than search and integrate existing solutions. And it's less difficult to fix upcoming issues ^^ 

But what I can do, is sending you my complete testing app, then you can implement your algorithm correctly or make the test function respect the offset in your generated data and handle it correctly, what probably is the concern with the failed validation additional you can test the performance of upcoming code. The app is just quick and dirty but meets the requirements, only want to do basic tests, as this is only 1% of my project ^^

If you want we can stay in contact to avoid redundant work in the future :)

My focus are hardware solutions, and I'm facing audio hardware from time to time.

masterxq, 2023-02-08:
> Thanks, seems great - have you tried it with the complete squeezelite-esp32 to measure the global cpu gain? I've not checked how much this loops weights in the cpu load vs the fact that we have to 4x i2s rate

Sorry my other answer is still correct, but did reread, and I have at least some information about the CPU usage, it needs about 10-12% (freertos taskmanager) of a single core in my application, but I'm still using i2s_write, what copies the buffer once more, what is not needed as long I remember from another project, it should be possible to just queue some buffers without copy them once more and directly handle them over to dma (no cpu), what makes much more sense. I can't remember what you are doing... The last measurement was before I wrote the new algorithm, I think it should save about 1-2% what is ok but not solves problems in the most cases. For me, it was more important to understand what's happening and have my data under control, being able to handle variable streams and have clean samples, maybe this can solve problems for some people, at least I have a better feeling with it :)
Ofc I complete see your arguments that noise in the LSB can not be heard!

Best Regards

masterxq, 2023-02-09:
[image]
I really wonder, but it seems like I did remember wrong, here are my esp32 CPU stats. And I'm still using i2s_write. My modifications should not be in a visible range if the total usage is 2%...

`(44,1 kHz 320 kBit/s mp3, streamed over https)`

I was that sure it was more than 10%. But ok it's not, at least no longer^^

Maybe I had a mistake in my code on my last measurement.

sle118, 2023-04-01:
@masterxq do you feel is't something worth exploring?

masterxq, 2023-08-05:
> @masterxq do you feel is't something worth exploring?

As the algorithm is completed and this will save bus time, cpu time and give better sample and is easier to maintain and expand (eg other sample width). I would implement it, maybe not as high priority task but, I would do it :) And I have in my audio toolchain. It's working flawless from day one.

But at the end it is up to you :) I just want to share my code, as yours helped me to understand the problem :)

sle118, 2023-08-13:
@masterxq thank you for exploring these improvements. Il see if/when we can get to it.

sle118, 2023-08-13:
Given that the solution is 16 bits, this isn't going to work for us. Closing

closed 2023-08-13 by sle118