sle118/squeezelite-esp32 #222: delete saved ssid

shahglobal, 2023-02-16:
Hi smart people,
My squeezelite esp32 modules sometimes connects to ssid that I do not want, I manually connect it to ssid2, but at reboots some times it switches back to the ssid that I do not want so breaking my setup for homeassistant. 
Can I somehow delete the ssid that I do not want it to connect. without re-flashing the firmware?
any tip would be awesome. I even tried removing the saved password in unwanted ssid saved settings but it do not fall back to the second ssid. it still connects to the unwanted ssid.
anyways any tweak that helps to remove unwanted ssid that is saved , would be great.
Thanks
Shah

sle118, 2023-02-17:
hello @shahglobal 
You will have to provide all the details of your configuration to expect any help.  There's a wide variety of configurations out there and several releases so the very least I'll need is the build you're running, what sort of hardware config it is running on, but more importantly you'll have to provide some logs from the system when this is happening. 

without all of these, there's really no way for me (or anyone else) to figure out exactly what you are experiencing.

thanks

shahglobal, 2023-02-17:
Thanks for the feedback, I went the long route by re-flashing it.  Only connected it to the ssid that I want. This will do for now. Moral of the story is connect it to the ssid that you want it, if you  connect to a second ssid, it will still connect to the first ssid if its available. May be a newer update will have an option to "delete" ssids that are saved.  Anyways my esp32 module is fine after re-flashing.
Cheers
Shah

sle118, 2023-02-17:
Ok. I'll close the issue for now then. You can reopen it as needed

closed 2023-02-17 by sle118