sle118/squeezelite-esp32 #224: Help wanted: building with a different flashmode (e.g. QOUT)

AndreasBVB, 2023-03-05:
Hi thank you very much for putting this together. I currently have multiple copy of a device with perfect 8MB PSRAM but flash up to QOUT. It looks to me the default build BIN is built with QIO flash mode. Kindly instruct how to build my own with QOUT or even DIO/DOUT?

Also not sure how these two options are working with each other:
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHMODE="dio"

sle118, 2023-03-06:
This is not recommended. Have a look at the readme:
[The bare minimum required hardware is a WROVER module with 4MB of Flash and 4MB of PSRAM](https://github.com/sle118/squeezelite-esp32#performances)

> The esp32 must run at 240 MHz, with Quad-SPI I/O at 80 MHz and a clock of 40 Mhz. Still, it's a lot to run, especially knowing that it has a serial Flash and PSRAM, so kudos to Espressif for their chipset optimization

Running at anything below QIO is going to fail because the memory bandwidth is going to be insufficient for all we do.

AndreasBVB, 2023-03-06:
Thank you very much for the info. Will keep this in mind when shopping for devices.

closed 2023-03-06 by AndreasBVB