sle118/squeezelite-esp32 #225: API remote restart/reboot

stephiegu, 2023-03-05:
Hi, is there any way to remotely (e.g. over calling an url or some other way) rebooting the esp - as I would on clicking the reboot button?
I am asking because I could not find any working solution. I would love to automatically reboot the esp because they sometimes do not respond and i am tired doing a reboot by browser or even by hand.

If there is  no way do you have any ideas how to accomplish this?

Happy for open discussion :)

Best,
Stephie

sle118, 2023-03-06:
I'd be really curious to understand why the system isn't responding in the first place. In other words, how do you use the platform and why it becomes unresponsive. As you can imagine, we didn't build this system to crash and addressed all the cases we could find out such occurrence. Please don't hesitate to collect logs and to open another issue for that, detailing the version of firmware in use, the hardware, the type of streaming being done, etc.

As for your question, see this
https://github.com/sle118/squeezelite-esp32/blob/982ac3dc8a0908e0cfa98dfe088293edad7f8593/components/wifi-manager/wifi_manager_http_server.c#L78
You can do a simple Post of, say, {ok} to reboot.json

philippe44, 2023-03-15:
Since we moved to 4.3 there are a few cases when the device looses WiFi connection, it never comes back. If you remember, we had that in cmake version and it was a race condition on the WiFi handler. I also did some changes to still have softAP while doing regular searches and come back to last known AP. I think the issue came back on 4.3, mostly when the WiFi AP disappear and come back, we never reconnect.

sle118, 2023-03-15:
I thought I had addressed this already in 4.3. I'll need to cross check if I have some uncommitted changes

philippe44, 2023-03-15:
My memory might be at fault but what I'm sure is that there are cases where reconnection never happens in case the AP disappear for a while. I have one device on battery and without reset switch so every time I need to let battery drain when it happens :-)

philippe44, 2023-03-16:
Indeed they are not what they use to be 😃 but in that case it's a Classic retrofitted with a SqueezeAMP

sle118, 2023-04-01:
I'll close this for now, given that there has not been any recent response from the original poster

closed 2023-04-01 by sle118