sle118/squeezelite-esp32 #227: Add support for Line input on ESP32-Audio-kit and others

nagyrobi, 2023-03-12:
Follow up from https://github.com/sle118/squeezelite-esp32/discussions/164, seconding @FHFS and jamesmyatt I'm chiming in for an option to use the Line Input on this board.

The use case is very simple - use the squeezelite system to route local sound (from a TV connected to LINEIN) to the speakers (just like it would be Bluetooth). The reason for using direct cable connection is zero delay introduced in the sound which is important due to lipsync with the TV picture.

One could use the board to place it between the TV and the speakers, and this way when not watching TV, all the LMS/SqueezeLite stuff can be enjoyed on the same speakers.

It's important to have the LINEIN feature published towards the server too, so the input can be selected through the server too (eg. using an automation related to powering ON/OFF the TV set and the speakers).

FHFS, 2023-03-12:
I do like this option. I would love the option to add my speakers to my TV, or connect to my laptop docking station. If someone could point me in the right direction maybe I could figure it out and create a pull request for it.

nagyrobi, 2023-03-13:
Wow. Please put me on the list, I'd be the first to buy and test your prototypes. I need at least 2 or 3 pieces in my home.
This new board emphasises on everything which important, especially the Ethernet connection.

May I add an extra argument in favour of Toslink input: TV sets these days don't offer an analog output anymore. Not even 3.5mm jacks for headphones. What they do offer is optical output via Toslink and HDMI ARC (Audio Return Channel) on one of the HDMI input connectors. No way to use any analog source. 
If we had that, we could design our own open source soundbar replacement!

I see you added a combo SPDIF/Toslink connector to the board. If you don't want to re-design the whole thing you could go hybrid and just wire them separately to the codec (if they are not terminated on the same pin inside the connector). We could then choose to use for example Toslink as input and SPDIF as output or vice-versa in the config. That should fit almost everyone, having the same signal duplicated on two connectors is usually redundant, at least this way we would also have a choice to select which one to be used for what.

sle118, 2023-03-13:
It's not possible to wire the board's output to the input side because on the driving transformer on the RCA as well as the fact that the combo has an optical output driver, which is not pin compatible with the optical input combo of the same product family.

I'll consider including digital input for the next revision of the prototype, as the codec supports it

nagyrobi, 2023-03-21:
Like described in https://github.com/sle118/squeezelite-esp32/issues/228#issuecomment-1478183653 perhaps this could be implemented in software rather easily, by just watching for the command `playlist play source:toslink` coming from LMS.

nagyrobi, 2023-03-23:
> we also want to take the I2S bitstream coming out of the DAC/ADC chip and funnel it through a network connection (e.g. http streaming link) and tell the server to make it available for other players to source from. I am not sure if the existing plugin was already taking care of this or not.

That should be a different topic. 

For the use case mentioned above, this wouldn't be beneficial, as it would likely introduce a considerable amount of delay in the sound stream which would make it out of sync with the TV picture.

As a first step, I think it would be just enough to route the selected input to the output within the codec, and just take care of the statuses within the ESP.

When the live input needs to be also routed through the ESP, that should be a separate feature.

nagyrobi, 2023-03-31:
Raspiaudio Muse Luxe, AI-Thinker ESP32-A1S Audio Kit and Espressif LyraT v4.3 are quite popular and they all contain the ES8388 DAC.

This is a great DAC! Because it has a built-in mixer and multiple outputs too:
- has two analog stereo inputs (one L/R pair with a switchable microphone preamp)
- has implicitly an I2S DAC
- has two analog outputs
- the inputs can be mixed with the DAC signal to the outputs

ESP32 Audio Kit with the ESP32-A1S module is a wonderful board because:
- it has small onboard microphones ready on one of the stereo input pair
- it has a stereo jack line input on the other pair
- has a 2x3W onboard amp on one of the output pair
- has a stereo jack headphone on the other output pair
- has a GPIO connected to the headphone jack for plug detection

This DAC could solve alone the whole task.

According to [ES8388 documentation](http://www.everest-semi.com/pdf/ES8388%20DS.pdf) it's very versatile:
- it has separate gain control on the inputs (independently on L/R channels, registers 39, 42)
- DAC-only volume (registers 26, 27) and mute (register 29) on L/R
- it has separate volume control for the sum (mixed) outputs (registers 46-49)

Since the mixer passes audio through there's no delay in the process, which is beneficial to keep lipsync with the TV picture if used as a soundbar. ESP32 Audio Kit could do this out of the box.

Afaik this DAC can also digitize the input signal so yes, it could also be usable to generate a stream for the network.

philippe44, 2023-04-01:
As said, the only part where there is an embryo of code is for controlling the jack from the LMS UI, because it's a feature of the Boom product line to have it as an output or an input and use the built in amp. 

Streaming up is another thing as we would have to activate an encoder and putting encoded audio in a ring buffer and send what we have (I don't recommend streaming pcm). Having an http server is fine

nagyrobi, 2023-04-01:
> As said, the only part where there is an embryo of code is for controlling the jack from the LMS UI, because it's a feature of the Boom product line to have it as an output or an input and use the built in amp.

Yes, that's exactly why I opened this feature request!

philippe44, 2023-10-16:
No I don't and it's potentially a bit disruptive for many other aspects of the code (side effects of programming the DACs) if you want to do that properly, including LMS integration, and if we do this, I really want it to be clean, not a quick hack job that we'll regret in a year or so, looking at it and wondering WTF were we trying to do 😄

wizmo2, 2023-10-16:
Been using LMS/Slim stuff since the early days, and what you have put together here is immense!  I'm just a tinkerer who like to help out were I can.
 
Let me play in my S3 fork and see what comes out.  At this stage, I want to get a better understanding of how your existing architecture is structured, and was checking to see if you had already put some thoughts into how any ADC modules "should" be integrated.   

Re, PR validation:  it's important, that you as developers, keep a handle on the project.   If you feel any of my Pulls are overstepping then let me know and or just close them.  I'm not easily offended 😉.

sle118, 2023-10-23:
@wizmo2  I'm doing an attempt at factoring the platform config stuff and I'll need some help there. I think I can do better than the initial implementation in terms of design and efficiency, but I still have lots to do before I get a satisfactory clean compile. My goal is to reduce the memory footprint as well as the flash storage that the config system takes, while centralizing access to configuration options

philippe44, 2023-10-23:
I think we should be careful with too many changes in // because we are not so many... people. This project is still up and running because we have been able to keep a pretty good ratio of stability vs features, i.e. it's not cranking a new thingy every other week at the expense of the result rebooting every other hour. 

I've seen many projects looking great on the features list and never "converging" (being stable), thus remaining a big bucket of spaghetti code. Our code is very far from perfect, but it's not spaghetti 😄 and AFAIK, it's essential to keep it under control because once you've started that, it's a slippery slope and we all know that the "I'll do that quick hack today to make it work for now and fix it later" never works. You **never** fix it later, move to the next shiny object and keep piling up junk till you have to give up (well, they'll call it refactor...)

Anyway, long message to say that my opinion is that we should now concentrate on @sle118 work to refactor the config which needs it but put the rest on hold for now until we have something that is stable **and** clean.

wizmo2, 2023-10-23:
@sle118, let me know how I can be of assistance.

sle118, 2023-11-08:
> @sle118, let me know how I can be of assistance.

As soon as I have a solid foundation, I'll commit the code back to a new branch. When it hits that branch, the code will pretty much have been stripped from all the NVS references (with will have been commented out). At this point of time, I will need help to re-attach all the settings relevant code to the new settings module.  So your help will definitely be appreciated!

alidaf, 2024-01-18:
@sle118 I really like the look of the SqueezeIO but I'd like to request some considerations after designing cases for various Squeezelite devices that would make it much more accessible and easier to create custom cases for it.

Please could you consider making duplicate connectors (screw terminals or JST etc) or alternative mounting options for all of the inputs/outputs, e.g. power supply, RCA, IR, SPDIF, speaker output, etc., so that panel mounted versions can be designed into the case. I've been really handicapped by PCB mounted socketry because everything has to be made as a pass-through or daisy chain and the existing PCB mounted stuff just takes up additional space.

Also consider make the mounting holes a decent size (at least M2.5) with adequate clearance underneath for a decent mount and spaced adequately for rigidity, i.e. sufficient to allow inserting/removing plugs without stressing the board. The M1.4 screws I had to use for the SqueezeAMP caused a lot of problems.

sle118, 2024-02-03:
> @sle118 I really like the look of the SqueezeIO but I'd like to request some considerations after designing cases for various Squeezelite devices that would make it much more accessible and easier to create custom cases for it.
> 
> Please could you consider making duplicate connectors (screw terminals or JST etc) or alternative mounting options for all of the inputs/outputs, e.g. power supply, RCA, IR, SPDIF, speaker output, etc., so that panel mounted versions can be designed into the case. I've been really handicapped by PCB mounted socketry because everything has to be made as a pass-through or daisy chain and the existing PCB mounted stuff just takes up additional space.
> 
> Also consider make the mounting holes a decent size (at least M2.5) with adequate clearance underneath for a decent mount and spaced adequately for rigidity, i.e. sufficient to allow inserting/removing plugs without stressing the board. The M1.4 screws I had to use for the SqueezeAMP caused a lot of problems.
> 
> IR pass-through would also be handy!

I will definitely consider this on the next iteration, as the current one was a increment that was done to prove myself I could design a working PHY for Ethernet, which requires great care in laying down the PCB traces.

That being said, I can definitely make mounting holes better; there's room on the board. Also, I believe the spdif/RCA combo connector I'm using has a mounting screw, which would contribute to increasing strength.

But I'm not sure I understand what you mean by pass through or daisy chain. 

Additionally, I can generate a 3D model of the board with connectors, something that might help with the case design effort. I was also thinking of making my board wider to fit the usb connector on the back.

[image]

alidaf, 2024-02-03:
> But I'm not sure I understand what you mean by pass through or daisy chain.

A pass through would be when using a panel mounted (on the back of the case) pass-through connector, e.g. spdif that basically connects an optical cable to another optical cable on the inside of the case (optical to optical), that then connects to the optical connector on the board.
A daisy chain is similar but in this case it is a panel mounted socket, e.g. power that is then wired to a plug that then plugs into the same type of socket on the board.

Have a look at the what I mean in the image below!

Rather than do this, I would rather be able to wire the panel mounted sockets to JST plugs (or similar) that can plug into sockets on the board, bypassing the connector (SPDIF, headphone, power, speaker, etc) sockets on the board. Oh, and not those tiny 1.25mm Molex Microlock 2 connectors! They are so difficult to source pre-crimped cables for.

[image]

sle118, 2024-02-04:
Ok I see what you mean. I'm guessing a ribbon connector could be used to hook up a daughter board for most of the components of my board, but I'm not sure how I would be able to offload the Ethernet connector. Maybe mount the entire PHY layer with the Ethernet controller on the daughter board could be an option, but I don't know how EMF might interfere with stability or worst noise percolating towards the analog stage. 

My design already has a generous gpio header with dedicated display connector, so at least the front panel can be separated easily.

As for the back panel, it would probably be easier to space out the connectors and provide room to screw some brackets. From there, only the back panel would require an adapter design and since the SqueezeIO doesn't have screw in terminals, there wouldn't be a challenge with having the screws exposed.

alidaf, 2024-02-04:
Yes, I see the difficulty for some components and the desire to have a comprehensive feature set. I know TOSlink / SPDIF transmitters need a capacitor connected across the terminals within a short range but I guess that, and ethernet would be up to the hobbyist (case designer) to implement or workaround. Plus, as much as modularity may sound like a great idea, I don't think many people bought Google's modular phone in the end!

It’s your baby at the end of the day and I don’t wish to dictate but any extra extra flexibility would really help.

m1ke02, 2024-03-23:
> Raspiaudio Muse Luxe, AI-Thinker ESP32-A1S Audio Kit and Espressif LyraT v4.3 are quite popular and they all contain the ES8388 DAC.
> 
> This also is a great DAC! Because it has a built-in mixer and multiple outputs too:
> 
> * has two analog stereo inputs (one L/R pair with a switchable microphone preamp)
> * has implicitly an I2S DAC
> * has two analog outputs
> * the inputs can be mixed with the DAC signal to the outputs
> 
> ESP32 Audio Kit with the ESP32-A1S module is a wonderful board because:
> 
> * it has small onboard microphones ready on one of the stereo input pair
> * it has a stereo jack line input on the other pair
> * has a 2x3W onboard amp on one of the output pair
> * has a stereo jack headphone on the other output pair
> * has a GPIO connected to the headphone jack for plug detection
> 
> This DAC could solve alone the whole task of input handling, switching, mixing.
> 
> According to [ES8388 documentation](http://www.everest-semi.com/pdf/ES8388%20DS.pdf) it's very versatile:
> 
> * it has separate gain control on the inputs (independently on L/R channels, registers 39, 42)
> * DAC-only volume (registers 26, 27) and mute (register 29) on L/R
> * it has separate volume control for the sum (mixed) outputs (registers 46-49)
> 
> Since the mixer passes audio through there's no delay in the process, which is beneficial to keep lipsync with the TV picture if used as a soundbar. ESP32 Audio Kit could do this out of the box.
> 
> Afaik this DAC can also digitize the input signal so yes, it could also be usable to generate a stream for the network.

Line input passthrough on ES8388 can be achieved by the following:
dac_controlset {"init": [{"reg":39, "val":208}, {"reg":42, "val":208}, {"reg":38, "val":9}]}
But the input signal is turned into crap with a lot of noise and glitches, even if powered by battery

VincentGijsen, 2024-12-22:
If your really only want pass-through from tv or LMs Audio, here is a thought:

wireup you inbound spdif connector (either toslink or coax) route it thought a multiplexor  chip, with also the outbound esp32 spdif as 2nd input. Switch the multipliplexor chip (or dumb mechanical relay) with the amp-power gpio.

this way: switch player off: relay/multiplexor routes inbound spiff. Turn on player -> esp/lms sound out..

zero software changes needed..

ths916, 2025-05-09:
Hello, I'm interested in Squeezelite with audio pass-through as a first step. However, I couldn't get the line input pass-through to work. Can someone help me?

I have an Audio Kit Board with ESP-A1S. The standard Squeezelite project, including Ethernet (W5500), is running.

Parameters:
dac_config: model=es8388, bck=27, ws=25, do=26, di=35, mclk=0, sda=33, scl=32, i2c=16
eth_config: model=w5500, cs=15, speed=10000000, intr=22
set_gpio: 21=amp, 19=green:0, 39=jack:0
spi_config: data=13, clk=14, miso=12

So far, so good.

If I set any kind of dac_controlset, for example: {"init": [{"reg":39, "val":208}, {"reg":42, "val":208}, {"reg":38, "val":9}]} or only {"init":[{"reg":18,"val":0}]} there is no audio. The log says that mclk is -1 instead of GPIO0. If I clear dac_controlset, mclk is routed to GPIO0, and everything works fine, without input pass-through.

Thank you for your help!
The project is a good job.