sle118/squeezelite-esp32 #228: Add support for toggling GPIO pins through the LMS server

nagyrobi, 2023-03-12:
Please add an option to present a few free GPIOs to LMS as external audio source selectors. For example Logitech Transporter  has two digital inputs which can be selected through LMS, just like with [line-in](https://github.com/sle118/squeezelite-esp32/issues/227), if the board we use doesn't have one, we could build electronics around it which could switch audio path (eg. MAX4908).

Use case, we could use Squeezelite-ESP32 instead of ESPHome in the drawing: https://forum.raspiaudio.com/t/suggestion-for-espmuse-multiple-analog-inputs/401

philippe44, 2023-03-16:
Maybe if one day there is a line-in but I'll differ that to @sle118. There is some code in the LMS player specific code to use line-in, but I've never finalized it

nagyrobi, 2023-03-21:
According to LMS Command Line Interface [documentation](https://github.com/elParaguayo/LMS-CLI-Documentation/blob/master/LMS-CLI.md#notes), it is possible to send an input source selection command from LMS to the Transporter hardware player:

> **Transporter Digital Inputs**
> Transporter Digital Inputs are handled as remote streams, with a URL starting with `source:` followed by `aes-ebu`, `bnc-spdif`, `toslink` or `rca-spdif`.
> To set Transporter to the TOSLINK input, use `<playerid> playlist play source:toslink`
> When set to a digital input, Transporter reports the URL scheme above to the various path, url or status queries. 

So all it would have to be done is that when LMS is trying to play for example `source:toslink`, toggle a GPIO pin.

Would be useful to support any string after the `source:` keyword, not only `aes-ebu`, `bnc-spdif`, `toslink` or `rca-spdif`.

As a first step maybe it would be enough to have a command line option to pass a pair of parameters like:
`source=<sourcestring>:<GPONUM>` where `sourcestring` could be the part coming after `source:` and `GPONUM` could be the number of the pin to be toggled.

So in the example above we could have `source=toslink:15` to turn on GPO15 when LMS asks to play `source:toslink`.

And when reporting back to the various path, url or status queries, just return `sourcestring`.

closed 2023-08-27 by sle118