sle118/squeezelite-esp32 #229: Add IR codes sending option similar to `<amp>` GPO

nagyrobi, 2023-03-12:
The use case is that when connecting the device to an external A/V receiver, it would be great to be able to switch that to the correct input (and back). So it's not only just powering the amp ON and OFF, it's about choosing the correct audio input...

We could connect an IR transmitter LED to [set_GPIO](https://github.com/sle118/squeezelite-esp32#set-gpio) and using similar to how `<amp>` works, transmit raw IR codes to turn on or switch the input of the A/V receiver, and transmit some other IR codes to switch to another input or turn off the A/V receiver.

We could use the already supported IR receiver to capture the IR codes sent by the original remote of the A/V receiver. We'd just note these down and copy/paste later in the config. I'd assume it would be useful sending max 5 codes (button press sequences) would be enough, separated with space, where space would mean about 100ms delay (eg. turn on receiver + switch input = two button presses).

sle118, 2023-03-13:
It is definitely a worthy idea, but not something trivial to implement as far as I know.

I will dedicate most of the sparse time at my disposition (it's been at least two months since I could contribute to the project) to implement some features of my own, and I have no remaining free time to spend on more features. In addition, we're very close to making out the capacity of today 4mb chips and I am unsure how we could fit more.

I would definitely merge someone else's contribution, but it's very unlikely that I'll ever have time to build something as involved as this.

Thank you for the suggestion

philippe44, 2023-03-16:
I'll make this one as "don't" - too much code to be added and w are too tight. And I d need to add all the IR transmit code which I'm not familiar with.

sle118, 2023-03-16:
So in the meantime I'll go ahead and close this

closed 2023-03-16 by sle118

nagyrobi, 2023-03-16:
> I d need to add all the IR transmit code which I'm not familiar with.

You would't, it would be enough to learn/transmit simply in RAW.

philippe44, 2023-03-16:
> You would't, it would be enough to learn/transmit simply in RAW. Just like you do now, when forward to LMS.
>
Yes but it's still additional code and I cannot make it a priority currently as my main objective as far as esp32 is concerned is to integrate the new cspot version and I don't know what will be the result in term of memory. I also have plenty of other projects and this feature is not something I'd use, so I can't allocate time to it - sorry