sle118/squeezelite-esp32 #231: FLAC files with empty SEEKTABLE metadata block do not play on squeezelite-esp32 players

CornelisJ, 2023-03-15:
A member of the Squeezebox forum has reported that some of his FLAC files do not play on his Polyvection DAC32 player (with latest DAC32 firmware based on squeezelite-esp32). On request he made one FLAC file that does not play on his DAC32 available for download. It was confirmed by another DAC32 owner that this FLAC does not play on his DAC32. And the same FLAC also does not play on my ESP Muse Luxe player with latest squeezelite-esp32 firmware (v1.0-1254-16). The FLAC does play on all other tested players including SB Boom and piCorePlayer with latest squeezelite firmware.

When trying to play the uploaded FLAC, squeezelite-esp32 gives the following error message:
flac_decode:267 flac error: FLAC__STREAM_DECODER_MEMORY_ALLOCATION_ERROR

Investigation shows that the uploaded FLAC file contains an "empty" SEEKTABLE metadata block with following content:
```
METADATA block #2
  type: 3 (SEEKTABLE)
  length: 0
  seek points: 0
```

It appears that the FLAC can be made playable on ESP32 players, without re-encoding the audio, by just removing this "empty" seektable metadata block. Further details can be found in following forum post:
https://forums.slimdevices.com/forum/user-forums/3rd-party-hardware/109412-announce-squeezelite-esp32-dedicated-thread?p=1633730#post1633730

It would be nice if squeezelite-esp32 can be updated such that FLAC files with "empty" seektable can be played on ESP32 players, considering that such FLAC files are playable on other tested players.

The example FLAC file that does not play with existing squeezelite-esp32 firmware is attached to this message.
Regards, CJS

[ESP32_NOPLAY.zip](https://github.com/sle118/squeezelite-esp32/files/10976892/ESP32_NOPLAY.zip)

philippe44, 2023-04-01:
It's corrected in the non-yet published releases. You can try it from the build artifacts

philippe44, 2023-04-01:
Now fixed and verified

closed 2023-04-01 by philippe44